ghdl/ghdl #985: configure fails: Sorry, you need 'cmp' to configure GHDL.

yurivict, 2019-10-22:
It fails on FreeBSD:
```
cd /usr/ports/cad/ghdl/work/ghdl-0.36-1072-ga08e8ce8 && /usr/bin/env PATH=/usr/local/gcc6-aux/bin:/usr/ports/cad/ghdl/work/.bin:/home/yuri/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin  ADA_PROJECT_PATH=/usr/local/lib/gnat MAKE=gmake XDG_DATA_HOME=/usr/ports/cad/ghdl/work  XDG_CONFIG_HOME=/usr/ports/cad/ghdl/work  HOME=/usr/ports/cad/ghdl/work SHELL=/bin/sh CONFIG_SHELL=/bin/sh ADDR2LINE="/usr/local/bin/addr2line" AR="/usr/local/bin/ar" AS="/usr/local/bin/as" CPPFILT="/usr/local/bin/c++filt" GPROF="/usr/local/bin/gprof" LD="/usr/local/bin/ld" NM="/usr/local/bin/nm" OBJCOPY="/usr/local/bin/objcopy" OBJDUMP="/usr/local/bin/objdump" RANLIB="/usr/local/bin/ranlib" READELF="/usr/local/bin/readelf" SIZE="/usr/local/bin/size" STRINGS="/usr/local/bin/strings" CMAKE_PREFIX_PATH="/usr/local" CONFIG_SITE=/usr/ports/Templates/config.site lt_cv_sys_max_cmd_len=262144 ./configure  --with-llvm-config=/usr/local/bin/llvm-config90 --prefix=/usr/local/ghdl
Sorry, you need 'cmp' to configure GHDL.
```

```cmp``` exists:
```
$ cmp
usage: cmp [-l | -s | -x] [-hz] file1 file2 [skip1 [skip2]]
```

```
$ cmp --help
cmp: unrecognized option `--help'
usage: cmp [-l | -s | -x] [-hz] file1 file2 [skip1 [skip2]]
```

1138-4EB, 2019-10-22:
The line to blame is https://github.com/ghdl/ghdl/blob/master/configure#L167. @tgingold, maybe we should use `which cmp` or `command -v cmp` instead?

yurivict, 2019-10-22:
Also: ```cmp``` is not a way to compare variables in shell scripts. From the ```cmp(1)``` man page:
```
NAME
     cmp — compare two files

SYNOPSIS
     cmp [-l | -s | -x] [-hz] file1 file2 [skip1 [skip2]]
```

Variable comparisons is done using ```test(1)``` or ```[ s1 = s2 ]```.

1138-4EB, 2019-10-22:
Do you mean https://github.com/ghdl/ghdl/blob/master/configure#L195?

yurivict, 2019-10-22:
Yes, and other similar places.

1138-4EB, 2019-10-22:
That line is comparing a file with stdin, which is another kind of file. Do you suggest to use `cat` or `sed` instead of `cmp`?

yurivict, 2019-10-22:
You should cat them both into variables, and compare as variables.

1138-4EB, 2019-10-22:
Thanks for clarifying. Pinging @tgingold, since he chose `cmp`. I don't know if it was because of some cross-platform compatibility issue with `cat`.

tgingold, 2019-10-22:
There are 2 uses of cmp: the first to compare configure files, the second to compare version.py
Feel free to propose a better solution.

yurivict, 2019-10-22:
```diff``` is a normal way to compare files, ```test``` or ```[ $var1 = $var2 ]``` is a normal way to compare variables.

tmeissner, 2019-10-22:
I think, you only use `diff` if you are interested what the difference is. I use `cmp` if I only want to know if they are equal or not.

yurivict, 2019-10-22:
```diff -q file1 file2 > /dev/null``` only compares files w/out the details.

1138-4EB, 2019-10-22:
The main point in this issue is a check I introduced when configuration failed on MSYS. The check is not cross-platform compatible, as `cmp` in FreeBSD does not support `--help`. That can be fixed with either `which` or `command`, as suggested.

@yurivict commented that `cmp` should not be used to compare variables. In `configure`, none of the two uses of `cmp` is to compare variables. The first one is to compare files, and the second one is to compare the content of a file with stdin. Then, @yurivict suggested that there are other similar places, to which @tgingold answered that there are not. Therefore, before discussing further modifications, I think we should find/point where is `cmp` used to compare two variables. 

Alternatively, I think we should be discussing why `diff` is more cross-platform compatible than `cmp` and what the check for it would be. Otherwise, discussing the syntax of `cmp` vs `diff` is pointless, as it is just a matter of preference/style.

I agree with changing line 195 to use `cat` or `sed`, instead of either `cmp` or `diff`. But I'd like to hear some argument about line 172.

PR #986 (2019-10-23): fix(configure): use '/dev/null' to check 'cmp', use 'sed' to compare versions
Close #985
closed 2019-10-23 by tgingold