ghdl/ghdl #987: Please clarify the backends section in README

yurivict, 2019-10-23:
It says that 3 backends are supported but LLVM is just typed after the table. Could you please make sure all of them are in the table.

It also says "easier to build". What does this mean? It either builds or not.

I am trying to restore the FreeBSD port for ghdl which was deleted because it fell into disrepair. The descriptions there said ```GCC (mature)``` and ```LLVM (experimental)```. Is this currently accurate? What versions of LLVM/GCC are supported? Any versions?

tgingold, 2019-10-23:
Easier to build means less steps or less dependencies.

I don't think that LLVM is experimental anymore.

yurivict, 2019-10-23:
You should just make it easier to read.

1138-4EB, 2019-10-23:
- The backends are mentioned in https://github.com/ghdl/ghdl#main-features
- A short table is shown in https://github.com/ghdl/ghdl#building-ghdl
- A larger table, along with multiple hints, is shown in https://ghdl.readthedocs.io/en/latest/getting/

> You should just make it easier to read.

Please, either suggest specific enhancements, or open a PR. 'easier to read' is, at least, as undefined as 'easier to build'.

For example, do you suggest to remove the table from the README and refer users to the docs directly?

> What versions of LLVM/GCC are supported? Any versions?

According to the docs (https://ghdl.readthedocs.io/en/latest/getting/LLVM.html#build-llvm and https://ghdl.readthedocs.io/en/latest/getting/GCC.html#build-gcc):

- LLVM (Low-Level-Virtual Machine) and CLANG (Compiler front-end for LLVM): 3.5, 3.8, 3.9, 4.0, 5.0, 6.0, 7.0, 8.0 or 9.0
- GCC source files. Download and untar the sources of version 4.9.x, 5.x, 6.x or 7.x.

However, I think that the list of GCC versions is outdated. GHDL is tested with GCC 8.x too. There might be issues with GCC 9.x (#883).

tgingold, 2020-03-04:
After re-reading the README, I think it is clear enough. PR are the best way to improve docs.

closed 2020-03-04 by tgingold