ghdl/ghdl #989: Build ignored the '-j NCPUs' option

yurivict, 2019-10-23:
It always builds on one CPU, when normally projects should build on multiple CPUs.

tgingold, 2019-10-23:
Yes, but `make` is not able to pass the `-j NCPU` option to a sub-builder (expected itself).
You can use: `make GNATMAKE="gnatmake -j4"`

yurivict, 2019-10-23:
```MAKEFLAGS``` or ```MFLAGS``` contain ```-jN``` supplied to GNU make.

1138-4EB, 2019-10-23:
> `MAKEFLAGS` or `MFLAGS` contain `-jN` supplied to GNU make.

Those variables are passed to GNU make, but not to gnatmake, which is the (sub-builder) tool that does the heavy compilation work actually. The syntax suggested by @tgingold will ensure that `-jN` is passed to gnatmake. See:

https://github.com/ghdl/ghdl/blob/a08e8ce832701dd0832330e783009b20b34f6782/Makefile.in#L596-L603

tgingold, 2019-10-23:
With gnu make, MAKEFLAGS does not contain -jN but parameters for the job server.

I fear there is no easy solutions.  Any proposal is welcome.

closed 2019-10-23 by tgingold