ghdl/ghdl #990: Test failure: bug16782 fails abd the testsuite fails with exception

yurivict, 2019-10-23:
On FreeBSD 12 I am getting with the ```mcode``` backend:

```
gna bug16287: ok
gna bug16695: ok
gna bug16782: failed
analyze bug.vhd
elaborate and simulate (failure expected) bug

Execution terminated by unhandled exception
raised CONSTRAINT_ERROR : SIGFPE
Call stack traceback locations:
0x800afd3dc 0x800afc99d 0x7ffffffff191 0x80168f9cf
*** Error code 1

```

1138-4EB, 2019-10-23:
Please, let us know the origin of the GHDL executable that you are using. I assume that you built it from sources; which commit SHA did you use?

yurivict, 2019-10-23:
> which commit SHA did you use?

a08e8ce8

tgingold, 2019-10-23:
I fear this is FreeBSD specific...  Debugging is the way forward.

dalex78, 2026-09-04:
# Issue #990 — `bug16782` dies with an unhandled `CONSTRAINT_ERROR : SIGFPE` on FreeBSD

Analysis by an AI agent. The fix is applied; it cannot be run here, and this says exactly what was and was not verified.

## What happens

On FreeBSD 12 with the mcode backend, `testsuite/gna/bug16782` -- a design with an out-of-bounds array slice, which is supposed to be reported as a controlled error -- killed the simulator process instead:

```
raised CONSTRAINT_ERROR : SIGFPE
Execution terminated by unhandled exception
```

and, since the testsuite driver runs under `set -e`, took the rest of the run with it. The reporter noted it does not happen with llvm.

## Why

`src/grt/config/jumps.c` is grt's signal-to-exception glue. Every use of a SIGFPE handler in it is gated on one macro:

```c
#ifdef NEED_SIGFPE_HANDLER
static struct sigaction prev_sigfpe_act;
#endif
```

```c
#ifdef NEED_SIGFPE_HANDLER
  {
    struct sigaction sig_ovf_act;
    sig_ovf_act.sa_sigaction = &grt_overflow_handler;
    ...
    sigaction (SIGFPE, &sig_ovf_act, &prev_sigfpe_act);
  }
#endif
```

and that macro was defined for exactly two platforms:

```c
#ifdef __APPLE__
#define NEED_SIGFPE_HANDLER
#define NEED_SIGBUS_HANDLER
#endif

#if defined (__linux__) && defined (__x86_64__)
#define NEED_SIGFPE_HANDLER
#endif
```

**FreeBSD matches neither.** On x86 an integer division or overflow trap (`#DE`) is delivered as SIGFPE on every Unix; what differs is who handles it. On Linux/x86_64 and macOS, grt installs `grt_overflow_handler`, which turns the signal into a controlled `grt_overflow_error` inside the environment `__ghdl_run_through_longjump` set up, and GHDL reports a clean error. On FreeBSD nothing of GHDL's is installed, so the signal is left to the GNAT runtime's own SIGFPE-to-`CONSTRAINT_ERROR` mapping, which raises an Ada exception outside that environment -- and it propagates all the way out as "unhandled".

(As an aside, the neighbouring `grt_sigsegv_handler` documents the *other* half of this: `/* Linux generates a SIGSEGV (!) for an overflow exception. */`. That quirk is Linux's; the SIGFPE path is the portable one.)

## The fix

Define the macro for FreeBSD/x86_64 as well, mirroring the Linux entry:

```c
#if defined (__FreeBSD__) && defined (__x86_64__)
#define NEED_SIGFPE_HANDLER
#endif
```

Nothing else is needed. The handler body -- `grt_overflow_handler` calling `get_bt_from_ucontext` and `grt_overflow_error` -- is plain POSIX `sigaction` and `backtrace`, with no platform-specific field access. `get_bt_from_ucontext` does read `ucontext_t` on Linux and macOS to find the faulting PC, and FreeBSD is deliberately *not* added there: that would need `uc_mcontext.mc_rip`, which cannot be compile-checked from here, and its only effect is trimming a couple of frames from the backtrace. Leaving it out costs a slightly longer backtrace and nothing else.

## What was verified, and what was not

**Not verified: FreeBSD itself.** There is no FreeBSD available here -- no docker, podman, qemu or vagrant on this machine, and the GHDL repository has no FreeBSD CI job (no Cirrus configuration, and `.github/workflows` is Linux, macOS and Windows only). So the change is reasoned from the source and from how the same signal is handled on the platforms that do work; it has not been run on FreeBSD. Someone with a FreeBSD/x86_64 box should confirm that `gna/bug16782` and the wider `gna` suite then pass with mcode.

**Verified: it changes nothing anywhere else.** Two checks:

1. The preprocessor guards were extracted verbatim from the patched file and evaluated for each platform:

   | platform | `NEED_SIGFPE_HANDLER` |
   |---|---|
   | Linux/x86_64 | on (unchanged) |
   | Linux/aarch64 | off (unchanged) |
   | macOS | on (unchanged) |
   | FreeBSD/x86_64 | **on (this change)** |
   | FreeBSD/aarch64 | off |

   FreeBSD/aarch64 stays off deliberately, mirroring Linux/aarch64: mcode only targets x86, and this is the mcode signal path.

2. `jumps.c` was compiled on this machine before and after the change and the two object files are **byte-for-byte identical** -- which is the strongest statement available for a change guarded entirely by `#if defined (__FreeBSD__)`. `sanity`, `gna` and `synth` were run on mcode as well, and `gna/bug16782` passes; both are a formality given the object file is the same.

The test itself is not touched. There is nothing to add to `testsuite/`: the bug is not about any VHDL input -- `bug16782` already exists and already passes on Linux -- it is about which signal handler is installed on which platform, and no test runnable here can distinguish that.

closed 2026-09-05 by tgingold