ghdl/ghdl #991: gcc backend fails to build:

yurivict, 2019-10-23:
The ```all``` target says:
```
For installing sources in gcc, do:
1) gmake copy-sources
2) configure, build and install gcc
3) gmake ghdllib
4) gmake install
```
but then the suggested ```copy-sources ghdllib``` targets fail:
```
(cd /usr/ports/cad/ghdl/work/ghdl-0.36-1072-ga08e8ce8; if ! /usr/bin/env PATH=/usr/local/gcc6-aux/bin:/usr/ports/cad/ghdl/work/.bin:/home/yuri/bin:/sbin:/bin:/usr/sbin:/usr/bin:/usr/local/sbin:/usr/local/bin  ADA_PROJECT_PATH=/usr/local/lib/gnat PERL_USE_UNSAFE_INC=1 XDG_DATA_HOME=/usr/ports/cad/ghdl/work  XDG_CONFIG_HOME=/usr/ports/cad/ghdl/work  HOME=/usr/ports/cad/ghdl/work NO_PIE=yes MK_DEBUG_FILES=no MK_KERNEL_SYMBOLS=no SHELL=/bin/sh NO_LINT=YES ADDR2LINE="/usr/local/bin/addr2line" AR="/usr/local/bin/ar" AS="/usr/local/bin/as" CPPFILT="/usr/local/bin/c++filt" GPROF="/usr/local/bin/gprof" LD="/usr/local/bin/ld" NM="/usr/local/bin/nm" OBJCOPY="/usr/local/bin/objcopy" OBJDUMP="/usr/local/bin/objdump" RANLIB="/usr/local/bin/ranlib" READELF="/usr/local/bin/readelf" SIZE="/usr/local/bin/size" STRINGS="/usr/local/bin/strings" PREFIX=/usr/local  LOCALBASE=/usr/local  CC="ada" CFLAGS="-O2 -pipe -fno-omit-frame-pointer  -DLIBICONV_PLUG -fstack-protector-strong -fno-strict-aliasing "  CPP="cpp" CPPFLAGS="-fno-omit-frame-pointer -DLIBICONV_PLUG"  LDFLAGS=" -fstack-protector-strong " LIBS=""  CXX="c++" CXXFLAGS="-O2 -pipe -fno-omit-frame-pointer -DLIBICONV_PLUG -fstack-protector-strong -fno-strict-aliasing -fno-omit-frame-pointer  -DLIBICONV_PLUG "  MANPREFIX="/usr/local" BSD_INSTALL_PROGRAM="install  -s -m 555"  BSD_INSTALL_LIB="install  -s -m 0644"  BSD_INSTALL_SCRIPT="install  -m 555"  BSD_INSTALL_DATA="install  -m 0644"  BSD_INSTALL_MAN="install  -m 444" gmake -f Makefile -j8  copy-sources ghdllib; then  if [ -n "Try to set MAKE_JOBS_UNSAFE=yes and rebuild before reporting the failure to the maintainer." ] ; then  echo "===> Compilation failed unexpectedly.";  (echo "Try to set MAKE_JOBS_UNSAFE=yes and rebuild before reporting the failure to the maintainer.") | /usr/bin/fmt 75 79 ;  fi;  false;  fi)
gmake[2]: Entering directory '/usr/ports/cad/ghdl/work/ghdl-0.36-1072-ga08e8ce8'
if test -d ./.git \
   && desc=`cd .; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
else \
  sub="s/tarball/tarball/"; \
        fi; \
sed -e "$sub" -e "s/@VER@/0.37-dev/" < src/version.in > version.tmp; \

if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
rm -f -rf /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
mkdir /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p version.ads ghdlsynth_maybe.ads /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/*.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/vhdl/*.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/vhdl/translate/*.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/synth/*.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/psl/*.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/grt/grt.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/grt/grt-types.ads /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/grt/grt-c.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/grt/grt-fcvt.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/grt/grt-algos.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/grt/grt-vstrings.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/ortho/*.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/ortho/gcc/*.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/ortho/gcc/*.c /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./src/ortho/gcc/*.opt /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
base_ver=`cat /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/BASE-VER`; \
case $base_ver in \
 4.9*)     gcc_ortho_lang=ortho-lang-49.c ;; \
 5.*)      gcc_ortho_lang=ortho-lang-5.c ;; \
 6 | 6.*)  gcc_ortho_lang=ortho-lang-6.c ;; \
 7.*)      gcc_ortho_lang=ortho-lang-7.c ;; \
 8.*)      gcc_ortho_lang=ortho-lang-8.c ;; \
 9.*)      gcc_ortho_lang=ortho-lang-9.c ;; \
 *) echo "Mismatch gcc version from /usr/ports/lang/gcc9/work/gcc-9.2.0"; \
    echo "Need gcc version 4.9.x, 5.x, 6.x, 7.x, 8.x or 9.x"; \
    exit 1 ;; \
esac; \
cp -p ./src/ortho/gcc/$gcc_ortho_lang \
  /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl/ortho-lang.c
cp -p ./doc/ghdl.texi ./doc/ghdl.1 /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
mkdir /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl/ghdldrv
cp -pR ./src/ghdldrv/*.ad? /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl/ghdldrv
cp -p ./dist/gcc/Make-lang.in /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./dist/gcc/config-lang.in /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./dist/gcc/lang-options.h /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
cp -p ./dist/gcc/lang-specs.h /usr/ports/lang/gcc9/work/gcc-9.2.0/gcc/vhdl
gmake -f ./libraries/Makefile.inc LIBSRC_DIR="./libraries" LIBDST_DIR="lib/ghdl" enable_openieee="false" LN="ln -s" CP="cp" MKDIR="mkdir" GHDL="/usr/local/bin/ghdl" GHDL_FLAGS=" " vhdl.libs.all libs.vhdl.standard
gmake[3]: Entering directory '/usr/ports/cad/ghdl/work/ghdl-0.36-1072-ga08e8ce8'
gmake[3]: *** No rule to make target '/usr/local/bin/ghdl', needed by 'lib/ghdl/std/v87/std-obj87.cf'.  Stop.
gmake[3]: Leaving directory '/usr/ports/cad/ghdl/work/ghdl-0.36-1072-ga08e8ce8'
gmake[2]: *** [Makefile:266: libs.vhdl.gcc] Error 2
gmake[2]: Leaving directory '/usr/ports/cad/ghdl/work/ghdl-0.36-1072-ga08e8ce8'
===> Compilation failed unexpectedly.
Try to set MAKE_JOBS_UNSAFE=yes and rebuild before reporting the failure to
the maintainer.
*** Error code 1
```

FreeBSD 12

tgingold, 2019-10-23:
To be pedantic, `copy-sources ghdllib` was not suggested.
The actions must be done in order.

yurivict, 2019-10-23:
I already have gcc sources extracted, built and installed. So these targets are what is remaining.

tgingold, 2019-10-23:
No, as `copy-sources` will put ghdl files in the gcc sources, so that the gcc backend for ghdl is built.

yurivict, 2019-10-23:
I see.

yurivict, 2019-10-23:
Is there any real benefit in having the gcc backend, provided that the llvm backend works fine?

tgingold, 2019-10-23:
The gcc backend provides coverage (and debugging).  Sometimes used.

closed 2019-10-24 by tgingold