ghdl/ghdl #1001: synth: Memories with (a)sync resets aren't supported

tmeissner, 2019-11-03:
**Description**
Memories (arrays of arrays) with (a)sync resets aren't supported. They lead to an error in Yosys like

Sync Reset:
```
vai_reg0.vhd:27:7:note: write to a non memory "s_register"
Importing module vai_reg.
ERROR: Unsupported(1): instance \6 of $memidx.
```

Async Reset:

```
vai_reg0.vhd:18:10:note: write to memory "s_register" is not synchronous
Importing module vai_reg.
ERROR: Unsupported(1): instance \6 of $memidx.
```

**Expected behaviour**
Memories with resets should be supported.

**How to reproduce?**

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;

entity test is
  port (
    Reset_n_i    : in  std_logic;
    Clk_i        : in  std_logic;
    Wen_i        : in  std_logic;
    Addr_i       : in  natural range 0 to 2**8-1;
    Din_i        : in  std_logic_vector(7 downto 0);
    Dout_o       : out std_logic_vector(7 downto 0)
  );
end entity test;

architecture rtl of test is

  type t_register is array(0 to 7) of std_logic_vector(7 downto 0);
  signal s_register : t_register;

begin

  Dout_o <= s_register(Addr_i);

  WriteP : process (Clk_i) is
  begin
    if rising_edge(Clk_i) then
      if Reset_n_i = '0' then
        s_register <= (others => (others => '0'));
      else
        if Wen_i = '1' then
          s_register(Addr_i) <= Din_i;
        end if;
      end if;
    end if;
  end process WriteP;

end architecture rtl;
```

```sh :image: ghdl/ghdl:formal
ghdl -a --std=08 test.vhd
yosys -m ghdl -p "ghdl --std=08 test"
```

**Context**

- OS: Debian Buster
- Origin:
  - Built from sources: `534f39a`

tmeissner, 2019-11-05:
The two cases are working now with the latest commits in GHDL and GHDLsynth 👍 

I have tested it with git commits
GHDL: `121afa1`
GHDLsynth : `4d1d02b`

1138-4EB, 2019-11-05:
@tmeissner, can this be closed? Or should it remain open until the examples are added to the testsuite?

tgingold, 2019-11-06:
Added.  Thanks!

closed 2019-11-06 by tgingold