ghdl/ghdl #1004: synth: Initialisation of memory from file hits exception

antonblanchard, 2019-11-05:
Code using readline/hread is hitting an exception:

```vhd :file: test.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use std.textio.all;

entity test is
    generic(
        MEMORY_SIZE   : natural := 4096;
        RAM_INIT_FILE : string := "firmware.hex"
        );
end entity test;

architecture behaviour of test is
    type ram_t is array(0 to (MEMORY_SIZE / 8) - 1) of std_logic_vector(63 downto 0);

    impure function init_ram(name : STRING) return ram_t is
        file ram_file : text open read_mode is name;
        variable ram_line : line;
        variable temp_word : std_logic_vector(63 downto 0);
        variable temp_ram : ram_t := (others => (others => '0'));
    begin
        for i in 0 to (MEMORY_SIZE/8)-1 loop
            exit when endfile(ram_file);
            readline(ram_file, ram_line);
            hread(ram_line, temp_word);
            temp_ram(i) := temp_word;
        end loop;

        return temp_ram;
    end function;

    signal memory : ram_t := init_ram(RAM_INIT_FILE);
begin
end architecture behaviour;
```

```sh :image: ghdl/synth:beta
ghdl -a --std=08 test.vhdl
yosys -m ghdl -p "ghdl --std=08 test"
```

The exception hit:

```
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-context.adb:332 access check failed
```

tmeissner, 2019-11-05:
Mhm, I wonder how such code should be synthesized. For my understanding, the file I/O is simulation code only

1138-4EB, 2019-11-05:
@tmeissner, related to #1005, I was about to comment the same. However, I think that this might be a non-standard feature that is supported by some vendors. IIRC, Xilinx uses initialization values of signals as reset values for memory elements (FFs, BRAMs, etc.), when no explicit reset signal/value is defined. So, in this example, the content of RAM_INIT_FILE is used to define the content of the corresponding BRAM(s) in the bitstream. Unfortunately, I have absolutely no idea about how is it done internally.

**EDIT**

https://forums.xilinx.com/t5/Vivado-Debug-and-Power/using-reset-value-as-an-initial-register-value-for-Vivado/m-p/674239/highlight/true#M9015

antonblanchard, 2019-11-05:
Interesting, I didn't realise this was a non standard way to initialize a BRAM. It looks like it started working on Vivado a few years ago, not sure about the other toolchains.

1138-4EB, 2019-11-05:
> As far as I know this is the most portable way to initialize a RAM from a file.

Does this mean that it is valid to use any non-sythesisable construct to define initial values of arrays because all of it is resolved during elaboration?

tgingold, 2019-11-05:
Yes, that's the idea.  Except maybe foreign calls.

tgingold, 2019-11-06:
I don't plan to work immediately on this issue (which is adding support for files and access types to synthesis).  Unless it is high priority for you.

antonblanchard, 2019-11-12:
Thanks for looking at this - we use it in microwatt to load programs into RAM, so it is of use.

tgingold, 2019-11-23:
The testcase is now supported.

Maybe there are still holes in the current implementation for other use of files/accesses, but at least the infrastructure is present.

Supporting files came earlier than expected.  I now have to work on memories.

closed 2019-11-23 by tgingold