ghdl/ghdl #1005: synth: Simulation commands (eg stop) result in a failure

antonblanchard, 2019-11-05:
The following code:

```vhd :file: test.vhdl

library ieee;
use ieee.std_logic_1164.all;
use std.env.stop;

entity test is
    generic (
        SIM : boolean := false
        );
    port (
        val : in std_ulogic
        );
end entity test;

architecture behaviour of test is
begin
    process_0: process(all)
    begin
        if SIM and val = '1' then
            stop;
        end if;
    end process;
end architecture behaviour;
```

```sh :image: ghdl/synth:beta
ghdl -a --std=08 test.vhdl
yosys -m ghdl -p "ghdl --std=08 test:"
```

fails because it tries to link in simulation control code:

```
../../src/std/env-body.vhdl:40:5:error: call to foreign procedure "control_simulation" is not supported
```

I would have thought all the simulation stuff should just get dropped when synthesizing, but I'm not sure.

1138-4EB, 2019-11-05:
@antonblanchard, even though it is undocumented, I think that you can use `--synthesis translate_off` and `--synthesis translate_on` to tell which code blocks to ignore for synthesis.

Refs: https://github.com/ghdl/ghdl/issues/903#issuecomment-541388163, [October 24, 2019 1:12 AM](https://gitter.im/ghdl1/Lobby?at=5db0de652a6494729c2a454f), https://github.com/ghdl/ghdl/search?q=translate_off&unscoped_q=translate_off

> I would have thought all the simulation stuff should just get dropped when synthesizing, but I'm not sure.

IMHO, it is not good practice to rely on the tools to drop parts of the design. In a *properly described* design the hardware should be clearly differentiated from simulation/verification code. This is because you sometimes describe helper functions/signals for simulation and it is not always easy for the tool to decide what to drop. Moreover, when checking the logs, it is harder to spot actual bugs (undesired optimizations). Does this sound sensible to you? Do you have 'many' simulation-only features mixed with actual hardware?

antonblanchard, 2019-11-05:
@1138-4EB makes sense, and I wasn't sure how best to proceed here. We've been using a SIM generic to wrap sim only things in microwatt with the hope of it optimising away on synthesis.

This worked on one proprietary toolchain, but appears to be having problems on another, so this may not be a great idea.

1138-4EB, 2019-11-05:
Using a generic only is not ideal, because you are relying on the tool to optimize a DAG, considering it knows that the generic is both constant and `false`. However, you can combine the generic with conditional `generate` statements. Note that this is a VHDL 2008 feature; it is supported by GHDL (#106), but it might fail with other tools.

In fact, I think that generic + conditional generate is the idiomatic (standard) way of doing `--synthesis translate_off`. But I'd say that the latter is supported by more tools.

closed 2019-11-06 by tgingold

1138-4EB, 2019-11-06:
@tgingold, does this mean that GHDL will attempt to automatically remove/ignore uneffective simulation-only statements during synthesis (and optimize processes/modules consequently)?

tgingold, 2019-11-07:
In this case, yes.

commit e4279a1e (2019-11-06): synth: handle short-circuit and.  Fix #1005