ghdl/ghdl #1008: synth: read from memory array hits assertion

antonblanchard, 2019-11-05:
The following code:

```vhd :file: test.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
    port(
         addr_in : in std_logic_vector(11 downto 0);
         dat_out : out std_ulogic_vector(63 downto 0)
         );
end entity test;

architecture behaviour of test is
    type ram_t is array(0 to (4096 / 8) - 1) of std_logic_vector(63 downto 0);
    signal memory : ram_t := (others => (others => '0'));
begin
    dat_out <= memory(to_integer(unsigned(addr_in(11 downto 3))));
end architecture behaviour;
```

```sh :image: ghdl/synth:beta
ghdl -a --std=08 test
ghdl --synth --std=08 test
```

Hits an assertion:

```
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists.adb:596
```

1138-4EB, 2019-11-05:
I feel that this might be a duplicate of tgingold/ghdlsynth-beta#69, but I am not sure.

tgingold, 2019-11-06:
No, it is a duplicate of undriven signals.

tgingold, 2019-11-12:
It now passes, but the write is asynchronous.  Not sure if it is realistic.

1138-4EB, 2019-11-12:
@tgingold, did you mean 'read' instead of 'write'?

tgingold, 2019-11-13:
Oops, you're right.  So it has to be correctly handled.

1138-4EB, 2019-11-13:
Then, I think that it is correct. The example code is describing an asynchronous ROM, isn't it?

tgingold, 2019-11-23:
I keep this issue open, as the memory is expanded to registers.

tgingold, 2019-12-05:
Now fixed.

closed 2019-12-05 by tgingold