ghdl/ghdl #1009: synth: crash when assigning a record to another one

marph91, 2019-11-05:
I have an record, which gets assigned to another record:

```vhd :file: record_test.vhd
entity record_test is
  port (
    o : out integer
  );
end record_test;

architecture rtl of record_test is
  type t_record is record
    int  : integer;
  end record t_record;
  constant rec_constant : t_record := (int => 27.777 us / 83.333 ns);
  constant rec_constant2 : t_record := rec_constant;
begin
  o <= rec_constant2.int;
end rtl;
```

Synthesizing fails with:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1151-g121afa1a) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-suse-linux
/home/workspace/vhdl/test_ghdlsynth/record/build/
Command line:
ghdl --synth
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-context.adb:434 index check failed
Call stack traceback locations:
0x585096 0x58552d 0x58566c 0x5860f7 0x586724 0x5a4305 0x58a526 0x58c97b 0x594587 0x5947e8 0x58a311 0x58a402 0x586977 0x5e3249 0x53354c 0x5ec4f9 0x40379f 0x7f61e466ef88 0x402888 0xfffffffffffffffe
******************************************************************
```
This only seems to fail when calculating the integer by division of two variables of type `time`.

closed 2019-11-06 by tgingold

arcturus140, 2019-11-06:
isn't a record required to have at least 2 elements?

diogratia, 2019-11-06:
The BNF in IEEE Std 1076-2008 5.3.3 Record types is authoritative where braces allow zero or more subsequent elements:

>record_type_definition ::=  
&nbsp;&nbsp;&nbsp;&nbsp;**record**  
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;element_declaration  
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;{ element_declaration }  
&nbsp;&nbsp;&nbsp;&nbsp;**end record** [ *record_type*_simple_name ]

1.3.2 Syntactic description
>...  
> f) Braces { } enclose a repeated item or items on the right-hand side of a production. The items may appear zero or more times; the repetitions occur from left to right as with an equivalent left-recursive rule. ...

commit f7887b62 (2019-11-06): synth-expr: do subtype conversion in fill_record_aggregate.  Fix #1009