ghdl/ghdl #1010: libghdl-<version>.dll has incorrect extension .exe

GlenNicholls, 2019-11-06:
`<ghdl>/lib/libghdl-<version>.dll` is generated with the incorrect extension of `.exe`: `libghdl-<version>.dll.exe`.

When setting up ghdl-ls, the following error occurs after installation steps when the extension tries to start in VSC:

```
Traceback (most recent call last):
  File "C:\Users\Glen.Nicholls\AppData\Local\Programs\Python\Python37-32\Scripts\ghdl-ls-script.py", line 11, in <module>
    load_entry_point('ghdl-language-server==0.1.dev0', 'console_scripts', 'ghdl-ls')()
  File "c:\users\glen.nicholls\appdata\local\programs\python\python37-32\lib\site-packages\pkg_resources\__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "c:\users\glen.nicholls\appdata\local\programs\python\python37-32\lib\site-packages\pkg_resources\__init__.py", line 2793, in load_entry_point
    return ep.load()
  File "c:\users\glen.nicholls\appdata\local\programs\python\python37-32\lib\site-packages\pkg_resources\__init__.py", line 2411, in load
    return self.resolve()
  File "c:\users\glen.nicholls\appdata\local\programs\python\python37-32\lib\site-packages\pkg_resources\__init__.py", line 2417, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "c:\users\glen.nicholls\appdata\local\programs\python\python37-32\lib\site-packages\vhdl_langserver\main.py", line 9, in <module>
    import libghdl
  File "c:\users\glen.nicholls\appdata\local\programs\python\python37-32\lib\site-packages\libghdl\__init__.py", line 75, in <module>
    _libghdl_path = _get_libghdl_path()
  File "c:\users\glen.nicholls\appdata\local\programs\python\python37-32\lib\site-packages\libghdl\__init__.py", line 71, in _get_libghdl_path
    raise Exception('Cannot find libghdl {}'.format(basename))
Exception: Cannot find libghdl libghdl-0_37_dev.dll
```

This occurs since the generated `.dll` has an extra extension that is not expected in the ghdl-ls setup. Simply changing the file name to remove `.exe` fixed the problem.

More information can be seen in the ghdl gitter

tgingold, 2019-11-07:
I have pushed a tentative fix.  Can you confirm it works for you ?

1138-4EB, 2019-11-07:
FTR, I'm trying to build the mingw64-llvm PKGBUILD using the latest `master` and it crashes. I think it is unrelated to this fix, tho.

```
gnatbind -Lgrt_ -o run-bind.adb -n -x ghdl_main.ali
cd grt; MSYS2_ARG_CONV_EXCL="-aI;-gnatec" gnatmake -j4 -u -c -g -gnatecT:/vhdl-hub/ghdl/ghdl/dist/msys2-mingw/llvm/src/builddir/../ghdl/src/grt/grt.adc -gnatdY -o run-bind.o run-bind.adb
gcc -c -g -gnatecT:/vhdl-hub/ghdl/ghdl/dist/msys2-mingw/llvm/src/builddir/../ghdl/src/grt/grt.adc -gnatdY run-bind.adb
gcc.exe: internal compiler error: Aborted signal terminated program gnat1
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://sourceforge.net/projects/msys2> for instructions.
gnatmake: "run-bind.adb" compilation error
make: *** [../ghdl/src/grt/Makefile.inc:126: grt/run-bind.o] Error 4
==> ERROR: A failure occurred in build().
    Aborting...
```

1138-4EB, 2019-11-07:
Something was broken on my local MSYS2 setup. After reinstalling, I can confirm that latest `master` works. @GlenNicholls, please reopen if it is still failing for you.

closed 2019-11-07 by 1138-4EB

GlenNicholls, 2019-11-08:
It looks like it works for me. I will let you know if I find any other issues