ghdl/ghdl #1012: Support macOS Xcode 11.1

1138-4EB, 2019-11-08:
Yesterday, I received the following email from GitHub:

> On October 7th Apple released Catalina, the latest update to macOS X. Since then, we’ve been busy preparing our macOS virtual environment for the update and today we’re making the switch.

> Any jobs in your workflows that run on the `macos-10.14` virtual environment must migrate to `macos-latest`. Going forward, we will only support the `macos-latest` environment which will run Catalina (10.15). Once the migration to Catalina is complete, jobs using `macos-10.14` will no longer run. 

> Other changes to the environment include:
>
> - Removed Xcode 10.*
> - Removed Xcode 9.*
> - Xcode 11.1 will be set as default

As a result, the GitHub Actions build on macOS (see #929) is broken: https://github.com/1138-4EB/ghdl/commit/e178192136f3d7594d04aabd52b82a0510e1fe65/checks?check_suite_id=301351962#step:3:680

Because of the error we get, this is probably a duplicate of #959. However, unlike @anquegi's computer, GHA is an environment where we can all run tests.

tgingold, 2019-11-08:
Please, try to use clang to compile the C files (CC=clang).  I am not sure how well gcc (which is the gcc from gnat) can deal with system headers.

1138-4EB, 2019-11-11:
Indeed...

- https://github.com/1138-4EB/ghdl/commit/d57c81cf4a45cd28ae970de07153e45e27ba443d
- https://github.com/1138-4EB/ghdl/runs/296926058

closed 2019-11-11 by 1138-4EB