ghdl/ghdl #1013: Having 2 similar entities in different libraries results in compilation error.

m-kru, 2019-11-08:
Having 2 similar (same name, ports, architecture name) entities in different libraries results in compilation error. I have created simple project to show the issue: [ghdl_bug](https://github.com/m-kru/ghdl_bug). Just clone and run make.

diogratia, 2019-11-08:
```none
diogratia@Macbook: make
ghdl -a --std=08 --work=lib_1 lib_1/my_entity.vhdl
ghdl -a --std=08 --work=lib_2 lib_2/my_entity.vhdl
ghdl -a --std=08 tb_my_entity.vhdl
ghdl --elab-run --std=08 tb_my_entity
Undefined symbols for architecture x86_64:
  "_lib_1__my_entity__ARCH__rtl__DECL_ELAB", referenced from:
      _lib_1__my_entity__LASTARCH__DECL_ELAB in e~tb_my_entity.o
  "_lib_1__my_entity__ARCH__rtl__STMT_ELAB", referenced from:
      _lib_1__my_entity__LASTARCH__STMT_ELAB in e~tb_my_entity.o
  "_lib_1__my_entity__PKG_ELAB", referenced from:
      _work__tb_my_entity__ARCH__tb__my_entity_1__COMP_ELAB in tb_my_entity.o
  "_lib_1__my_entity__RTI", referenced from:
      _work__tb_my_entity__ARCH__tb__my_entity_1__RTI in tb_my_entity.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ghdl:error: compilation error
make: *** [all] Error 1
```
The my_entity.o object file is over written:
```none
diogratia@Macbook: ll
/Users/diogratia/Desktop/ghdl_bug
total 96
drwxr-xr-x@ 13 diogratia  staff   442 Nov  9 05:13 ./
drwxr-x---+ 52 diogratia  staff  1768 Nov  9 05:10 ../
-rw-r--r--@  1 diogratia  staff   233 Nov  9 05:12 Makefile
-rw-------   1 diogratia  staff   447 Nov  9 05:13 e~tb_my_entity.lst
-rw-------   1 diogratia  staff  4240 Nov  9 05:13 e~tb_my_entity.o
drwxr-xr-x@  3 diogratia  staff   102 Nov  9 04:54 lib_1/
-rw-------   1 diogratia  staff   193 Nov  9 05:13 lib_1-obj08.cf
drwxr-xr-x@  3 diogratia  staff   102 Nov  9 04:55 lib_2/
-rw-------   1 diogratia  staff   193 Nov  9 05:13 lib_2-obj08.cf
-rw-------   1 diogratia  staff  3904 Nov  9 05:13 my_entity.o
-rw-------   1 diogratia  staff  9004 Nov  9 05:13 tb_my_entity.o
-rw-r--r--@  1 diogratia  staff  1149 Nov  7 23:47 tb_my_entity.vhdl
-rw-------   1 diogratia  staff   195 Nov  9 05:13 work-obj08.cf
```
and it doesn't contain the missing symbols, reflecting the last analyzed my_entity: 
```none
diogratia@Macbook: nm my_entity.o
0000000000000140 s __UI00000000
                 U ___ghdl_process_add_sensitivity
                 U ___ghdl_sensitized_process_register
                 U ___ghdl_signal_add_direct_driver
                 U ___ghdl_signal_direct_assign
                 U ___ghdl_signal_merge_rti
                 U _ieee__std_logic_1164__ELABORATED
                 U _ieee__std_logic_1164__ELAB_BODY
                 U _ieee__std_logic_1164__rising_edge
                 U _ieee__std_logic_1164__std_logic__RTI
                 U _lib_2__RTI
0000000000000080 T _lib_2__my_entity__ARCH__rtl__DECL_ELAB
000000000000016c S _lib_2__my_entity__ARCH__rtl__INSTSIZE
00000000000000f0 T _lib_2__my_entity__ARCH__rtl__P0__PROC
0000000000000300 S _lib_2__my_entity__ARCH__rtl__P0__RTI
0000000000000178 s _lib_2__my_entity__ARCH__rtl__P0__RTIARRAY
0000000000000170 s _lib_2__my_entity__ARCH__rtl__P0__RTISTR
00000000000002c0 S _lib_2__my_entity__ARCH__rtl__RTI
0000000000000338 s _lib_2__my_entity__ARCH__rtl__RTIARRAY
0000000000000180 s _lib_2__my_entity__ARCH__rtl__RTISTR
00000000000000a0 T _lib_2__my_entity__ARCH__rtl__STMT_ELAB
0000000000000000 T _lib_2__my_entity__DECL_ELAB
0000000000000060 T _lib_2__my_entity__PKG_ELAB
0000000000000250 S _lib_2__my_entity__RTI
0000000000000290 s _lib_2__my_entity__RTIARRAY
0000000000000161 s _lib_2__my_entity__RTISTR
0000000000000050 T _lib_2__my_entity__STMT_ELAB
0000000000000190 S _lib_2__my_entity__clk__RTI
0000000000000155 s _lib_2__my_entity__clk__RTISTR
00000000000001f0 S _lib_2__my_entity__i__RTI
000000000000015d s _lib_2__my_entity__i__RTISTR
0000000000000220 S _lib_2__my_entity__o__RTI
000000000000015f s _lib_2__my_entity__o__RTISTR
00000000000001c0 S _lib_2__my_entity__rst__RTI
0000000000000159 s _lib_2__my_entity__rst__RTISTR
```
You would think the object files ought to be saved in the directory containing the library which would likely entail searching for them in the right place as well.

You could fix this in several ways in a makefile. For example change directory to each library directory before analysis and pass the library directory paths to ghdl for elaboration:

makefile_paths  
```make

all:
	ghdl -a --std=08 --work=lib_1 --workdir=lib_1 lib_1/my_entity.vhdl
	ghdl -a --std=08 --work=lib_2 --workdir=lib_2 lib_2/my_entity.vhdl
	ghdl -a --std=08 -P./lib_1 -P./lib_2 tb_my_entity.vhdl
	ghdl --elab-run --std=08 -P./lib_1 -P./lib_2 tb_my_entity
clean:
	-rm *.o lib_1/*.o lib_2/*.o
	-rm *.cf lib_1/*.cf lib_2/*.cf
```
When invoked:
```none
diogratia@Macbook: make -f makefile_paths
ghdl -a --std=08 --work=lib_1 --workdir=lib_1 lib_1/my_entity.vhdl
ghdl -a --std=08 --work=lib_2 --workdir=lib_2 lib_2/my_entity.vhdl
ghdl -a --std=08 -P./lib_1 -P./lib_2 tb_my_entity.vhdl
ghdl --elab-run --std=08 -P./lib_1 -P./lib_2 tb_my_entity
simulation finished @70ns
```

(And those so inclined might want to change the VHDL file suffixes in the makefile to DOS compatible `.vhd`, which is already bespoken in Windows for Virtual Hard Disk.)

m-kru, 2019-11-09:
Thanks for help @diogratia!

closed 2019-11-09 by m-kru