ghdl/ghdl #1015: Can't compile a package named "std" on a library other than "work".

hdl4fpga, 2019-11-12:
**Description**
Compiling this code 

`package std is
end;
`

ghdl reports the following error 

std.vhd:1:9:error: identifier "std" already used for a declaration
*std_standard*:1:1:error: previous declaration: library "std"

**Expected behaviour**

Compiled with no errors as all the other tools.

**How to reproduce?**

```vhd :file: ent.vhd
package std is
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- Gentoo/Linux OS:
- Origin:
  - [ ] Released binaries: https://github.com/ghdl/ghdl/releases/tag/v0.36
  

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2019-11-12:
What do you means by:  on a library other than "work" ?

hdl4fpga, 2019-11-12:
that I used

ghdl -a --work=hdl4fpga ent.vhd

tgingold, 2019-11-12:
I fear that ghdl is correct here.  According to LRM08 13.2, there is an implicit
```vhdl
library std;
```
in all design units, so you cannot name a library unit ``std``.

hdl4fpga, 2019-11-12:
are you sure ? ... none of the vhdl tools such as modeltech, xilinx ise,
vivado, or diamond lattice even warning about that .
my project uses a library with std and i wanted to use ghdl as simulator.

I think one thing is a library unit and other is a library name that is
defined by the tools and only used in the code to access library units...
but probably i am wrong

sorry

hdl4fpga, 2019-11-13:
I think the attachment can throw some light to the this problem.  As you
can read on 11.2, the first lines state that a design library is not a
design unit, it is *an implementation-dependent storage facility* This the
IEEE std1076-1987 and I don't think it has been changed on the new ones.
So you can name a package *std* even though a library is named that way and
called on a design unit.

So, I'm pretty sure that it is a bug. Even though I'll change the name of
the package to other than *std*

hdl4fpga, 2019-11-13:
The image wasn't attached on the email. Here it is.

[image]

1138-4EB, 2019-11-13:
@hdl4fpga, note that you are reading LRM87, you executed the example according to LRM93, and Tristan is reading LRM08. I am pretty sure that it is not relevant for this discussion, but I think it'd be good if you tested the example with `--std=87` and `--std=08` too.

I read LRM08 (https://ieeexplore.ieee.org/document/4772740) and compared section 13.2 (pages 195-196) to your attachment. It is essentially the same:

> The library denoted by the library logical name STD contains no library units other than packages STANDARD, TEXTIO, and ENV.

Therefore, I think that @hdl4fpga is correct in:

> So you can name a package *std* even though a library is named that way and
called on a design unit.

That is:

```
library std, work;    -- Default and implicit
use std.standard.all; -- Default and implicit
use std.textio.all;   -- Default and implicit
use std.env.all;      -- Default and implicit (>=08 only)

use std.std.all;      -- Not allowed
use std.mypkg.all;    -- Not allowed

use work.mypkg.all;   -- Allowed
use work.std.all;     -- Allowed (not supported in GHDL)

use mylib.mypkg.all;  -- Allowed
use mylib.std.all;    -- Allowed (not supported in GHDL)
```

However, I think that the title of this issue is misleading. GHDL in docker image `ghdl/vunit:llvm` fails when analising `package std is end`. It is not possible to compile a package named `std`, neither to library `work` nor to any other library.

diogratia, 2019-11-13:
The first paragraph on your comment relating to the attachment has a logic fallacy.

>...As you can read on 11.2, the first lines state that a design library is not a
design unit, it is *an implementation-dependent storage facility* This the
IEEE std1076-1987 and I don't think it has been changed on the new ones.
So you can name a package *std* even though a library is named that way and
called on a design unit.

The issue isn't with what the library logical name references in the host environment it's that it's declared in the same declarative region with a package of the same name.

IEEE Std 1076-2008 13.2 Design libraries, para 3:

>A library clause defines logical names for design libraries in the host environment. **A library clause appears as part of a context clause, either at the beginning of a design unit** or within a context declaration. **For the former case, the declaration of each logical name defined by the library clause occurs immediately within the root declarative region associated with the design unit.** For a library clause that appears within a context declarative region, the logical names are not declared; rather, there is an equivalent library clause that declares the logical names (see 13.4).

paras 7-9:

>Every design unit except a context declaration and package STANDARD is assumed to contain the following implicit context items as part of its context clause:  
>  
>**library** STD, WORK; use STD.STANDARD.**all**;  
>  
>Library logical name STD denotes the design library in which packages STANDARD, TEXTIO, and ENV reside (see Clause 16). (The use clause makes all declarations within package STANDARD directly visible within the corresponding design unit; see 12.4.) Library logical name WORK denotes the current working library during a given analysis. Library logical name IEEE denotes the design library in which the mathematical, multivalue logic and synthesis packages, and the synthesis context declarations reside (see Clause 16).  
>  
>The library denoted by the library logical name STD contains no library units other than packages STANDARD, TEXTIO, and ENV.

12.1 Declarative region, para 4:

>In addition to the preceding declarative regions, there is a _root declarative region_, not associated with a portion of the text of the description, but encompassing any given primary unit. At the beginning of the analysis of a given primary unit, there are no declarations whose scopes (see 12.2) are within the root declarative region. Moreover, the root declarative region associated with any given secondary unit is the root declarative region of the corresponding primary unit.

The design file comprised of the primary design unit:
```vhdl
package std is
end;
```
has a root declarative region containing a declaration of the library logical name (for  library work as well).

So is the package declaration in the same declarative region? (Can `std` be overloaded?)

The package declaration occurs within the root declarative region. 12.2 para 2:

>In each of these cases, the declarative region is said to be associated with the corresponding declaration or statement. A declaration is said to occur immediately within a declarative region if this region is the innermost region that encloses the declaration, not counting the declarative region (if any) associated with the declaration itself.

12.3 Visibility para 10:

>**Two declarations that occur immediately within the same declarative region**, other than the declarative region of a block implied by a component instantiation or the declarative region of a generic-mapped package or subprogram equivalent to a package instance or a subprogram instance, **shall not be homographs, unless exactly one of them is the implicit declaration of a predefined operation or is an implicit alias of such an implicit declaration**. In such cases, a predefined operation or alias thereof is always hidden by the other homograph. Where hidden in this manner, an implicit declaration is hidden within the entire scope of the other declaration (regardless of which declaration occurs first); the implicit declaration is visible neither by selection nor directly. For a declarative region of a block implied by a component instantiation or the declarative region of a generic-mapped package or subprogram equivalent to a package instance or a subprogram instance, the rules of this paragraph are applied to the corresponding entity declaration, component declaration, uninstantiated package declaration, or uninstantiated subprogram declaration, as appropriate.

`shall` is mandatory. See 1.3 Structure and terminology of this standard, 1.3.1.

Para 7:

>A declaration is said to be hidden within (part of) an inner declarative region if the inner region contains a homograph of this declaration; the outer declaration is then hidden within the immediate scope of the inner homograph. **Each of two declarations is said to be a _homograph_ of the other if and only if both declarations have the same designator, and they denote different named entities, and either overloading is allowed for at most one of the two, or overloading is allowed for both declarations and they have the same parameter and result type profile (see 4.5.1).**

`designator` is defined 6. Declarations, 6.1 para 3:

[... cut ...]
You'd also find that any name declared in in one these other locations couldn't also be used as a package name:

```vhdl
package foo_pkg is
    constant foo: string := "foo";
    function foo return string;
end package;
```
```none
ghdl -a foo_pkg.vhdl
foo_pkg.vhdl:3:14:error: identifier "foo" already used for a declaration
foo_pkg.vhdl:2:14:error: previous declaration: constant "foo"
ghdl:error: compilation error
```
because of the same rule that two declarations can't have the same designator in the same declarative region unless they can be overloaded.

hdl4fpga, 2019-11-14:
> The first paragraph on your comment relating to the attachment has a logic
> fallacy.
>
> ...As you can read on 11.2, the first lines state that a design library is
> not a
> design unit, it is *an [...]
You cannot declare a llibrary in VHDL, you can only access it, tools
declare libraries, a library name is not a design unit. I don't see the any
logic fallacy there. That is clear when std1076-87 says:  a design library
is *an implementation-dependent storage facility. *A library name is not a
design unit neither primary unit nor secondary.

> Use of the name std in any of these would hide the library logical name
> declared in an outer declarative region. If you've been using std as a
> package name it could have been as other than a primary unit.
Yes, I did use it as primary unit: a package name and I haven't got any
errors or warnings by the tools I going to enumerate later. In the code I
also access std.textio. It works on mentor modeltech and synthesizer tools
such as: Xilinx ISE, Xilinx Vivado and Lattice Diamond. Here you have the
link to the file:
https://github.com/hdl4fpga/hdl4fpga/blob/188b094e89eb16573614fdea9b5622c9e34f98b9/library/common/std.vhd#L1.
if you want to try it.

> In Tristan's comment *library unit* is defined in 13.1 Design units as
> either a primary unit or a secondary unit, wherein there's an implicit
> library clause in the associated root declarative region.
>
> From this you can also take that packages declared other than as primary
> units are not library units, they appear in the declarative region of their
> respective descriptions.
>
> You'd also find that any name declared in in one these other locations
> couldn't also be used as a package name:
>
> package foo_pkg is
>     constant foo: string := "foo";
>     function foo return string;end package;
>
> ghdl -a foo_pkg.vhdl
> foo_pkg.vhdl:3:14:error: identifier "foo" already used for a declaration
> foo_pkg.vhdl:2:14:error: previous declaration: constant "foo"
> ghdl:error: compilation error
>
> because of the same rule that two declarations can't have the same
> designator in the same declarative region unless they can be overloaded.

But I insist: the library is not declared by the code,  it makes the
library visible.

tgingold, 2019-11-14:
Yes, "a library name is not a design unit neither primary unit nor secondary".  That's correct, but so are signals, variables, components and many other named entities.

But when you use a library clause like ``library mylib;``, you declare ``mylib``.  This is what LRM08 says in 13.2: "the declaration of each logical name defined by the library clause occurs immediately within the root declarative region associated with the design unit".

As a consequence, the name ``mylib`` has a scope and a visibility.  Those rules are explained in chapter 12.

In LRM08 section 6.1, this is clearly said:  A declaration is one of the following:
...
— A logical name in a library clause

In this discussion, we don't really care what is behind the logical name.  The problem is that you are trying to declare two names in the same declarative region.

If you deny that a library clause is a declaration, you will have to explain how ``mylib`` denotes the library in a clause like: ``use mylib.all``.

diogratia, 2019-11-14:
> But I insist: the library is not declared by the code,  it makes the
library visible.

I recall having exchanges in the early to mid '90s about the incompletely defined semantics of VHDL serving as a barrier to 'late-adopters' and preserving commercial advantage to 'early-adopters'.

I hold up for your perusal [Issue Report 1050](http://www.eda-twiki.org/isac/IRs-VHDL-93/IR1050.txt) summarized `Visibility outside of design units is not well-defined`.

```none
Description of Problem
----------------------
Consider the folowing example:

   library L;                 --Line 1
   entity E is                --Line 2
                              --Line 3 **
   end E;                     --Line 4
                              --Line 5
   use L.p.all;               --Line 6
   architecture A of E is     --Line 7
                              --Line 8  **_continued
   begin                      --Line 9  **_continued
                              --Line 10 **_continued
   end A;                     --Line 11

The LRM is not clear about the visibility of L outside the architecture 
(line 6). The scope of L starts after the ";" on line 1 and ends at the end of 
the declarative region associated with the entity (in our example line 11). 
The declarative region for the entity e is marked by "**" and "**_continued"
for the continuation inside the architecture outer block. This declarative
region does not include line 6 in our example. However, according to the 
majority interpretation, the use clause on line 6 is legal. It is not clear 
how the scope of L gets out of the declarative region for e-a and then back in.

Proposed Resolution
-------------------
Assume an "outer declarative region" that has the special properties like
the following: 

1. Library logical names and the name of the design unit are considered to be
declared in this "outer or global declarative region."

2. If a string of design files is analyzed in order, the "outer declarative
region" starts at the top of the first analyzed design file and continues
into the next analyzed design file. 

3. The second and subsequent appearances of the same library logical name
in the same library clause or in diferent library clauses defining libraries
for the same design unit have no effect and should be ignored.

4. An architecture body identifier may be the same as the identifier for
the corresponding entity declaration. 

5. Other special rules as required. 
```
The elements of the author's proposed resolution have all since been adopted. The date of this Issue Report was January 10, 1992. The clarification of the root declarative region was present in [IEEE Std P1076-2000/D3 Chapter 10](https://www.csee.umbc.edu/portal/help/VHDL/p1076/P1076_Chap_10.pdf) but didn't make it into the approved revision, subsequently present in IEEE Std 1076-2002.

```none
VASG-ISAC Analysis & Rationale
------------------------------ 
Missing or purely defined semantic for visibility outside design units is  
apparently leading to similar problems as described in the following issue
reports:

IR-0092: Visibility of an architecture body identifier is unclear.

IR-0110: May library names be redefined between the analysis
         of a primary unit and a corresponding secondary unit?

IR-0111: What are the semantics of duplicate library name definitions?

[... 24 lines cut ...]
```

Spelling 'poorly' as 'purely' aside. the effect of the lack of semantic definition of the scope and visibility of library logical names is profound (you'd find that IRs through 259 are VHDL-87 issues). 

You could note that the VHDL products you mentioned (Xilinx ISE, Xilinx Vivado and Lattice Diamond) are all -1993 based and are synthesis software, where semantics can be both at once looser and overly restrictive (see IEEE Std 1076.6-2004 RTL Synthesis, withdrawn, there's also an historic expectation that synthesized code had been previously simulated, and as we see here it's a matter of whose implementation in the face of semantic uncertainty).

Note that scope is defined for declarations. IEEE Std 1076-1993 10.2 Scope of declarations:

>For each form of declaration, the language rules define a certain portion of the description text called the _scope of the declaration_. The scope of a declaration is also called the scope of any named entity declared by the declaration. Furthermore, if the declaration associates some notation (either an identifier, a character literal, or an operator symbol) with the named entity, this portion of the text is also called the scope of this notation. Within the scope of a named entity, and only there, there are places where it is legal to use the associated notation in order to refer to the named entity. These places are defined by the rules of visibility and overloading.

In the context of the Issue Report the example library L's scope and visibility is not clearly defined with far reaching implications. You can also see the implication that the declaration of the library logical name was assumed to occur in the library clause and that a library was a named entity so decorated.

You could also note that in -1993 Section 4 Declarations there's a BNF rule for declarations since deprecated, it appear no where in the standard in a right hand side production that isn't inclusive for declarations (while -2008 as noted by Tristan in 6. Declarations, 6.1 also includes implicit declarations of labels as well as library logical names).

[... cut ...]
That ghdl adheres historically to semantics as defined today is the result of scouring issue reports and otherwise examining the history of the VHDL standard (which is unfortunately incomplete and likely to remain so for any future revisions). Discovering semantic ambiguity results from implementation or comparing implementations whose results vary.

As far as dealing with semantic ambiguity (as opposed to changed syntax and semantics) the single best thing an end-user can do today is use the -2008 standard as a starting point.

closed 2019-12-03 by tgingold