ghdl/ghdl #1019: GCC version running slower then mcode version instead of faster.

janslab, 2019-11-13:
Hi,

I am new to GHDL and am trying to shorten the time that simulations are running. On the machine I am using there was already an mcode version of GHDL installed. Because the GCC version should be faster, I built a GCC version for my machine. When I tested a testbench with it, it was surprisingly enough slower then the mcode version instead of 20% faster. 
The testbench was a pipelined multiplexer and demultiplexer (I am afraid I can not share the code). The tests results were as following.

```
GHDL mcode (with '--wave')          1m41,866s; 100% CPU
GHDL GCC (with '--wave')            2m32,203s; 100% CPU
GHDL GCC (with '--wave' and '-O')   2m24,093s; 100% CPU
GHDL GCC (with '--wave' and '-O2')  2m26,154s; 100% CPU
Questasim (with 'as8')              38.09s; ~198% CPU
```

I measured with the commands 'time' and 'top' (questasim with stopwatch).

The installed mcode version:
```
GHDL 0.37-dev (v0.36-902-g47cb9e6b) [Dunoon edition]
 Compiled with GNAT Version: 7.4.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

the --disp-config
```
command_name: ghdl
command line prefix (--PREFIX): (not set)
environment prefix (GHDL_PREFIX): (not set)
exec prefix (from program name): /usr/local

library prefix: /usr/local/lib/ghdl
library directory: /usr/local/lib/ghdl
default library paths:
 /usr/local/lib/ghdl/ieee/v93/
 /usr/local/lib/ghdl/
```

The installed GCC version:
```
GHDL 0.37-dev (v0.36-1113-g53a18886) [Dunoon edition]
 Compiled with GNAT Version: 7.4.0
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

the --disp-config
```
Paths at configuration:
compiler command: libexec/gcc/x86_64-pc-linux-gnu/7.4.1/ghdl1
assembler command: as
linker command: gcc
default lib prefix: lib/ghdl

command line prefix (--PREFIX): (not set)
environment prefix (GHDL_PREFIX): (not set)
exec prefix (from program name): /home/myname/bin/ghdl-gcc

library prefix: /home/myname/bin/ghdl-gcc/lib/ghdl
library directory: /home/myname/bin/ghdl-gcc/lib/ghdl
compiler path: /home/myname/bin/ghdl-gcc/libexec/gcc/x86_64-pc-linux-gnu/7.4.1/ghdl1
assembler path: /usr/bin/as
linker path: /usr/bin/gcc

default library paths:
 /home/myname/bin/ghdl-gcc/lib/ghdl/ieee/v93/
 /home/myname/bin/ghdl-gcc/lib/ghdl/
```

GCC version:
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.4.0-1ubuntu1~18.04.1' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1) 
```

I built the GHDL GCC version on Linux Mint 19.1 Tessa 'Cinnamon' with the following script that I made.
```bash
#! /bin/bash

GCC_VERSION="7-20181206"
WORKDIR="$HOME/Downloads/src"
INSTALLDIR="$HOME/bin/ghdl-gcc"

mkdir -p $WORKDIR

# get the GHDL source code
cd $WORKDIR
git clone https://github.com/ghdl/ghdl.git

# get the GCC source code
wget http://mirror.koddos.net/gcc/snapshots/${GCC_VERSION}/gcc-${GCC_VERSION}.tar.xz
tar -xf gcc-${GCC_VERSION}.tar.xz

# download the prerequisites
cd gcc-${GCC_VERSION}
./contrib/download_prerequisites

# create the build directory
cd $WORKDIR
mkdir gcc-ghdl-build
cd gcc-ghdl-build

# configure GHDL
$WORKDIR/ghdl/configure --with-gcc=$WORKDIR/gcc-${GCC_VERSION} --prefix=${INSTALLDIR} --default-pic
make copy-sources

# configure GCC and build
mkdir gcc-objs; cd gcc-objs
$WORKDIR/gcc-${GCC_VERSION}/configure                \
    --prefix=${INSTALLDIR}                           \
    --enable-languages=c,vhdl                        \
    --disable-bootstrap                              \
    --disable-lto                                    \
    --disable-multilib                               \
    --disable-libssp                                 \
    --disable-libgomp                                \
    --disable-libquadmath                            \
    --enable-default-pie
make -j2 && make install
#cd path/to/ghdl/source/dir/build
cd $WORKDIR/gcc-ghdl-build
make ghdllib
make install
```

Could someone please tell me what I did wrong to get the performance results I am getting?
Did I configure GHDL or GCC wrong or something? Other than '--default-pic' in the GHDL config and '--enable-default-pie' in the GCC config, the parameters are the same as in the example in the documentation.

1138-4EB, 2019-11-13:
Hi @janslab! It is possible that you did nothing wrong. GHDL is currently single-threaded and not heavily optimized. From this point of view, the execution time with mcode vs Questasim seems good. However, I am also curious about the difference between mcode and GCC. I think we need some more info about the setup to better understand what's going on.

- Your design. Even if you cannot share the sources, is it a large design (both in terms of the size of the sources and the logic that is described)?
- The commands. Did you measure simulation time only? Did you include analysis and elaboration time? Did you include startup time for QuestaSim? Do you get the same behaviour without `--wave`?
- The simulator config. Is the same time resolution being used (see #672)?

---

I run the following test:

```
git clone https://github.com/VUnit/vunit
cd vunit/examples/vhdl/array_axis_vcs

# manually C&P the content of src/test/data/in.csv until it has 2470 lines

docker run --rm -v /$(pwd)://src -w //src ghdl/vunit:mcode bash -c "python3 run.py -v"
...
Total time was 36.7 seconds
Elapsed time was 36.8 seconds
...
rm -rf vunit_out
docker run --rm -v /$(pwd)://src -w //src ghdl/vunit:mcode bash -c "python3 run.py -v"
...
Total time was 38.7 seconds
Elapsed time was 38.8 seconds
...

docker run --rm -v /$(pwd)://src -w //src ghdl/vunit:llvm bash -c "python3 run.py -v"
...
Total time was 18.7 seconds
Elapsed time was 18.7 seconds
...
rm -rf vunit_out
docker run --rm -v /$(pwd)://src -w //src ghdl/vunit:llvm bash -c "python3 run.py -v"
...
Total time was 18.7 seconds
Elapsed time was 18.7 seconds
...

docker run --rm -v /$(pwd)://src -w //src ghdl/vunit:gcc bash -c "python3 run.py -v"
...
Total time was 35.4 seconds
Elapsed time was 35.4 seconds
...
rm -rf vunit_out
docker run --rm -v /$(pwd)://src -w //src ghdl/vunit:gcc bash -c "python3 run.py -v"
...
Total time was 36.0 seconds
Elapsed time was 36.0 seconds
...
```

tgingold, 2019-11-23:
Feel free to re-open if you are interested in investigating.

closed 2019-11-23 by tgingold