ghdl/ghdl #1023: Synthetisation of barrel_shifter fails

troubas, 2019-11-19:
First of all thank you for providing us with a free software solution to simulate and maybe soon even synthesize VHDL code! I hope more and more universities will start using this free alternative to the proprietary solutions available.

I have, as an exercise during my studies, implemented a simple barrel shifter.

```

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1183-g22775978) [Dunoon edition]
Compiled with GNAT Version: 9.2.0
Target: x86_64-pc-linux-gnu
/vhdl/bte5023_combinational_logic_students/work/
Command line:
/ghdl/build/ghdl_mcode --synth barrel_shifter
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-environment.adb:297
Call stack traceback locations:
0x55e517772ce2 0x55e5175845a7 0x55e517665b39 0x55e517665d61 0x55e51766555e 0x55e5176659e1 0x55e517665bf2 0x55e517665d61 0x55e51765ad24 0x55e51765ae41 0x55e517656f59 0x55e5176a5b2d 0x55e5175fd19a 0x55e5177585a1 0x55e5173cbc94 0x7f2b2ca17151 0x55e5173ca55c 0xfffffffffffffffe
******************************************************************
```
The simulation works as expected. The source code I'm operating on:
``` vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.math_real.all;

-------------------------------------------------------------------------------

entity barrel_shifter is

  generic (
    NBITS : positive := 8);

  port (
    din, d : in  std_logic_vector(NBITS-1 downto 0);  -- data in, shift distance
    dout   : out std_logic_vector(NBITS-1 downto 0));  -- data out

end entity barrel_shifter;

-------------------------------------------------------------------------------

architecture dfl of barrel_shifter is
  -- TODO: Calculate the number of required shift stages as a constant.
  constant nshift : natural := natural(log2(real(NBITS)));

  -- custom vector for the shift stages
  type my_vec is array(0 to nshift) of std_logic_vector(NBITS-1 downto 0);
  signal vector : my_vec;
  -- vector of zeros
  constant ZEROS : std_logic_vector(NBITS-1 downto 0) := (others => '0');

begin  -- architecture dfl
    vector(0) <= din;

    gen: for i in 0 to nshift-1 generate
        vector(i+1) <= (vector(i)(NBITS-1-2**i downto 0)
                       & ZEROS(2**i-1 downto 0))
                   when d(i) = '1'
                   else vector(i);
    end generate gen;
    dout <= vector(nshift);

end architecture dfl;
```

tgingold, 2019-11-20:
I will have a look, but please be patient as I have several issues to fix.

closed 2019-11-23 by tgingold

troubas, 2019-11-23:
Confirmed. Thank you!

commit dd34ca9d (2019-11-23): synth-expr: add strip_const for indexed names.  Fix #1023