ghdl/ghdl #1025: synth: crash on procedures with discrete subtype ports

Xiretza, 2019-11-20:
**Description**
Procedures with an `inout` integer subtype port cause ghdl to crash during synthesis.

**Expected behaviour**
ghdl synthesizes the design.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
	port (
		clk : in std_logic;
		counter : out natural
	);
end entity;

architecture a of ent is
	procedure incr(signal i : inout natural) is
	begin
		i <= i + 1;
	end procedure;
begin
	process(clk)
	begin
		if rising_edge(clk) then
			-- works:
			--counter <= counter + 1;
			incr(counter);
		end if;
	end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=08 ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 7b3afe30

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1265-g7b3afe30) [Dunoon edition]
Compiled with GNAT Version: 9.2.0
Target: x86_64-pc-linux-gnu
/tmp/
Command line:
ghdl --synth --std=08 ent.vhd -e ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-expr.adb:858
Call stack traceback locations:
0x5586f745867a 0x5586f744dc1c 0x5586f745c240 0x5586f744085e 0x5586f7443a08 0x5586f744a51d 0x5586f744880f 0x5586f7448e82 0x5586f744a60c 0x5586f7443fd0 0x5586f744a4fe 0x5586f744ab93 0x5586f744c588 0x5586f744c737 0x5586f7440604 0x5586f7440721 0x5586f743b0df 0x5586f7490f4f 0x5586f73e1312 0x5586f7543ab7 0x5586f71a9d0e 0x7f3771d4e151 0x5586f71a855c 0xfffffffffffffffe
******************************************************************
```

tgingold, 2019-11-23:
Inout + signals as subprogram ports is not yet well supported...

tgingold, 2019-11-23:
I have a first fix for the crash.

tgingold, 2019-11-23:
Now fixed.  Thank you for the issue.

closed 2019-11-23 by tgingold