ghdl/ghdl #1026: can't build gcc-based ghdl

sharkcz, 2019-11-20:
Commit 19c87d08ad5cb06c8f537bbe2306508eebe0978d fixed one problem, but the build still fals. Now with
````
gnatmake: "ghdl1" up to date.
gnatmake -o ghdl -aI../../gcc/vhdl -aI../../gcc/vhdl/ghdldrv \
 -aIvhdl -aOvhdl ghdl_gcc \
 -bargs -E -cargs  -gnata -gnaty3befhkmr -gnatwa -largs libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a 
gnatbind -aI../../gcc/vhdl -aI../../gcc/vhdl/ghdldrv -aIvhdl -aOvhdl -E -x ghdl_gcc.ali
gnatlink ghdl_gcc.ali -o ghdl libcommon.a ../libcpp/libcpp.a ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a
/usr/bin/ld: ./grt-files_operations.o: in function `grt__files_operations__file_open':
grt-files_operations.adb:(.text+0xa19): undefined reference to `__ghdl_get_stdin'
/usr/bin/ld: grt-files_operations.adb:(.text+0xa51): undefined reference to `__ghdl_get_stdout'
collect2: error: ld returned 1 exit status
gnatlink: error when calling /usr/bin/gcc
gnatmake: *** link failed.
make[2]: *** [../../gcc/vhdl/Make-lang.in:115: ghdl] Error 4
make[2]: Leaving directory '/builddir/build/BUILD/gcc-9.2.1-20190827/obj-x86_64-fedora_ghdl-linux/gcc'
make[1]: *** [Makefile:4300: all-gcc] Error 2
make[1]: Leaving directory '/builddir/build/BUILD/gcc-9.2.1-20190827/obj-x86_64-fedora_ghdl-linux'
make: *** [Makefile:937: all] Error 2
````
Which is due `grt-cstdio.c` not included for the gcc build, but I can't figure out where it's missing.

````
diff --git a/Makefile.in b/Makefile.in
index a49d07dc..50b51c4c 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -235,6 +235,7 @@ copy-sources.gcc: version.ads
        $(CP) -p $(srcdir)/src/grt/grt-stdio.ad? $(gcc_vhdl_dir)
        $(CP) -p $(srcdir)/src/grt/grt-table.ad? $(gcc_vhdl_dir)
        $(CP) -p $(srcdir)/src/grt/grt-files_operations.ad? $(gcc_vhdl_dir)
+       $(CP) -p $(srcdir)/src/grt/grt-cstdio.c $(gcc_vhdl_dir)
        $(CP) -p $(srcdir)/src/ortho/*.ad? $(gcc_vhdl_dir)
        $(CP) -p $(srcdir)/src/ortho/gcc/*.ad? $(gcc_vhdl_dir)
        $(CP) -p $(srcdir)/src/ortho/gcc/*.c $(gcc_vhdl_dir)
````
^^^ clearly isn't enough.

sharkcz, 2019-11-20:
for a complete build log see https://koji.fedoraproject.org/koji/taskinfo?taskID=39106309

tgingold, 2019-11-21:
Now fixed.  Thanks!

closed 2019-11-21 by tgingold

1138-4EB, 2019-11-21:
In #1016, we were getting the same error and now we get https://github.com/ghdl/ghdl/pull/1016/checks?check_run_id=313037847#step:3:8552

tgingold, 2019-11-21:
Almost there.  We need to deal with -no-pie.

tgingold, 2019-11-21:
In the log, we don't see the 'configure' line for gcc.

1138-4EB, 2019-11-21:
No, we don't: https://github.com/1138-4EB/ghdl/blob/actions-push/dist/ci-run.sh#L276. I'll update it.

1138-4EB, 2019-11-21:
Done: https://github.com/ghdl/ghdl/runs/313079643#step:3:172

tgingold, 2019-11-21:
Thanks.  I test a fix.

sharkcz, 2019-11-21:
All builds now, thanks.

1138-4EB, 2019-11-21:
They do! #1016