ghdl/ghdl #1027: Not able to compile PSL verification unit file and bind to module.

steamka90, 2019-11-20:
**Description**
Not able to compile PSL verification unit file and bind to module. 

**How to reproduce?**
I was trying to compile `dff.vhd` and `dff_psl.vhd` file using VHDL-2008 and VHDL-93 mode.

```vhd :file: dff.vhd
library ieee;                                                                                                                                                                                                      
use ieee.std_logic_1164.all;                                                                                                                                                                                       
                                                                                                                                                                                                                   
entity dff is                                                                                                                                                                                                      
  port(                                                                                                                                                                                                            
    reset : in std_logic;                                                                                                                                                                                          
    clk : in std_logic;                                                                                                                                                                                            
    d : in std_logic;                                                                                                                                                                                              
    q : out std_logic                                                                                                                                                                                              
  );                                                                                                                                                                                                               
end entity dff;                                                                                                                                                                                                    
                                                                                                                                                                                                                   
architecture rtl of dff is                                                                                                                                                                                         
  signal q_int : std_logic;                                                                                                                                                                                        
begin                                                                                                                                                                                                              
                                                                                                                                                                                                                   
  dff_proc : process(clk, reset)                                                                                                                                                                                   
  begin                                                                                                                                                                                                            
    if reset = '1' then                                                                                                                                                                                            
      q_int <= '0';                                                                                                                                                                                                
    elsif rising_edge(clk) then                                                                                                                                                                                    
      q_int <= d;                                                                                                                                                                                                  
    end if;                                                                                                                                                                                                        
  end process dff_proc;                                                                                                                                                                                            
                                                                                                                                                                                                                   
  -- drive q_int to output port                                                                                                                                                                                    
  q <= q_int;                                                                                                                                                                                                      
                                                                                                                                                                                                                   
end rtl;  
```

```vhd :file: dff_psl.vhd
vunit dff_psl (dff(rtl))                                                                                                                                                                                           
{                                                                                                                                                                                                                  
  default clock is rising_edge(clk);                                                                                                                                                                               
  async_reset_check:                                                                                                                                                                                               
    assert always {reset = '1'} |-> {not q_int}                                                                                                                                                                    
      report "Async reset activated, q_int /= '0' during reset";                                                                                                                                                   
}   
```
Using VHDL-2008 complication option:
```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 dff.vhd dff_psl.vhd
```

Using VHDL-1993 complication option:
```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=93 dff.vhd dff_psl.vhd
```

**Context**
Please, provide the following information:

- OS: Manjaro Linux 18.1.3
- Origin:
```
[slee@manjaro-1804-pc vhdl]$ ghdl -v
GHDL 0.37-dev (v0.36-1262-gb57c47f3) [Dunoon edition]
 Compiled with GNAT Version: 9.2.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
[... 16 lines cut ...]
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o dff_psl.o dff_psl.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x55f55c853b80 0x55f55c924cdd 0x55f55c9f0426 0x55f55c9f1ded 0x55f55c9f3779 0x55f55c7aa34c 0x7f71d5120151 0x55f55c7a94cc 0xfffffffffffffffe
[... cut ...]
dff_psl.vhd:4:3:error: missing entity, architecture, package or configuration
dff_psl.vhd:5:19:error: '{' is an invalid character, replaced by '('
dff_psl.vhd:5:31:error: '}' is an invalid character, replaced by ')'
dff_psl.vhd:5:37:error: '{' is an invalid character, replaced by '('
dff_psl.vhd:5:47:error: '}' is an invalid character, replaced by ')'
dff_psl.vhd:7:1:error: '}' is an invalid character, replaced by ')'
dff_psl.vhd:7:1:error: missing entity, architecture, package or configuration
ghdl:error: compilation error
```

tgingold, 2019-11-20:
vunits are not supported before vhdl08, hence the errors when using --std=93.

vunits aren't yet supported outside synthesis.  It's on my todo list, but not on the top...

tmeissner, 2019-11-20:
> vunits aren't yet supported outside synthesis.

Oh, in that case I will add a hint to the PSL documentation. The current PSL doc could be  the cause that people try to use vunits out of synthesis context.

tgingold, 2019-11-20:
That's also an hint it's expected!

steamka90, 2019-11-21:
Ok, no problem. Yes, as @tmeissner mentioned there that I actually followed the doc to try out the PSL feature. That's why I was a bit puzzled whether it's supported yet in any way.

tmeissner, 2020-07-31:
I've updated the documentation with information about that vunits are only supported with VHDL-08 & synthesis. See commit 5a5419c.

tgingold, 2023-04-17:
I am working on the infrastructure to make that possible!

sschmitz86, 2023-04-18:
that is great to hear, thanks for the effort so far!