ghdl/ghdl #1028: error: simple uses VFP register arguments, e~simple.o does not

OpaStefanVogel, 2019-11-20:
**Description**
ghdl -m `error: simple uses VFP register arguments, e~simple.o does not`

**How to reproduce?**
[https://www.raspberrypi.org/products/raspberry-pi-4-model-b/](url)
[https://www.raspberrypi.org/downloads/raspbian/](url)

```
pi@raspberry:~/Desktop/GHDL  $ sudo apt-get install ghdl
pi@raspberry:~/Desktop/GHDL  $ wget https://www.fpgarelated.com/blockimages/MartinStrubel/ghdl/simple.vhdl
pi@raspberry:~/Desktop/GHDL  $ ghdl -i simple.vhdl
pi@raspberry:~/Desktop/GHDL  $ ghdl -m simple
analyze simple.vhdl
elaborate simple
/usr/bin/ld: error: simple uses VFP register arguments, e~simple.o does not
/usr/bin/ld: failed to merge target specific data of file e~simple.o
/usr/bin/ld: error: simple uses VFP register arguments, /usr/lib/ghdl/gcc/vhdl/std/v93/std_standard.o does not
...
```

**Additional context**
installing latest version does not work too:
```
pi@raspberry:~/Desktop/GHDL  $ git clone https://github.com/ghdl/ghdl.git
pi@raspberry:~/Desktop/GHDL  $ cd ghdl
pi@raspberry:~/Desktop/GHDL/ghdl  $ ./configure --prefix=~/Desktop/Test
pi@raspberry:~/Desktop/GHDL/ghdl  $ make install
mkdir -p ~/Desktop/Test
mkdir -p ~/Desktop/Test/bin
mkdir -p ~/Desktop/Test/lib
mkdir -p ~/Desktop/Test/lib/ghdl
mkdir -p ~/Desktop/Test/include
gcc -c -g -DWITH_GNAT_RUN_TIME -o jumps.o /home/pi/Desktop/GHDL/ghdl/./src/grt/config/jumps.c
gcc -c -g -DWITH_GNAT_RUN_TIME -o times.o /home/pi/Desktop/GHDL/ghdl/./src/grt/config/times.c
gcc -c -g -DWITH_GNAT_RUN_TIME -o grt-cstdio.o /home/pi/Desktop/GHDL/ghdl/./src/grt/grt-cstdio.c
gcc -c -g -DWITH_GNAT_RUN_TIME -o grt-cgnatrts.o /home/pi/Desktop/GHDL/ghdl/./src/grt/grt-cgnatrts.c
gcc -c -g -DWITH_GNAT_RUN_TIME -o grt-cvpi.o /home/pi/Desktop/GHDL/ghdl/./src/grt/grt-cvpi.c
gcc -c -g -DWITH_GNAT_RUN_TIME -o grt-cdynload.o /home/pi/Desktop/GHDL/ghdl/./src/grt/grt-cdynload.c
gcc -c -g -DWITH_GNAT_RUN_TIME -o fstapi.o /home/pi/Desktop/GHDL/ghdl/./src/grt/fst/fstapi.c -I/home/pi/Desktop/GHDL/ghdl/./src/grt/fst
gcc -c -g -DWITH_GNAT_RUN_TIME -o lz4.o /home/pi/Desktop/GHDL/ghdl/./src/grt/fst/lz4.c
gcc -c -g -DWITH_GNAT_RUN_TIME -o fastlz.o /home/pi/Desktop/GHDL/ghdl/./src/grt/fst/fastlz.c
echo "with Grt.Backtraces.Jit;" > grt-backtraces-impl.ads
echo "package Grt.Backtraces.Impl renames Grt.Backtraces.Jit;" >> grt-backtraces-impl.ads
gcc -c -g -o memsegs_c.o src/ortho/mcode/memsegs_c.c
gcc -c -g -DWITH_GNAT_RUN_TIME -o chkstk.o /home/pi/Desktop/GHDL/ghdl/./src/grt/config/chkstk.S
chkstk.S: Assembler messages:
chkstk.S:37: Error: unrecognized symbol type ""
chkstk.S:41: Error: bad instruction `testl %eax,%eax'
chkstk.S:42: Error: bad instruction `je 0f'
chkstk.S:43: Error: bad instruction `subl $4,%eax'
chkstk.S:44: Error: bad instruction `subl %eax,%esp'
chkstk.S:45: Error: bad instruction `jmp *(%esp,%eax)'
chkstk.S:47: Error: bad instruction `ret'
make: *** [src/grt/Makefile.inc:200: chkstk.o] Fehler 1
```

1138-4EB, 2019-11-20:
@OpaStefanVogel, I think that the URL/file is not public. You can just drag and drop it to your message to have it attached.

1138-4EB, 2019-11-20:
> installing latest version does not work too:

`mcode` backend is for x86_64/i386 only. On RPi, you need to use either LLVM or GCC backends. Current `master` is broken for GCC. Hence, today, you need to use LLVM. Should you have docker installed on raspbian, image `aptman/dbhi:bionic` (https://hub.docker.com/layers/aptman/dbhi/bionic/images/sha256-e703ebe9a76a79b4f0cefdaf1b0d9852b8d1888043cb95738e5ee06173d2c490) contains up-to-date GHDL. It's a multiarch manifest, so it works on arm, arm64 or amd64.

OpaStefanVogel, 2019-11-20:
On arm64 [(https://github.com/sakaki-/raspbian-nspawn-64)](https://github.com/sakaki-/raspbian-nspawn-64) it works in the 64-bit Terminal.

But why not in arm? I haven't installed docker, but I will test it.

1138-4EB, 2019-11-20:
> On arm64 [(https://github.com/sakaki-/raspbian-nspawn-64)](https://github.com/sakaki-/raspbian-nspawn-64) it works in the 64-bit Terminal.

What does work? Building GHDL? Running the MWE?

Anyway, please provide the output of `ghdl --version`, since that allows us to know which backend is being used.

OpaStefanVogel, 2019-11-21:
Sorry (I thought, `sudo apt-get install ghdl` is enough):
```
ghdl --version
GHDL 0.36-dev (Raspbian 0.35+git20181129+dfsg-3) [Dunoon edition]
 Compiled with GNAT Version: 8.2.0
 GCC back-end code generator
...
```
Sorry again, on arm64 [(https://github.com/sakaki-/raspbian-nspawn-64)](https://github.com/sakaki-/raspbian-nspawn-64) `sudo apt-get install ghdl` does not work at the moment (ghdl not found, I don't know why???). If MWE = mimimal working example, then yes, I had tested `ghdl -i simple.vhdl`  `ghdl -m simple` `./simple` and another examples too, they all works. Building GHDL would have an error, but i don't recall error `unrecognized symbol type ""`. At the moment I can't reproduce it.

OpaStefanVogel, 2019-11-21:
Sorry 3, the correct URL/file is `wget https://www.fpgarelated.com/blogimages/MartinStrubel/ghdl/simple.vhdl` with following content

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all; -- Unsigned

entity simple is end simple;
architecture behaviour of simple is
        signal clk : std_logic := '0';
        signal sigterm : std_logic := '0';
        signal counter : unsigned(7 downto 0) := x"00";
begin
        process
        begin
                wait for 5 us;
                clkloop : loop
                        wait for 1 us;
                        clk <= not clk;
                        if sigterm = '1' then
                                exit;
                        end if;
                end loop clkloop;
                wait for 5 us;
                wait;
        end process;

        process (clk)
        begin
                if rising_edge(clk) then
                        if counter = 16 then
                                sigterm <= '1';
                        end if;
                        counter <= counter + 1;
                end if;
        end process;
end behaviour;
```

I have installed docker and `aptman/dbhi:bionic`

```
root@7eefd06d72a2:~# ghdl --version
GHDL 0.37-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 7.4.0
 llvm code generator
```
and now I get

```
root@7eefd06d72a2:~# ghdl -i simple.vhdl 
simple.vhdl:1:1:error: missing entity, architecture, package or configuration
ghdl:error: importation has failed due to compilation error
root@7eefd06d72a2:~# 
```

OpaStefanVogel, 2019-11-21:
I have created file simple2.vhdl
```
library ieee;
use ieee.std_logic_1164.all;

entity simple2 is end simple2;
architecture behaviour of simple2 is
        signal clk : std_logic := '0';
begin
        clk <= '1';
end behaviour;
```

and this works

```
root@1dc19290aec0:/# ghdl -i simple2.vhdl
root@1dc19290aec0:/# ghdl -m simple2
analyze simple2.vhdl
elaborate simple2
root@1dc19290aec0:/# ./simple2
root@1dc19290aec0:/# 
```

1138-4EB, 2019-11-21:
I think that we are getting multiple issues mixed here:

The permissions on the URL are correct now, and I could successfully download the file. I could test that latest GHDL can successfully analyse (`-a`), import (`-i`) or elaborate (`-e`) it. I tested it with docker images `ghdl/vunit:llvm` and `aptman/dbhi:bionic` on amd64. Therefore, this is likely to be an installation issue. I suggest that you change the title of this issue to 'Build/installation on RaspberryPi 4', since that's what we actually need to focus on.

---

I assume that this error:

`simple uses VFP register arguments, /usr/lib/ghdl/gcc/vhdl/std/v93/std_standard.o does not`

was produced by this version of ghdl:

```
ghdl --version
GHDL 0.36-dev (Raspbian 0.35+git20181129+dfsg-3) [Dunoon edition]
 Compiled with GNAT Version: 8.2.0
 GCC back-end code generator
```

which you installed on Raspbian through `sudo apt-get install ghdl`.

This might be a legit bug. `std_standard.o` and `e~simple.o` are both created by GCC backend. So, there should be no imcompatibilities between them. However, I think that no one has reported having used ghdl with GCC backend on ARM before. Please:

- Open a new issue about 'GCC backend support on ARM(64)'.
- Remove temporal files from your workdir (i.e. `e~simple.o` and so), and try again.
- That version is relatively old (a year). Build GHDL with GCC backend from master.

Note that the 32/64 thing is quite tricky on RPi3 and RPi4. The hardware supports aarch32 and aarch64, but Raspbian itself is for 32-bit. However, it is possible to run either arm32v7 or arm32v8 docker images on Raspbian. Hence, you can use applications for 64-bit on a RPi4 with Raspbian, as long as you use containers.

---

> Sorry again, on arm64 (https://github.com/sakaki-/raspbian-nspawn-64) `sudo apt-get install ghdl` does not work at the moment (ghdl not found, I don't know why???).

To be honest, I am even surprised that you could `apt install ghdl` on Raspbian. I assume it is because the latest version is based on Debian Buster: https://packages.debian.org/buster/ghdl. If search after `sudo update -qq` does not work, you might want to open an issue in raspbian-nspawn-64 or download the deb file manually.

---

`simple.vhdl:1:1:error: missing entity, architecture, package or configuration`

I think that this might be because the content of `simple.vhd` is not correct. Please check that `wget` did write the expected content, and not some HTML.

---

To build GHDL on ARM, I suggest you use LLVM backend: https://ghdl.readthedocs.io/en/latest/getting/LLVM.html. You will need the following dependencies: `gcc gnat-* make zlib1g-dev clang-* llvm-*-dev`. I don't know the specific numbers that replace those `*`.

OpaStefanVogel, 2019-11-21:
> `simple.vhdl:1:1:error: missing entity, architecture, package or configuration`
> 
> I think that this might be because the content of `simple.vhd` is not correct. Please check that `wget` did write the expected content, and not some HTML.

Sorry 4, again
```
root@1dc19290aec0:/# wget https://www.fpgarelated.com/blogimages/MartinStrubel/ghdl/simple.vhdl
root@1dc19290aec0:/# ghdl -i simple.vhdl
root@1dc19290aec0:/# ghdl -m simple     
analyze simple.vhdl
elaborate simple
root@1dc19290aec0:/# ./simple --disp-time
Now is 5us +0
Now is 6us +0
...
root@1dc19290aec0:/# 
```
correct.

OpaStefanVogel, 2019-11-21:
> > Sorry again, on arm64 (https://github.com/sakaki-/raspbian-nspawn-64) `sudo apt-get install ghdl` does not work at the moment (ghdl not found, I don't know why???).

I had raspbian-nspawn-64 new installed and then I need `apt update...` first:
```
pi@raspberry:~ $ ds64-shell 
Connected to machine debian-buster-64. ....
pi@debian-buster-64:~ $ sudo apt-get update
pi@debian-buster-64:~ $ sudo apt-get upgrade
pi@debian-buster-64:~ $ sudo apt-get install gcc
pi@debian-buster-64:~ $ sudo apt-get install ghdl
pi@debian-buster-64:~ $ cd Desktop/GHDL
pi@debian-buster-64:~/Desktop/GHDL $ ghdl -i simple.vhdl
pi@debian-buster-64:~/Desktop/GHDL $ ghdl -m simple
analyze simple.vhdl
elaborate simple
pi@debian-buster-64:~/Desktop/GHDL $ ./simple --disp-time
Now is 5us +0
Now is 6us +0
...
pi@debian-buster-64:~/Desktop/GHDL $ 
``` 
This is correct.

> To be honest, I am even surprised that you could `apt install ghdl` on Raspbian. I assume it is because the latest version is based on Debian Buster: https://packages.debian.org/buster/ghdl. 

Yes, `pi@debian-buster-64:~ $ sudo apt-get ` is connected to https://deb.debian.org/debian buster.

> To build GHDL on ARM, I suggest you use LLVM backend: https://ghdl.readthedocs.io/en/latest/getting/LLVM.html. You will need the following dependencies: `gcc gnat-* make zlib1g-dev clang-* llvm-*-dev`. I don't know the specific numbers that replace those `*`.

Now is an other error in `make`:
```
pi@raspberry:~/Desktop/GHDL/ghdl $ sudo apt get install llvm

pi@raspberry:~/Desktop/GHDL/ghdl $ whereis llvm-config
llvm-config: /usr/bin/llvm-config /usr/share/man/man1/llvm-config.1.gz

pi@raspberry:~/Desktop/GHDL/ghdl $ ./configure --with-llvm-config=/usr/bin/llvm-config  --prefix=~/Desktop/Test
Use full IEEE library
Build machine is: arm-linux-gnueabihf
Debugging is not enabled with llvm 7.0.1
Creating ghdl.gpr
Creating Makefile
for d in src/std src/std/v87 src/std/v93 src/std/v08 src/synopsys src/synopsys/v08 std/v87 ieee/v87 synopsys/v87 std/v93 ieee/v93 synopsys/v93 std/v08 ieee/v08 src/openieee src/ieee src/vital95 src/vital2000 src/mentor src/ieee/v87 src/ieee/v93 src/ieee2008 mentor/v93 synopsys/v08; do \
  mkdir -p lib/ghdl/$d; \
done
Generate ghdlsynth_maybe.ads
Generate default_paths.ads

pi@raspberry:~/Desktop/GHDL/ghdl $ make
if test -d ./.git \
   && desc=`cd .; git describe --dirty`; then \
          sub="s/[(].*[)]/($desc)/"; \
else \
  sub="s/tarball/tarball/"; \
        fi; \
sed -e "$sub" -e "s/@VER@/0.37-dev/" < src/version.in > version.tmp; \

if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f ./src/ortho/llvm4-nodebug/Makefile \
 ortho_srcdir=./src/ortho ortho_exec=ghdl1-llvm \
 GNATFLAGS="-aI./src -aI./src/vhdl -aI./src/grt -aI./src/psl -aI./src/vhdl/translate -aI./src/ghdldrv -aI./src/ortho -aI./src/ortho/llvm4-nodebug -aI./src/synth -gnaty3befhkmr -gnatwa -gnatf -g -gnata -gnatwe" LDFLAGS="" \
 LLVM_CONFIG="/usr/bin/llvm-config" CXX="clang++" GNATMAKE="gnatmake" \
 all
make[1]: Verzeichnis „/home/pi/Desktop/GHDL/ghdl“ wird betreten
rm -f -f src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--  DO NOT MODIFY - this file was generated from:" > src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--  ortho_nodes.common.ads and ortho_llvm.private.ads" \
  >> src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--" >> src/ortho/llvm4-nodebug/ortho_llvm.ads
sed -e '/^private/,$d' \
  < ./src/ortho/llvm4-nodebug/ortho_llvm.private.ads >> src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--  Start of common part" >> src/ortho/llvm4-nodebug/ortho_llvm.ads
sed -e '1,/^package/d' -e '/^private/,$d' < src/ortho/ortho_nodes.common.ads >> src/ortho/llvm4-nodebug/ortho_llvm.ads
echo "--  End of common part" >> src/ortho/llvm4-nodebug/ortho_llvm.ads
sed -n -e '/^private/,$p' \
  < ./src/ortho/llvm4-nodebug/ortho_llvm.private.ads >> src/ortho/llvm4-nodebug/ortho_llvm.ads
chmod a-w src/ortho/llvm4-nodebug/ortho_llvm.ads
clang++ -c `/usr/bin/llvm-config --cxxflags` -o llvm-cbindings.o src/ortho/llvm4-nodebug/llvm-cbindings.cpp
/bin/sh: 1: clang++: not found
make[1]: *** [src/ortho/llvm4-nodebug/Makefile:18: llvm-cbindings.o] Fehler 127
make[1]: Verzeichnis „/home/pi/Desktop/GHDL/ghdl“ wird verlassen
make: *** [Makefile:342: ghdl1-llvm] Fehler 2

pi@raspberry:~/Desktop/GHDL/ghdl $ 

```

1138-4EB, 2019-11-21:
Did you install `clang-7`?

OpaStefanVogel, 2019-11-21:
> To build GHDL on ARM, I suggest you use LLVM backend: https://ghdl.readthedocs.io/en/latest/getting/LLVM.html. You will need the following dependencies: `gcc gnat-* make zlib1g-dev clang-* llvm-*-dev`. I don't know the specific numbers that replace those `*`.

Sorry 5, the dependendcies I have not checked because missing dependendcies are diplayed? Now `./ghdl_llvm -i` is running, but error in `./ghdl_llvm -m`:
```
pi@raspberry:~/Desktop/GHDL/ghdl $ sudo apt-get install clang
...
Entpacken von clang (1:7.0-47) ...
clang (1:7.0-47) wird eingerichtet ...
...
pi@raspberry:~/Desktop/GHDL/ghdl $ make
pi@raspberry:~/Desktop/GHDL/ghdl $ make install
pi@raspberry:~/Desktop/GHDL/ghdl $ cd ..
pi@raspberry:~/Desktop/GHDL $ ./ghdl/ghdl_llvm -i simple.vhdl
pi@raspberry:~/Desktop/GHDL $ ./ghdl/ghdl_llvm -m simple
/usr/bin/ld: /home/pi/Desktop/GHDL/ghdl/lib/ghdl/libgrt.a(grt-lib.o): in function `grt__lib__do_report':
/home/pi/Desktop/GHDL/ghdl/src/grt/grt-lib.adb:55: undefined reference to `__gnat_last_chance_handler'
/usr/bin/ld: /home/pi/Desktop/GHDL/ghdl/src/grt/grt-lib.adb:57: undefined reference to `__gnat_last_chance_handler'
/usr/bin/ld: /home/pi/Desktop/GHDL/ghdl/src/grt/grt-lib.adb:59: undefined reference to `__gnat_last_chance_handler'
/usr/bin/ld: /home/pi/Desktop/GHDL/ghdl/lib/ghdl/libgrt.a(grt-lib.o): in function `__ghdl_program_error':
/home/pi/Desktop/GHDL/ghdl/src/grt/grt-lib.adb:161: undefined reference to `__gnat_last_chance_handler'
/usr/bin/ld: /home/pi/Desktop/GHDL/ghdl/lib/ghdl/libgrt.a(grt-lib.o): in function `grt__lib__hi':
...
...
...
/usr/bin/ld: /home/pi/Desktop/GHDL/ghdl/lib/ghdl/libgrt.a(grt-wave_opt-design.o): in function `grt__wave_opt__design__check$
/home/pi/Desktop/GHDL/ghdl/src/grt/grt-wave_opt-design.adb:211: undefined reference to `__gnat_last_chance_handler'
/usr/bin/ld: /home/pi/Desktop/GHDL/ghdl/src/grt/grt-wave_opt-design.adb:218: undefined reference to `__gnat_last_chance_han$
collect2: error: ld returned 1 exit status
./ghdl/ghdl_llvm: compilation error
```

OpaStefanVogel, 2019-11-21:
Now I have tested the last up to `./ghdl_llvm -m` on a complete new installed Raspbian buster (not raspian-nspawn-64 in the 32-Bit-Terminal) with no error. Thank you very much!!!

1138-4EB, 2019-11-21:
Great! I'm closing this issue, since it seems that we sorted it out. Nevertheless, feel free to continue the discussion.

closed 2019-11-21 by 1138-4EB