ghdl/ghdl #1029: GCC backend support on ARM(64)

OpaStefanVogel, 2019-11-21:
Continuation from https://github.com/ghdl/ghdl/issues/1028#issuecomment-556819444

I have used GCC backend because I'v got it with `sudo apt-get install ghdl`. But this is not necessary for me, your suggestion is LLVM. I open this issue only for completness.

tgingold, 2019-11-21:
I think the ghdl binary you got from apt-get is not correctly built.  To be more precise, the configuration of the gcc makes it use a different ABI than your system.

OpaStefanVogel, 2019-11-21:
I have compared few days ago the checksum SHA-256 of

https://packages.debian.org/buster/armel/ghdl/download
https://packages.debian.org/buster/armhf/ghdl/download

and they were equal 'as far as I can remember'. Now they are different. apt-get in my Raspbian is connected to http://raspbian.raspberrypi.org/raspbian/ buster and not to Debian. Therefore I have tried to compile the latest package with `./configure` and `make install` in https://github.com/ghdl/ghdl/blob/master/README.md section 'TL;DR', but with error `unrecognized symbol type`, Logfile in https://github.com/ghdl/ghdl/issues/1028#issue-526259033 section 'Additional context'.

sharkcz, 2019-11-21:
you are trying to build the mcode backend, which is x86-only

OpaStefanVogel, 2019-11-21:
In https://github.com/ghdl/ghdl/blob/master/README.md section 'TL;DR' I found `make` "At that place, you can already use the ghdl_mcode built in the directory. You can also install GHDL:" `make install`.

Here in https://github.com/ghdl/ghdl/issues/1028#issuecomment-556949849 I have build ghdl_llvm with `make install`. `make install` is not x86-only but armhf-included.

OpaStefanVogel, 2019-11-21:
Next attempt, in https://ghdl.readthedocs.io/en/latest/getting/GCC.html - Example I can't find the right `/path/to/gcc/source/dir`:
```
pi@raspberrypi:~/Desktop/GHDL/ghdl/build $ ../configure --with-gcc=/usr/bin --prefix=/usr/local
Use full IEEE library
Build machine is: arm-linux-gnueabihf
cannot find gcc/BASE-VER in /usr/bin
pi@raspberrypi:~/Desktop/GHDL/ghdl/build $ ../configure --with-gcc=/usr/lib --prefix=/usr/local
Use full IEEE library
Build machine is: arm-linux-gnueabihf
cannot find gcc/BASE-VER in /usr/lib
pi@raspberrypi:~/Desktop/GHDL/ghdl/build $ 

```

sharkcz, 2019-11-21:
ghdl needs gcc sources to be available

OpaStefanVogel, 2019-11-21:
```
pi@raspberry:~/Desktop/GHDL/ghdl/build $ ls
... (no content) ...
pi@raspberry:~/Desktop/GHDL/ghdl/build $ sudo nano /etc/apt/sources.list
```
--- uncomment the prepared line #deb-scr (content equal to first line deb) and save the file ---

```
pi@raspberry:~/Desktop/GHDL/ghdl/build $ sudo apt-get update
pi@raspberry:~/Desktop/GHDL/ghdl/build $ sudo apt-get source gcc
...
dpkg-source: Information: gcc-defaults wird nach gcc-defaults-1.181+rpi2 extrahiert
dpkg-source: Information: gcc-defaults_1.181+rpi2.tar.gz wird entpackt
...
pi@raspberry:~/Desktop/GHDL/ghdl/build $ ls
gcc-defaults-1.181+rpi2      gcc-defaults_1.181+rpi2.tar.gz
gcc-defaults_1.181+rpi2.dsc
pi@raspberry:~/Desktop/GHDL/ghdl/build $ ../configure --with-gcc=gcc-defaults-1.181+rpi2 --prefix=/usr/local
Use full IEEE library
Build machine is: arm-linux-gnueabihf
cannot find gcc/BASE-VER in gcc-defaults-1.181+rpi2
pi@raspberry:~/Desktop/GHDL/ghdl/build $ ../configure --with-gcc=gcc-defaults-1.181+rpi2/debian --prefix=/usr/local
Use full IEEE library
Build machine is: arm-linux-gnueabihf
cannot find gcc/BASE-VER in gcc-defaults-1.181+rpi2/debian
pi@raspberry:~/Desktop/GHDL/ghdl/build $ 
```

Is `~/Desktop/GHDL/ghdl/build/gcc-defaults-1.181+rpi2` the right source file? It doesn't contain `gcc/BASE-VER`.

Another attempt for example [https://github.com/gcc-mirror/gcc/tree/master](https://github.com/gcc-mirror/gcc/tree/master) doesn't contain `gcc/BASE-VER`.

1138-4EB, 2019-11-21:
@OpaStefanVogel, the 'TL;DR' section is to be used as a fast recipe, in case it works. If it fails, users are expected to read the README properly, and the Building section of the docs too. As you say `make install` is used for any backend and platform. That's not the relevant command, tho. You need to pay attention to how you `configure` the build. The difference between building with mcode or LLVM backends is precisely the configuration.

Hence, in https://github.com/ghdl/ghdl/issues/1029#issuecomment-556904528 you did neither use `--with-llvm-config` nor `--with-gcc`. As a result, you were building `mcode` backend, which is not supported on ARM. That's what @sharkcz pointed out.

> Another attempt for example [gcc-mirror/gcc@`master`](https://github.com/gcc-mirror/gcc/tree/master) [](/gcc-mirror/gcc/tree/HEAD@{2019-11-21T14:41:08Z}) doesn't contain `gcc/BASE-VER`.

That's the source we use for CI:

https://github.com/ghdl/ghdl/blob/03862a4607fd127e6570a3e141a92265a23c2a68/dist/travis/build.sh#L76-L92

I think that you might have checked some incorrect branch/tag, because BASE-VER exists in `master`: https://github.com/gcc-mirror/gcc/blob/master/gcc/BASE-VER

OpaStefanVogel, 2019-11-21:
> @OpaStefanVogel, the 'TL;DR' section is to be used as a fast recipe, in case it works. If it fails, users are expected to read the README properly, and the Building section of the docs too. 

This I didn't know. @sharkcz Thank You for Your answers.

> I think that you might have checked some incorrect branch/tag, because BASE-VER exists in `master`: https://github.com/gcc-mirror/gcc/blob/master/gcc/BASE-VER

No, I have expected and only seached BASE-VER as a directory. Mistake.

Now I will try to find the proper version of the gcc source file...

1138-4EB, 2019-11-21:
> Now I will try to find the proper version of the gcc source file...with gcc/BASE-VER...

Instead of trying to do it manually, you can try the scripts in `dist`. For example:

`./dist/build.sh -b gcc-9.1.0 -c --pkg mypkg.tgz`

The result should be a tarball (`mypkg.tgz`) that you can extract/install. Alternatively, `build-*` and `install-*` folders will contain temporal build files. Referring to the README, *"At that place, you can already use the ghdl_mcode built in the directory"*; but it's `ghdl_llvm`, not ghdl_mcode, in this case.

As a matter of fact, that's the script that is used to build the ghdl installations available in `ghdl/*` or `aptman/dbhi:*` images, to the tarballs that are published to GitHub Releases.

OpaStefanVogel, 2019-11-21:
```
pi@Radio:~/Desktop/GHDL/ghdl/build $ cd ..
pi@Radio:~/Desktop/GHDL/ghdl $ ls dist
ansi_color.sh  gcc  macosx  travis  windows
pi@Radio:~/Desktop/GHDL/ghdl $ ls dist/travis
build.sh  man.sh  travis-ci.sh  utils.sh
```
`./dist/build.sh` does not exist, I try `./dist/travis/build.sh`:
```
pi@Radio:~/Desktop/GHDL/ghdl $ ./dist/travis/build.sh -b gcc-defaults-1.181+rpi2 -c --pkg mypkg.tgz
./dist/travis/build.sh -b gcc-defaults-1.181+rpi2 -c --pkg mypkg.tgz
SHELL=/bin/bash
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
XDG_MENU_PREFIX=lxde-pi-
...
OLDPWD=/home/pi/Desktop/GHDL/ghdl/build
TEXTDOMAIN=Linux-PAM
_=/usr/bin/env
https://github.com/gcc-mirror/gcc/archive/gcc-defaults-1_181+rpi2-release.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   149    0   149    0     0    238      0 --:--:-- --:--:-- --:--:--   238
100    15  100    15    0     0     15      0  0:00:01 --:--:--  0:00:01     0

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
```

1138-4EB, 2019-11-21:
> `./dist/build.sh` does not exist, I try `./dist/travis/build.sh`:

My bad. It's `./dist/travis/build.sh` indeed.

> ```
> pi@Radio:~/Desktop/GHDL/ghdl $ ./dist/travis/build.sh -b gcc-defaults-1.181+rpi2 -c --pkg mypkg.tgz
> ```

Whatever you attach to `gcc-` must be a branch or tag from https://github.com/gcc-mirror/gcc. `defaults-1.181+rpi2` is not a valid ref. The error you get is because it is trying to download an archive that does not exist.

OpaStefanVogel, 2019-11-21:
```
pi@raspberrypi:~/Desktop/GHDL/ghdl $ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
```
The version number of `gcc --version` hadn't to be equal to x.y.z in `./dist/build.sh -b gcc-x.y.z -c --pkg mypkg.tgz`? 

I have tested `gcc-8.3.0`, `gcc-9.1.0`, `gcc-0_2_0`, all of them ended after 20 min or so with (red)`error: gengtype-lex.c: Datei oder Verzeichnis nicht gefunden`

```
pi@raspberrypi:~/Desktop/GHDL/ghdl $ ./dist/travis/build.sh -b gcc-8.3.0 -c --pkg mypkg.tgz

... 20 min ...

make[2]: [Makefile:2876: gengtype-lex.c] Fehler 127 (ignoriert)
g++ -c   -g -O2 -DIN_GCC     -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -Wno-error -DHAVE_CONFIG_H -DGENERATOR_FILE -fno-PIE -I. -Ibuild -I../../gcc-srcs/gcc -I../../gcc-srcs/gcc/build -I../../gcc-srcs/gcc/../include  -I../../gcc-srcs/gcc/../libcpp/include  \
	-o build/gengtype-lex.o gengtype-lex.c
g++: error: gengtype-lex.c: Datei oder Verzeichnis nicht gefunden
g++: fatal error: no input files
compilation terminated.
make[2]: *** [Makefile:2670: build/gengtype-lex.o] Fehler 1
make[2]: Verzeichnis „/home/pi/Desktop/GHDL/ghdl/build-gcc-8.3.0/gcc-objs/gcc“ wird verlassen
make[1]: *** [Makefile:4343: install-gcc] Fehler 2
make[1]: Verzeichnis „/home/pi/Desktop/GHDL/ghdl/build-gcc-8.3.0/gcc-objs“ wird verlassen
make: *** [Makefile:2273: install] Fehler 2
```

1138-4EB, 2019-11-21:
I think that the error you are getting is because you are missing some dependencies. `gengtype-lex.c` is probably not found because you need to install `flex`. You will need `gcc gnat-* make zlib1g-dev autogen bzip2 curl dejagnu flex g++ lbzip2 texinfo wget`, where `*` is a number (6, 7, 8...).

Either `gcc-8.3.0` or `gcc-9.1.0` should compile. However, there might be runtime issues when using some advanced features with 9.1.0. See #883.

OpaStefanVogel, 2019-11-22:
Now `libgcc_eh.a` not found.
```
pi@raspberrypi:~/Desktop/GHDL/ghdl $ sudo apt-get install flex autogen dejagnu lbzip2 texinfo
pi@raspberrypi:~/Desktop/GHDL/ghdl $ rm -r build-gcc-8.3.0/
pi@raspberrypi:~/Desktop/GHDL/ghdl $ sudo rm -r build-gcc-8.3.0/
pi@raspberrypi:~/Desktop/GHDL/ghdl $ ./dist/travis/build.sh -b gcc-8.3.0 -c --pkg mypkg.tgz
... a lot of output, then ...
make[2]: Verzeichnis „/home/pi/Desktop/GHDL/ghdl/build-gcc-8.3.0/gcc-objs/armv7l-unknown-linux-gnueabihf/libgcc“ wird betreten
/bin/bash ../../../gcc-srcs/libgcc/../mkinstalldirs /home/pi/Desktop/GHDL/ghdl/install-gcc-8.3.0/lib/gcc/armv7l-unknown-linux-gnueabihf/8.3.0
/usr/bin/install -c -m 644 libgcc_eh.a /home/pi/Desktop/GHDL/ghdl/install-gcc-8.3.0/lib/gcc/armv7l-unknown-linux-gnueabihf/8.3.0/
/usr/bin/install: der Aufruf von stat für 'libgcc_eh.a' ist nicht möglich: Datei oder Verzeichnis nicht gefunden
make[2]: *** [Makefile:1152: install-shared] Fehler 1
make[2]: Verzeichnis „/home/pi/Desktop/GHDL/ghdl/build-gcc-8.3.0/gcc-objs/armv7l-unknown-linux-gnueabihf/libgcc“ wird verlassen
make[1]: *** [Makefile:12673: install-target-libgcc] Fehler 2
make[1]: Verzeichnis „/home/pi/Desktop/GHDL/ghdl/build-gcc-8.3.0/gcc-objs“ wird verlassen
make: *** [Makefile:2273: install] Fehler 2
```
`libgcc_eh.a` is contained in https://packages.debian.org/buster/armhf/libgcc-8-dev/filelist and `libgcc-8-dev` is installed:
```
pi@raspberrypi:~/Desktop/GHDL/ghdl $ dpkg -l libgcc-8-dev
Gewünscht=Unbekannt/Installieren/R=Entfernen/P=Vollständig Löschen/Halten
| Status=Nicht/Installiert/Config/U=Entpackt/halb konFiguriert/
         Halb installiert/Trigger erWartet/Trigger anhängig
|/ Fehler?=(kein)/R=Neuinstallation notwendig (Status, Fehler: GROSS=schlecht)
||/ Name               Version      Architektur  Beschreibung
+++-==================-============-============-=======================================
ii  libgcc-8-dev:armhf 8.3.0-6+rpi1 armhf        GCC support library (development files)
pi@raspberrypi:~/Desktop/GHDL/ghdl $ 
```
I have no idea how to solve such a dependency.

tgingold, 2019-11-22:
libgcc_eh.a should have been generated during the build.  That's not a problem of dependency but a build problem.

Maybe --disable-shared is not supported on armv7

I would recommend to build with llvm backend (and not gcc); that's way simpler to build.

OpaStefanVogel, 2019-11-22:
Just finished the version raspbian-nspawn-64 Terminal 64-Bit SUCCESSFULL!
```
pi@debian-buster-64:~/Desktop/GHDL/ghdl $ ./dist/travis/build.sh -b gcc-8.3.0 -c --pkg mypkg.tgz
...
./lib/ghdl/synopsys/v93/math_real-body.o
./lib/ghdl/synopsys/v93/std_logic_signed.o
./lib/libghdl-0_37_dev.so
./lib/libcc1.so
[SUCCESSFUL]
pi@debian-buster-64:~/Desktop/GHDL/ghdl $ ls
... 
ghdlmain.ali                   mypkg.tgz.tgz                    trans-chap12.ali     vhdl-ieee-std_logic_arith.ali     vpi_thunk.o
...
```
Maybe I compare the outputs of 32-Bit and 64-bit `./dist/travis/build.sh` for further search, but llvm backend is ok for me. Thank you very much.

closed 2019-11-23 by tgingold