ghdl/ghdl #1030: Crash with some fmf SPI FLASH models.

donnie-j, 2019-11-22:
**Description**
When using newer SPI FLASH models from the Free Model Foundation, illegal instruction is triggered in GHDL

**How to reproduce?**
Using the attached file, untar and run ./build-run.sh

```
[flash_mem_crash.tar.gz](https://github.com/ghdl/ghdl/files/3881541/flash_mem_crash.tar.gz)
[flash_mem_crash.tar.gz](https://github.com/ghdl/ghdl/files/3881544/flash_mem_crash.tar.gz)
[flash_mem_crash.tar.gz](https://github.com/ghdl/ghdl/files/3881545/flash_mem_crash.tar.gz)
[rom_contents.txt](https://github.com/ghdl/ghdl/files/3881547/rom_contents.txt)
ibuprofen:flash_mem_crash jeff$ ./build-run.sh
m25p80.vhd:204:35:warning: declaration of "se" hides concurrent procedure call [-Whide]
m25p80.vhd:205:35:warning: declaration of "be" hides concurrent procedure call [-Whide]
m25p80.vhd:206:35:warning: declaration of "pp" hides concurrent procedure call [-Whide]
m25p80.vhd:207:35:warning: declaration of "dp" hides concurrent procedure call [-Whide]
s25fs128s.vhd:322:13:warning: declaration of "sck" hides port "sck" [-Whide]
s25fs128s.vhd:323:13:warning: declaration of "csneg" hides port "csneg" [-Whide]
s25fs128s.vhd:322:13:warning: declaration of "sck" hides port "sck" [-Whide]
s25fs128s.vhd:323:13:warning: declaration of "csneg" hides port "csneg" [-Whide]
s25fs128s.vhd:373:36:warning: declaration of "wrr" hides concurrent procedure call [-Whide]
s25fs128s.vhd:392:36:warning: declaration of "be" hides concurrent procedure call [-Whide]
s25fs128s.vhd:413:36:warning: declaration of "ees" hides concurrent procedure call [-Whide]
s25fs128s.vhd:415:36:warning: declaration of "se4" hides concurrent procedure call [-Whide]
s25fs128s.vhd:434:36:warning: declaration of "dpd" hides concurrent procedure call [-Whide]
s25fs128s.vhd:695:16:warning: declaration of "esusp" hides concurrent procedure call [-Whide]
s25fs128s.vhd:790:33:warning: declaration of "bottomboot" hides signal "bottomboot" [-Whide]
s25fs128s.vhd:791:33:warning: declaration of "topboot" hides signal "topboot" [-Whide]
s25fs128s.vhd:820:33:warning: declaration of "bottomboot" hides signal "bottomboot" [-Whide]
s25fs128s.vhd:821:33:warning: declaration of "topboot" hides signal "topboot" [-Whide]
s25fs128s.vhd:3562:24:warning: declaration of "addr" hides variable "addr" [-Whide]
s25fs128s.vhd:3611:24:warning: declaration of "addr" hides variable "addr" [-Whide]
s25fs128s.vhd:3612:24:warning: declaration of "block_e" hides variable "block_e" [-Whide]
s25fs128s.vhd:3660:24:warning: declaration of "addr" hides variable "addr" [-Whide]
s25fs128s.vhd:3669:24:warning: declaration of "rdar_reg" hides variable "rdar_reg" [-Whide]
s25fs128s.vhd:3670:24:warning: declaration of "addr" hides variable "addr" [-Whide]
s25fs128s.vhd:3757:30:warning: declaration of "latency_code" hides variable "latency_code" [-Whide]
s25fs128s.vhd:3757:54:warning: declaration of "dummy_cnt" hides variable "dummy_cnt" [-Whide]
./build-run.sh: line 20: 48527 Illegal instruction: 4  ghdl -r flash_tb --stop-time=31ms --wave=flash_tb.ghw
```
[flash_mem_crash.tar.gz](https://github.com/ghdl/ghdl/files/3881551/flash_mem_crash.tar.gz)

tgingold, 2019-11-23:
I can reproduce the crash.

tgingold, 2019-11-23:
The process memload needs a lot of memory, hence a crash due to a stack overflow.  I will see what could be done.

As a workaround, you could reduce the memory size.

tgingold, 2019-11-23:
The crash occurs at line 8326 of s25fs128s.vhd:
```
            Mem := (OTHERS => MaxData);
```

tmeissner, 2019-11-23:
Do they use an ordinary array for the memory? Better use a dictionary like data structure which only hold the data which is really read/write.

donnie-j, 2019-11-23:
Thanks for debugging this.  I really need to make a debugging build of GHDL so I can produce smaller bug reports.

> Do they use an ordinary array for the memory? Better use a dictionary like data structure which only hold the data which is really read/write.

I don't think any practical test bench will use the entire array.  For what we are trying to do, we just need something like a 32k byte array for program code, to verify operation of eXecute In Place from QSPI on a J-Core processor SoC... which would seem like a fairly typical thing to do with such a model.

Sparse allocation might be the best approach if I was going to change the flash model... but I'm just going to reduce the size to a smaller chip in the same family and move on for now :)

tgingold, 2019-11-23:
As a workaround, you can replace line 8330 s25fs128s.vhd like this:
```
            for i in mem'range loop
              mem (i) := maxdata;
            end loop;
            --  Mem := (OTHERS => MaxData);
```

Then the load fails because the addresses must be on 7 characters.

donnie-j, 2019-11-23:
> As a workaround, you can replace line 8330 s25fs128s.vhd like this:
> 
> ```
>             for i in mem'range loop
>               mem (i) := maxdata;
>             end loop;
>             --  Mem := (OTHERS => MaxData);
> ```
> 
> Then the load fails because the addresses must be on 7 characters.

Even then, adding leading 0s in the rom_contents.txt doesn't quite fix it, but at least the problem is detected...

```
ibuprofen:flash_mem_crash jeff$ ./build-run.sh
m25p80.vhd:204:35:warning: declaration of "se" hides concurrent procedure call [-Whide]
m25p80.vhd:205:35:warning: declaration of "be" hides concurrent procedure call [-Whide]
m25p80.vhd:206:35:warning: declaration of "pp" hides concurrent procedure call [-Whide]
m25p80.vhd:207:35:warning: declaration of "dp" hides concurrent procedure call [-Whide]
s25fs128s.vhd:322:13:warning: declaration of "sck" hides port "sck" [-Whide]
s25fs128s.vhd:323:13:warning: declaration of "csneg" hides port "csneg" [-Whide]
s25fs128s.vhd:322:13:warning: declaration of "sck" hides port "sck" [-Whide]
s25fs128s.vhd:323:13:warning: declaration of "csneg" hides port "csneg" [-Whide]
s25fs128s.vhd:373:36:warning: declaration of "wrr" hides concurrent procedure call [-Whide]
s25fs128s.vhd:392:36:warning: declaration of "be" hides concurrent procedure call [-Whide]
s25fs128s.vhd:413:36:warning: declaration of "ees" hides concurrent procedure call [-Whide]
s25fs128s.vhd:415:36:warning: declaration of "se4" hides concurrent procedure call [-Whide]
s25fs128s.vhd:434:36:warning: declaration of "dpd" hides concurrent procedure call [-Whide]
s25fs128s.vhd:695:16:warning: declaration of "esusp" hides concurrent procedure call [-Whide]
s25fs128s.vhd:790:33:warning: declaration of "bottomboot" hides signal "bottomboot" [-Whide]
s25fs128s.vhd:791:33:warning: declaration of "topboot" hides signal "topboot" [-Whide]
s25fs128s.vhd:820:33:warning: declaration of "bottomboot" hides signal "bottomboot" [-Whide]
s25fs128s.vhd:821:33:warning: declaration of "topboot" hides signal "topboot" [-Whide]
s25fs128s.vhd:3562:24:warning: declaration of "addr" hides variable "addr" [-Whide]
s25fs128s.vhd:3611:24:warning: declaration of "addr" hides variable "addr" [-Whide]
s25fs128s.vhd:3612:24:warning: declaration of "block_e" hides variable "block_e" [-Whide]
s25fs128s.vhd:3660:24:warning: declaration of "addr" hides variable "addr" [-Whide]
s25fs128s.vhd:3669:24:warning: declaration of "rdar_reg" hides variable "rdar_reg" [-Whide]
s25fs128s.vhd:3670:24:warning: declaration of "addr" hides variable "addr" [-Whide]
s25fs128s.vhd:3757:30:warning: declaration of "latency_code" hides variable "latency_code" [-Whide]
s25fs128s.vhd:3757:54:warning: declaration of "dummy_cnt" hides variable "dummy_cnt" [-Whide]
s25fs128s.vhd:3021:13:@10300ns:(report warning): *s25fs128s: Device is selected during Power Up
ghdl:error: overflow detected
ghdl:error: simulation failed
ibuprofen:flash_mem_crash Jeff$
```

tgingold, 2019-11-23:
Yes, but this occurs in around line 1658, when the value of opcode_cnt is 8 (which is too large to index opcode_in).

donnie-j, 2019-11-23:
Yes, you're right, it looks like something is buggy in the FMF's model around mode switching between regular SPI and quad mode.  I don't immediately see how they expected the command shift register to do the correct thing here (seems like it always increments the index by 1)

But this one isn't a GHDL issue... I might have to dig into this SPI flash model after all :(

ozbenh, 2020-06-10:
Ah I've been chasing the same problem.

Talking of which @donnie-j, using s25fl128s.vhd from fmf, I had to take out the various path delay processes at the end and replace them with things like:

        HOLDNegOut <= HOLDNegOut_zd;
        WPNegOut <= WPNegOut_zd;

etc..

Otherwise ghdl complains that the signals have multiple drivers and makes them all U

I'm not sure what's up here, I don't really understand the VITAL stuff :)

tgingold, 2020-06-14:
After the fix for #1361, you don't need the workaround anymore.  So I close this issue as fixed.

closed 2020-06-14 by tgingold