ghdl/ghdl #1038: uvvm_vvc_framework  compilation ends in - Report this bug --

ZirconfleX, 2019-11-27:
**Description**
- Latest version of GHDL installed on Linux-Mint_19.2(Tara).
- GHDL installed using LLVM approach. (Used LLVM version 9).
- GHDL installed in /opt/Ghdl (directory is owned by the <user>)
- Compiling UVVM libraries.

- Downloaded the libraries fro UVVM github page (UVVM version 2019.11.25).
- Downloaded zip extracted in /opt/Ghdl/lib/ghdl/UVVM 
  - a subdirectory UVVM-2019.11.25 is created containing all UVVM stuff (files and directories)
- Adapted the config.sh file with these settings:
  - InstallationDirectories[UVVM]="/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25"	
  - DestinationDirectories[UVVM]="/opt/Ghdl/lib/ghdl/UVVM"
  - SourceDirectories[UVVM]="."

In a terminal changed the working directory to: cd opt/Ghdl/lib/vendors
Run from here command: ./compile-uvvm.sh --uvvm-utilities
This end in a successful compile and a uvvm_util is created as sub-directory in /opt/Ghdl/lib/ghdl/UVVM

Run as second command:  ./compile-uvvm.sh --uvvm-vvc-framework
This ends in below BUG report, but the uvvm_vvc_framework directory is created.

**Expected behaviour**
Should expect this to finish without issues

**How to reproduce?**
Do as described above in teh description paragraph.

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-pc-linux-gnu
/opt/Ghdl/lib/ghdl/UVVM/uvvm_vvc_framework/v08/
Command line:
/opt/Ghdl/bin/ghdl1-llvm -fexplicit -frelaxed-rules --no-vital-checks --warn-binding --mb-comments --std=08 --work=uvvm_vvc_framework -P/opt/Ghdl/lib/ghdl/UVVM/ -P/opt/Ghdl/lib/ghdl/ieee/v08/ -P/opt/Ghdl/lib/ghdl/ -c -fpic -o ti_vvc_framework_support_pkg.o /opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_vvc_framework_support_pkg.vhd
Exception PROGRAM_ERROR raised
Exception information:
raised PROGRAM_ERROR : ortho_llvm.adb:827 explicit raise
Call stack traceback locations:
0x414a0a 0x5687f0 0x56b8f0 0x56c632 0x56c994 0x56e8e5 0x56ff71 0x57150e 0x5b167a 0x5b1915 0x59bedd 0x60d869 0x60ec21 0x61000e 0x4084bd 0x7feb798a5b95 0x4076d8 0xfffffffffffffffe
...
******************************************************************
```

**Additional context**
Below is the terminal output after the bug message.

/opt/Ghdl/bin/ghdl: compilation error
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd'
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd:69:16:warning: declaration of "scope" hides constant interface "scope" [-Whide]
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd:72:16:warning: declaration of "scope" hides constant interface "scope" [-Whide]
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd:550:14:warning: declaration of "scope" hides constant interface "scope" [-Whide]
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_generic_queue_pkg.vhd:574:14:warning: declaration of "scope" hides constant interface "scope" [-Whide]
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_data_queue_pkg.vhd'
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_data_queue_pkg.vhd:27:19:warning: type of a shared variable must be a protected type [-Wshared]
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_data_fifo_pkg.vhd'
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_data_stack_pkg.vhd'
Analyzing package '/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_uvvm_engine.vhd'
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_uvvm_engine.vhd:24:24: unit "ti_vvc_framework_support_pkg" not found in library "uvvm_vvc_framework"
/opt/Ghdl/lib/ghdl/UVVM/UVVM-2019.11.25/uvvm_vvc_framework/script/../src/ti_uvvm_engine.vhd:29:14: entity 'ti_uvvm_engine' was not analysed
/opt/Ghdl/bin/ghdl: compilation error
--------------------------------------------------------------------------------
Compiling UVVM packages [FAILED]

diogratia, 2019-11-28:
If you look at line 827 in ortho_llvm.adb that's the error call out in function New_Global.

If you look at the differences between UVVM  v2019.11.25 and v2019.09.02 for ti_vvc_framework_support_pkg.vhd you see added declarations for various things including procedure activity_watchdog also found in the package body which waits on signal global_trigger_testcase_inactivity_watchdog declared in the package.

With the 'global' signal showing up in the later package I wouldn't bet money this doesn't represent a ghdl bug or an incompatibility with ghdl.  The signal is only mentioned two places in the uvvm_vvc_framework directory (the other being vvc_generator.py).

These declarations also show up in v2019.11.21 which you note also fails to analyze in UVVM issue [BUG while compiling lates UVVM with GHDL \#91](https://github.com/UVVM/UVVM/issues/91).

ZirconfleX, 2019-11-29:
Thanks for this,

I also filed the BUG report with GHDL, as it is GHDL erroring out.
I wll add your remarks to the bug case with GHDL.

Many thanks,

Marc

ZirconfleX, 2019-11-29:
HI,

I saw you also reported this, with even a lot more comments, to GHDL.
Thanks for doing that!

Kind regards,

Marc

diogratia, 2019-11-29:
By definition there's a ghdl bug, an uncaught exception.

Referring to the diff for  ti_vvc_framework_support_pkg.vhd you can comment out all the sequential statements in procedure activity_watchdog and still get the same error. That tells us the problem is with one of the remaining added declarations. 

Commenting out the type and constant declarations following the declaration of procedure activity_watchdog in package ti_vvc_framework_support_pkg allows  ti_vvc_framework_support_pkg.vhd to analyze.

Isolating which type declarations fail shows two: t_hvvc_to_brdige and t_bridge_to_hvvc which both contain record elements of type t_byte_array, Either declaration by itself is sufficient to cause the crash:

```vhdl
  type t_hvvc_to_bridge is record
    trigger                   : boolean;
    operation                 : t_vvc_operation;
    num_data_bytes            : positive;
    data_bytes                : t_byte_array;
    dut_if_field_idx          : natural;
    current_byte_idx_in_field : natural;
    msg_id_panel              : t_msg_id_panel;
    field_timeout_margin      : time;
  end record;

  type t_bridge_to_hvvc is record
    trigger        : boolean;
    data_bytes     : t_byte_array;
  end record;
```

`t_byte_array is from library uvvm_util fournd in types_pkg.vhd:
```vhdl
  type t_slv_array      is array (natural range <>) of std_logic_vector;
  ...
  subtype t_byte_array  is t_slv_array(open)(7 downto 0);
```
where the base type is an unbounded array with an unconstrained element index range.

If the record elements data_bytes have constraints provided by subtype indication in the record declarations analysis succeeds.

This looks like a second ghdl issue besides the uncaught exception.

andmeyee, 2019-12-17:
I observe a similar bug if compiling axistream_bfm_pkg.vhd from bitvis_vip_axistream which is relaying on uvvm_util.

If checking out UVVM v2019.09.02, everything compiles. For later versions, eg. v2019.11.21, I get the following error log:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-linux-gnu
/workspaces/bbuffers/build/bbuffers_0/sim-ghdl/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -frelaxed-rules -fexplicit --no-vital-checks --warn-binding --mb-comments --work=bitvis_vip_axistream -Plib/ -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o axistream_bfm_pkg.o ../../../../UVVM/bitvis_vip_axistream/src/axistream_bfm_pkg.vhd
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7f546a89283e 0x7f546bbba600 0x7f546badff18 0x7f546baf8c5a 0x7f546bb07b81 0x41942c 0x6266e2 0x60d803 0x60dd40 0x6118c1 0x5e6db6 0x5dcb7d 0x5e6f0a 0x5d6469 0x5dc6df 0x5e6f0a 0x58be45 0x58bf1f 0x5a5133 0x5a52cf 0x58dcad 0x58de47 0x5a516f 0x5a52cf 0x58dcad 0x58de47 0x5a516f 0x5a5307 0x62d55d 0x5cc370 0x62e76c 0x57b003 0x63d8ac 0x63f188 0x640a81 0x409e19 0x7f546a87f099 0x409128 0xfffffffffffffffe
******************************************************************
```
I am using the following Makefile for compilation:

``` Makefile
STD = --std=08
ANALYZE_OPTIONS = -frelaxed-rules -fexplicit --no-vital-checks --warn-binding --mb-comments -Plib

all: test

test:
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/types_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/adaptations_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/string_methods_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/protected_types_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/global_signals_and_shared_variables_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/hierarchy_linked_list_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/alert_hierarchy_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/license_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/methods_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/bfm_common_pkg.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=uvvm_util ./UVVM/uvvm_util/src/uvvm_util_context.vhd
	ghdl -a $(STD) $(ANALYZE_OPTIONS) --work=bitvis_vip_axistream ./UVVM/bitvis_vip_axistream/src/axistream_bfm_pkg.vhd
clean:
	rm -rf *.o
	rm -rf *.cf
```
If I only replace axistream_bfm_pkg.vhd with a later version (e.g. v2019.11.25) and keep uvvm_util files by v2019.09.02, everything compiles. That lets me assume that not the compilation of axistream_bfm_pkg.vhd but the compilation of uvvm_util is the problem.

mavric315, 2019-12-19:
I am seeing this issue also with windows build:

`GHDL 0.36 (v0.36) [Dunoon edition]
 Compiled with GNAT Version: 8.2.0
 llvm code generator`

Not sure if this was also reported but to get the latest version of UVVM to build, well to get at least till you encounter the bug,  the compile_uvvm.ps1/sh scripts need to be updated to include the ti_protected_types_pkg file at the top of the file list for the frame work:

`$UVVM_VVC_Files = @(
            "uvvm_vvc_framework\src\ti_protected_types_pkg.vhd",`

This file was added to the framework in 11.25.2019 and to the corresponding compile_order.txt file but the powershell scripts only use the compile_order.txt file for the VVCs not the utils nor the framework.

-Don

closed 2020-01-27 by tgingold