ghdl/ghdl #1044: synth: handle remaining real operations

Xiretza, 2019-11-28:
**Is your feature request related to a problem? Please describe.**
I use a lot of real math to calculate ROM contents in a project, and since always hunting down the next missing operation whenever one gets fixed isn't very efficient, I'll just list the missing ones here.

**Describe the solution you'd like**
The following operations are implemented for static values:

- [x] `Iir_Predefined_Floating_Equality`
- [x] `Iir_Predefined_Floating_Inequality`
- [x] `Iir_Predefined_Floating_Less`
- [x] `Iir_Predefined_Floating_Less_Equal`
- [x] `Iir_Predefined_Floating_Greater`
- [x] `Iir_Predefined_Floating_Greater_Equal`
- [ ] `Iir_Predefined_Floating_Identity`
- [ ] `Iir_Predefined_Floating_Negation`
- [ ] `Iir_Predefined_Floating_Absolute`
- [ ] `Iir_Predefined_Floating_Plus`
- [ ] `Iir_Predefined_Floating_Minus`
- [ ] `Iir_Predefined_Floating_Mul`
- [x] `Iir_Predefined_Floating_Div`
- [ ] `Iir_Predefined_Floating_Exp`
- [ ] `Iir_Predefined_Floating_Minimum`
- [ ] `Iir_Predefined_Floating_Maximum`
- [ ] `Iir_Predefined_Floating_To_String`

tgingold, 2019-11-29:
Can you also provide a reproducer of your design to improve testing of real operations ?  That would be nice.

tgingold, 2019-12-03:
I think that all vhdl93 operations are now implemented.

I close this issue; if you also need 08 operations, please create a new issue!

closed 2019-12-03 by tgingold