ghdl/ghdl #1048: synth: instantiations in generate statement get duplicate names

Fatsie, 2019-11-30:
**Description**
When using for ... generate and doing cell instantiations in the loop the exanded synth output will have instantiations will duplicate names. This then errors when trying to get it into yosys.
Although #1007 also concerns duplicate names, that ticket handled two instantiations of the same cell with different value for the generics.

**Expected behaviour**
No duplicate names, likely names need to be uniqified using loop label and loop index.

**How to reproduce?**
Given following code:
```vhdl
library ieee;
use ieee.std_logic_1164.ALL;

entity child is
  port (
    I: in std_logic;
    O: out std_logic
  );
end entity child;

architecture rtl of child is
begin
  O <= I;
end architecture rtl;

library ieee;
use ieee.std_logic_1164.ALL;

entity top is
  port (
    I: in std_logic;
    O: out std_logic
  );
end entity top;

architecture rtl of top is
  component child is
    port (
      I: in std_logic;
      O: out std_logic
    );
  end component child;

  constant N: integer := 3;

  signal T: std_logic_vector(N-1 downto 0);
begin
  CHILDGEN: for it in 0 to N-2 generate
    childinst : child
      port map(
        I => T(it),
        O => T(it+1)
      );
  end generate CHILDGEN;
  
  T(0) <= I;
  O <= T(2);
    
end architecture rtl;
```
And doing analysis and synth results in following output:
```vhdl
library ieee;
use ieee.std_logic_1164.all;
entity top is
  port (
    I: in std_logic;
    O: out std_logic
  );
end entity top;
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity child is
  port (
    i : in std_logic;
    o : out std_logic);
end entity child;

architecture rtl of child is
  signal n13_o : std_logic;
begin
  o <= n13_o;
  -- top.vhdl:7:5
  n13_o <= i; -- (output)
end rtl;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of top is
  signal wrap_i: std_logic;
  signal wrap_o: std_logic;
  signal n2_o : std_logic;
  signal t : std_logic_vector (2 downto 0);
  signal n3_o : std_logic;
  signal n5_o : std_logic;
  signal n6_o : std_logic;
  signal n8_o : std_logic;
  signal n9_o : std_logic;
  signal n10_o : std_logic_vector (2 downto 0);
begin
  wrap_i <= i;
  o <= wrap_o;
  wrap_o <= n2_o;
  -- top.vhdl:23:5
  n2_o <= n9_o; -- (output)
  -- top.vhdl:37:10
  t <= n10_o; -- (signal)
  -- top.vhdl:42:15
  n3_o <= t (0);
  childinst : entity work.child port map (
    i => n3_o,
    o => n5_o);
  -- top.vhdl:42:15
  n6_o <= t (1);
  childinst : entity work.child port map (
    i => n6_o,
    o => n8_o);
  -- top.vhdl:48:9
  n9_o <= t (2);
  n10_o <= n8_o & n5_o & wrap_i;
end rtl;
```
One can see that there are two ``work.child`` instances with label ``childinst``.

diogratia, 2019-12-01:
The synthesized results represent a flattened net list where a hierarchical net list would have two blocks labeled childgen each containing instances labeled childinst. 

Path name elements for the blocks would be of the form  
```none
generate_statement_label [ ( static_expression ) ]
```
 (see IEEE Std 1076-2008 8.7 External names) "If the generate statement is a for generate statement, the pathname element shall include a static expression, the type of the expression shall be the same as the type of the generate parameter, and the value of the expression shall belong to the discrete range specified for the generate parameter."

It would seem net list flattening lost an element of the path name.

tgingold, 2019-12-01:
Yes, I knew.  That's the next issue!

commit 8b13ca7a (2019-12-02): synth-insts: use current prefix for instantiations.  Fix #1048
closed 2019-12-02 by tgingold