ghdl/ghdl #1049: Missing function declaration in fixed_pkg

ktbarrett, 2019-12-01:
Am I insane or is there no function declaration for this function in fixed_generic_pkg.vhdl?
https://github.com/ghdl/ghdl/blob/3bd9a185da8178e878e8034e04aab2238bab50ff/libraries/ieee2008/fixed_generic_pkg-body.vhdl#L266-L283
 It's not marked as a a purely internal function, and it's reciprocal function `to_sfixed(ufixed)` is declared in the pkg.

Note I was using the files here as a source reference, I did not run GHDL to observe this issue.

tgingold, 2019-12-01:
Apparently, you're right.

diogratia, 2019-12-01:
It appears intentional note the comments lines 366 - 368 in fixed_generic_pkg-body.vhdl:

```vhdl
  ---------------------------------------------------------------------------
  -- Visible functions
  ---------------------------------------------------------------------------
```

indicating function specifications following are also found as declarations in fixed_generic_pkg.vhdl.

There is a function specification found prior to line 366 that does show up as a declaration in 
fixed_generic_pkg.vhdl:

```vhdl
   335    function to_ufixed (
   336      arg                  : STD_ULOGIC_VECTOR;  -- shifted vector
   337      constant left_index  : INTEGER;
   338      constant right_index : INTEGER)
   339      return unresolved_ufixed
   340    is
   341      variable result : UNRESOLVED_ufixed (left_index downto right_index);
   342    begin
   343      if (arg'length < 1 or right_index > left_index) then
   344        return NAUF;
   345      end if;
   346      if (arg'length /= result'length) then
   347        report fixed_generic_pkg'instance_name & "TO_UFIXED(SLV) "
   348          & "Vector lengths do not match.  Input length is "
   349          & INTEGER'image(arg'length) & " and output will be "
   350          & INTEGER'image(result'length) & " wide."
   351          severity error;
   352        return NAUF;
   353      else
   354        result := to_fixed (arg         => UNRESOLVED_UNSIGNED(arg),
   355                            left_index  => left_index,
   356                            right_index => right_index);
   357        return result;
   358      end if;
   359    end function to_ufixed;
```

The fixed_generic_pkg.vhdl file includes:

```vhdl
    26  --   Note      :  This package may be modified to include additional data
    27  --             :  required by tools, but it must in no way change the
    28  --             :  external interfaces or simulation behavior of the
    29  --             :  description. It is permissible to add comments and/or
    30  --             :  attributes to the package declarations, but not to change
    31  --             :  or delete any original lines of the package declaration.
    32  --             :  The package body may be changed only in accordance with
    33  --             :  the terms of Clause 16 of this standard.
    34  --             :
```

denying permission to alter the package declaration other than comments or attributes where:

16.10 Fixed-point package  
>The library denoted by the library logical name IEEE contains packages FIXED_FLOAT_TYPES, FIXED_GENERIC_PKG, and FIXED_PKG.12 The following conformance rules shall apply as they pertain to the use and implementation of these packages:  
>>a) The package declarations may be modified to include additional data required by tools, but modifications shall in no way change the external interfaces or simulation behavior of the description. It is permissible to add comments and/or attributes to the package declarations, but not to change or delete any original lines of the approved package declaration.  
>>b) The FIXED_GENERIC_PKG package body and the FIXED_PKG package instantiation declaration provided in the machine-readable files accompanying this standard (see Annex A) represent the for- mal semantics of the implementation of the FIXED_GENERIC_PKG and FIXED_PKG packages. Implementers of these packages may choose to simply compile the package body and package instantiation declaration as it is, or they may choose to implement the packages in the most efficient form available to the user. **_Implementers shall not implement semantics that differ from the formal semantics provided herein._**  
>  
>The text of the FIXED_GENERIC_PKG package (both package declaration and package body) and the text of the FIXED_PKG instantiated package are included with this standard (see Annex A). Those texts are official parts of this standard.  

doing so would alter the semantics (making the function declaration visible). Doing so would be in the purview of VASG.

tgingold, 2019-12-03:
I am closing this issue as there is nothing to be done in ghdl.  Se use the official implementation.

closed 2019-12-03 by tgingold

1138-4EB, 2019-12-03:
ping @JimLewis, @Paebbels