ghdl/ghdl #1051: GHDL Crash in psi_common_i2c_master_tb

obruendl, 2019-12-02:
**Description**
GHDL Crashes. The same simulation works fine in Modelsim.

**Expected behaviour**
GHDL executes simulation without crash.

**How to reproduce?**
1. Clone https://github.com/paulscherrerinstitute/psi_fpga_all (with --recursive switch)
2. In the file <root>/VHDL/psi_common/sim/config.tcl, remove the line that contains "tb_run_skip Vivado GHDL" (line 287)
3. In the working copy, navigate to <root>/VHDL/psi_common/sim
4. In a TCL shell, execute "source ./runGhdl.tcl

**Context**

- OS: Win10

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36 (v0.36) [Dunoon edition]
Compiled with GNAT Version: 7.3.0
Target: i686-w64-mingw32
In directory: D:\github\psi_fpga_all\VHDL\psi_common\sim\
Command line:
C:\tools\GHDL\bin\ghdl.exe --elab-run --std=08 -frelaxed-rules -Wno-shared --work=psi_common --workdir=psi_common psi_common_i2c_master_tb -gInternalTriState_g=true --ieee-asserts=disable
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:3642
Call stack traceback locations:
0x63f38c 0x64604f 0x68e965 0x645e0b 0x68fba1 0x60bbd8 0x6098d1 0x6149d2 0x699569 0x5f2674 0x589224 0x69b94f 0x402635 0x401389 0x743862c2 0x77221e37 0x77221e02
******************************************************************
```

diogratia, 2019-12-03:
This problem can be demonstrated with fewer source files:
```none
psi_tb_txt_util.vhd
psi_tb_compare_pkg.vhd
psi_tb_activity_pkg.vhd
psi_tb_i2c_pkg.vhd
psi_common_logic_pkg.vhd
psi_common_array_pkg.vhd
psi_common_math_pkg.vhd
psi_common_bit_cc.vhd
psi_common_i2c_master.vhd
psi_common_i2c_master_tb.vhd
```
It also shows up when analyzing psi_tb_i2c_pkg.vhd using an llvm backend:

```none
ghdl -a --std=08 -frelaxed-rules psi_tb_i2c_pkg.vhd
psi_tb_i2c_pkg.vhd:162:24:warning: type of a shared variable must be a protected type [-Wshared]

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-892-ga462aa2) [Dunoon edition]
Compiled with GNAT Version: GPL 2017 (20170515-63)
Target: x86_64-apple-darwin14.5.0
/Users/diogratia/Desktop/psi_common-0503ad59cd2d7aa9f4d042f4f377b7f936b740bb/hdl/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -frelaxed-rules -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o psi_tb_i2c_pkg.o psi_tb_i2c_pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap7.adb:3634
Load address: 0x100e1b000
Call stack traceback locations:
0x100feece2 0x100fef1d0 0x100fccee6 0x100fcd4a7 0x100fcd5a6 0x100fd786d 0x101038ca0 0x100fd7342 0x10103a032 0x100f89709 0x101047bef 0x10104909c 0x10104a565 0x100e1e894
******************************************************************
ghdl:error: compilation error
```

 The mcode trans-chap7.adb line numbers were the same in both error messages trying with both mcode and llvm, apparently in procedure Translate_Record_Aggregate_Bounds. 

The cause appears to be the record declaration with elements of unbounded array types (string) in the psi_tb_i2c_pkg.vhd package psi_tb_i2c_pkg package body:

```vhdl
    type MsgInfo_r is record
        Prefix    : string;
        Func    : string;
        User    : string;
    end record;
```
I'm personally sympathetic to having VHDL bugs in ghdl. Not so much for using VHDL as a verification language. On the plus side all these design units were compiled into work, meaning friendly use clause suffixes, and a shorter test case can be built without modifying design files. I'd suspect an even smaller test case could be forthcoming.

It's likely the error will show up in ghdl late build mcode versions.

1138-4EB, 2019-12-03:
> Not so much for using VHDL as a verification language.

Just because of curiosity, why?

obruendl, 2019-12-16:
Well, there is more capable languages out there for verification (System Verilog would most likely be the tool of choice). 

Our decision for using VHDL as verification language accross all PSI VHDL Libraries was taken for the following reasons:
- Everybody using VHDL Libraries can understand VHDL. Not all developers using VHDL know system verilog 
- VHDL is available with all simulators. System Verilog not.

These decisions certainly have good and bad sides. I just wanted to give the reasoning.

obruendl, 2019-12-16:
Thank you for the hint with the unbound array types in the record. I will check if I can find a workaround to make our code running.

However, I would appreciate if this could be fixed in GHDL. I guess we will not be the last ones trying to use something like this.

commit 91faff64 (2020-01-02): vhdl: handle untranslated subtypes for record aggregates.  Fix #1051
closed 2020-01-02 by tgingold