ghdl/ghdl #1057: use of alias of enumeration should implicitly make literals visible

kraigher, 2019-12-17:
```vhdl
package pkg is
  type enum_t is (alpha, beta);
  alias alias_t is enum_t;
end package;

-- Enum literals should be made implicitly visible here
use work.pkg.alias_t;

package pkg2 is
  -- Fails with no declaration for "alpha"
  constant c : alias_t := alpha;
end package;
```

I found it in `UVVM/uvvm_vvc_framework/src_target_dependent/td_vvc_entity_support_pkg.vhd`

diogratia, 2019-12-17:
IEEE Std 1076-2008 6.6.3 Nonobject aliases:

>The following rules apply to nonobject aliases:  
>>...  
>>c) If the name denotes an enumeration type or a subtype of an enumeration type, then one implicit alias declaration for each of the literals of the base type immediately follows the alias declaration for the enumeration type; each such implicit declaration has, as its alias designator, the simple name or character literal of the literal and has, as its name, a name constructed by taking the name of the alias for the enumeration type or subtype and substituting the simple name or character literal being aliased for the simple name of the type or subtype. Each implicit alias has a signature that matches the parameter and result type profile of the literal being aliased.

No so much the enumeration literals (not so implicitly declared see 5.2.2 Enumeration types) declaration being visible here as implicit aliases of the enumeration literal declarations. The distinction allows enumeration literal overloading to work properly (as it should here).

kraigher, 2019-12-18:
I do not see how the text you quoted rules out the behavior I expect. It must be reasonable to treat using an alias the same as using the actual type so that it is transparent. I would also expect implicit functions to be made visible as well for an alias of say an integer type.

kraigher, 2019-12-18:
Also as this type of code is found in UVVM it is likely that other simulator implementations have made the interpretation that it is the reasonable behavior.

diogratia, 2019-12-18:
You missed the point. The quoted portion of the standard supports your contention. You're attempting to spar for no reason.

kraigher, 2019-12-18:
Ah then I misunderstood the initial "No" in your reply :-)

commit 2274e498 (2019-12-26): vhdl: handle non-object aliases in selected use clause.  Fix #1057
closed 2019-12-26 by tgingold