ghdl/ghdl #1060: Replace IEEE libraries with VHDL-2019 packages (Apache license)

eine, 2019-12-30:
> [December 30, 2019 10:32 PM](https://gitter.im/ghdl1/Lobby?at=5e0a6cdd0fd3413f4c7df244)
>
> On 20.12.2019, [IEEE Std. 1076-2019](https://standards.ieee.org/standard/1076-2019.html) was officially released.
> In parallel, all [VHDL packages](https://opensource.ieee.org/vasg/Packages) (std.standard, std.env, ..., ieee.std_logic_1164, ...) have been release as **OPEN SOURCE** under Apache License, 2.0.

> [December 30, 2019 10:50 PM](https://gitter.im/ghdl1/Lobby?at=5e0a71282d574c7d1ee3c7a6)
>
> The good with VHDL-2019 packages and Apache license is as follows:
> 1. GHDL project creates a clone of the original sources and keeps the master branch.
> 2. GHDl implements [LCS-2016-061](http://www.eda-twiki.org/cgi-bin/view.cgi/P1076/LCS2016_061) to support conditional analysis
> 3. GHDl creates a modification branch, which adds conditional analysis statements into the sources.  
>    Thus, we can use VHDL-2019 sources for 1987, 1993, 2002, 2008, 2019

This issue is to track the migration of current non free IEEE sources, and openieee, to the open source [VHDL packages](https://opensource.ieee.org/vasg/Packages).

I think that the natural procedure would be to fork/clone the repo to this organization and add it as a git submodule in subdir `libraries`. Then, fix the makefiles accordingly.

However, there have been previous reluctances to using submodules in this org. @tgingold, what do you think?

tgingold, 2019-12-31:
It was already done.  The files in the `ieee2008` directory use the open source licensed files from 1076-2019.

What is missing:

* have a  downgrade version for vhdl-93

* there is no open-source version for the VITA packages,

eine, 2019-12-31:
> It was already done. The files in the `ieee2008` directory use the open source licensed files from 1076-2019.

Oh, I missed that. I now checked the history of that subdir... However, `libraries/openieee/v08` still exists. Shouldn't it be possible not to require `openieee` for v08?

Moreover, IEEE libraries are now expected to change faster, since a public repo exists. Manually copying them can make it harder to keep them up to date or to contribute fixes upstream. I think we should either have a script to do it automatically, or have these sources added as a subdir which is copied raw.

> have a  downgrade version for vhdl-93

What do you think about supporting all the versions with the same sources through conditional analysis?

> there is no open-source version for the VITA packages,

Does this mean that we still need to keep the `--gpl` option for Debian?

tgingold, 2019-12-31:
> Oh, I missed that. I now checked the history of that subdir... However, libraries/openieee/v08 still exists. Shouldn't it be possible not to require openieee for v08?

IIRC, by default openieee are not used.  To be confirmed.

> Moreover, IEEE libraries are now expected to change faster, since a public repo exists. Manually copying them can make it harder to keep them up to date or to contribute fixes upstream. I think we [...]

That's true for vhdl-2019, but not for vhdl-08 which is frozen.  So currently it is just a backport to vhdl 2008.

I agree that for vhdl 2019 we'd better to have a raw copy.

> have a downgrade version for vhdl-93
> 
> What do you think about supporting all the versions with the same sources through conditional analysis?

I really prefer to have pure vhdl-93 files, so that they could be used by any tool.

> Does this mean that we still need to keep the --gpl option for Debian?

Yes, I do think so.

eine, 2019-12-31:
> That's true for vhdl-2019, but not for vhdl-08 which is frozen. So currently it is just a backport to vhdl 2008.

I'm not sure about vhdl-08 libs being frozen. I know it has been discussed before, but I am not aware of a definitive outcome. Anyway, it is fair enough to consider it to be frozen from GHDL's perspective.

> I agree that for vhdl 2019 we'd better to have a raw copy.

Any strong feeling between a (kind of manual) raw copy or a git submodule?

> I really prefer to have pure vhdl-93 files, so that they could be used by any tool.

Good. If VASG decides to implement conditional sources upstream in the future, we can always adapt to them.

eine, 2020-07-23:
Ref https://github.com/ghdl/ghdl/issues/1255#issuecomment-619308878.

PR #1602 (2021-01-16): libraries: unuse 'openieee', rename enable_openieee to enable_gplcompat
Close #1060.
Ref: [December 10, 2020 6:06 AM](https://gitter.im/ghdl1/Lobby?at=5fd1acbf0697c1210db1246f)

This PR removes `openieee` (except for `upf` packages), since current ieee libraries are GPL compatible. At the same time, configure option `enable-openieee` is renamed to `enable-gplcompat`. Now, that option decides whether VITAL libraries (95 and 00) are included or not. The configure script, makefiles and CI scripts are updated accordingly.

@tgingold, did you actually expect to remove the sources from `libraries/op [... cut]
closed 2021-01-16 by tgingold