ghdl/ghdl #1062: synth: allow defining top-level generics using -g command-line

Xiretza, 2019-12-31:
`-gGENERIC=VALUE` is currently [only available as a run option](https://ghdl.readthedocs.io/en/latest/using/Simulation.html), but would come in very useful for synthesis as well. Additionally, being able to supply a definitions file (e.g. newline-separated `key=value` pairs) would be nice for designs needing many generics.

eine, 2019-12-31:
For the second requirement, I think you could use [Paebbels/JSON-for-VHDL](https://github.com/Paebbels/JSON-for-VHDL). Although I have only used it for simulation, I believe that it is synthesisable. In VUnit, JSON-for-VHDL is supported as a mechanism to pass either the path to a JSON file or stringified JSON content as a generic of type `string`. Anyway, you should be able to hardcode the path to the JSON file in a constant. The advantage of JSON-for-VHDL is that it supports complex config structures (nested lists/objects).

Paebbels, 2019-12-31:
JSON-for-VHDL ist fully synthezisable. It's based on file I/O and arrays of fixed size. Be aware, it's a very hard test for synthesis engines or in other words: almost all will fail to run that code :).

commit f171b7a7 (2020-01-01): vhdl: handle -gGEN=VAL for --synth.  Fix #1062
closed 2020-01-01 by tgingold