ghdl/ghdl #1063: [Compliance-Tests] vhdl_2008/tb_matching_operator crash during analysis

eine, 2020-01-01:
Source file https://github.com/VHDL/Compliance-Tests/blob/master/vhdl_2008/tb_matching_operator.vhd contains multiple tests for VHDL 2008 "matching operator" features. Unfortunately, none of them can be tested because GHDL crashes.

```vhdl :file:test.vhd
library vunit_lib;
context vunit_lib.vunit_context;

library ieee;
use ieee.std_logic_1164.all;

entity tb_matching_operators is
  generic (runner_cfg : runner_cfg_t);
end entity tb_matching_operators;

architecture tb of tb_matching_operators is
begin
  test_runner: process is
  begin
    test_runner_setup(runner, runner_cfg);
    check_equal('1' ?= '1', '1');
    test_runner_cleanup(runner);
    wait;
  end process test_runner;
end architecture tb;

```

```py :run:
#!/usr/bin/env python3

from os.path import join, dirname
from vunit import VUnit, VUnitCLI

root = dirname(__file__)

ui = VUnit.from_argv(vhdl_standard="2008")
ui.add_library("vhdl_2008").add_source_files(join(root, "*.vhd"))

ui.main()

#:image: ghdl/vunit:llvm
```

```
=== Command used: ===
/usr/local/bin/ghdl -a --workdir=/root/vunit_out/ghdl/libraries/vhdl_2008 --work=vhdl_2008 --std=08 -P/root/vunit_out/ghdl/libraries/vunit_lib -P/root/vunit_out/ghdl/libraries/vhdl_2008 /root/test.vhd

=== Command output: ===

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-linux-gnu
/root/
Command line:
/usr/local/bin/ghdl1-llvm --workdir=/root/vunit_out/ghdl/libraries/vhdl_2008 --work=vhdl_2008 --std=08 -P/root/vunit_out/ghdl/libraries/vunit_lib/ -P/root/vunit_out/ghdl/libraries/vhdl_2008/ -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o /root/vunit_out/ghdl/libraries/vhdl_2008/test.o /root/test.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-evaluation.adb:1760
Call stack traceback locations:
0x4ed66a 0x4f1855 0x4f261f 0x4f2b71 0x5482b8 0x548ad9 0x5513d4 0x551da1 0x552e0d 0x564194 0x5643bf 0x5646cf 0x54733c 0x547583 0x547cd8 0x530e0b 0x530f37 0x531b78 0x531fe8 0x5321fe 0x532355 0x565f2d 0x56cadb 0x529686 0x63cf87 0x63ebba 0x6404b3 0x409e19 0x7f8c22840099 0x409128 0xfffffffffffffffe
******************************************************************
/usr/local/bin/ghdl: compilation error
```

Using an assertion instead of VUnit's `check_equal`, does not crash:

```
assert '1' ?= '1' severity failure;
```

tgingold, 2020-01-01:
Not sure this is an interesting testcase: ghdl was failing at evaluating matching equality during analysis time, but this works at run time.

closed 2020-01-01 by tgingold

eine, 2020-01-02:
Thanks! Now 6 of 8 equality tests pass. The following fail:

```
check_equal('1' ?= 'H', '1');
0 fs - check                -   ERROR - Equality check failed - Got X. Expected 1.

check_equal('1' ?= 'W', 'X');
0 fs - check                -   ERROR - Equality check failed - Got 0. Expected X.
```

According to the table at page 122 of the LRM, I think that GHDL is wrong here.

Apart from that, other similar operators (`?/=`, `?<`, `?<=`, `?>`, `?>=`) are not implemented yet. Shall we reopen this issue to track all these here, or is it better to create separate issues?

tgingold, 2020-01-02:
Humm, there is a stupid mistake.

commit c7a4c4e1 (2020-01-01): vhdl: evaluate std_logic matching equality.  Fix #1063