ghdl/ghdl #1066: --ieee=synopsys doesn't actually add packages?

jgarvin, 2020-01-04:
**Description**
I'm trying to compile this code:

```
library ieee;
use ieee.std_logic_1164.all;
use ieee.std_logic_signed.all;
use ieee.std_logic_unsigned.all;
use ieee.numeric_std.all;

entity Add4 is port (
  Data1, Data2 : in std_logic_vector(3 downto 0);
  Cin : in std_logic;
  Cout : out std_logic;
  Sum : out std_logic_vector(3 downto 0) );
end entity Add4; 

architecture RTL of Add4 is
  signal Out5bit : unsigned(4 downto 0);
begin
  Out5bit <= ('0' & Data1) + ('0' & Data2) + Cin;
  Sum <= Out5bit(3 downto 0);
  Cout <= Out5bit(4);
end architecture RTL;
```
I get errors about the signed and unsigned package, suggesing I use `--ieee=synopsys` but I have already done that, here is my command line:

`ghdl -a add4.vhd -fno-color-diagnostics --ieee=synopsys |& cat`

The cat is to inhibit colors since `-fno-color-diagnostics` doesn't appear to work. Here are the errors:

```
add4.vhd:3:10: unit "std_logic_signed" not found in library "ieee"
add4.vhd:3:10:  (use --ieee=synopsys for non-standard synopsys packages)
add4.vhd:4:10: unit "std_logic_unsigned" not found in library "ieee"
add4.vhd:4:10:  (use --ieee=synopsys for non-standard synopsys packages)
add4.vhd:14:14: entity 'add4' was not analysed
```

This is confusing on two counts:

1. That flag is documented as solving this exact problem, but seems not to.
2. It's weird to tell the user to use a flag they're already using.

**Expected behaviour**
I'd expect it to compile successfully :)

**How to reproduce?**
Compile the code above with the command above.

**Context**
Please, provide the following information:

- OS: Ubuntu 18.10
- Origin
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `669dc74413a1b263f3377503d1b2703884550290`

tgingold, 2020-01-04:
Same issue: the options must appear before the files (as documented).

jgarvin, 2020-01-04:
Thanks that worked. That said though I think if you print a message telling users to add something to their command line most of them are just going to naively put it on the end. Maybe there should be an error if you have options that aren't followed by any sources? That would have clued me in immediately.

eine, 2020-01-04:
@jgarvin, GHDL is following the GNU/POSIX convention of not supporting flags after positional arguments. It might feel strange sometimes, for those of us used to other cli tools that support mixing them; but it is easy to remember that options go first and `--` means "option".

Nonetheless, I agree that it would be useful to print a warning (not error) if unexpected arguments are provided (and ignored).

commit 2361fe47 (2020-03-05): Warn on unexpected options for command '-a'.  Fix #1066
closed 2020-03-05 by tgingold