ghdl/ghdl #1067: Unconstrained Array Types failing in ghw

nfrancque, 2020-01-05:
**Description**
Similar to #538, designs using unconstrained arrays seem to simulate properly but I can't pull them up in a waveform viewer.  Using ghdl's version of ghwdump with all verbose outputs results in this:
```
Number of strings: 20
String table size: 125
 string 1 (pl=0): '-'
 string 2 (pl=1): '0'
 string 3 (pl=1): '1'
 string 4 (pl=1): 'H'
 string 5 (pl=1): 'L'
 string 6 (pl=1): 'U'
 string 7 (pl=1): 'W'
 string 8 (pl=1): 'X'
 string 9 (pl=1): 'Z'
 string 10 (pl=0): integer
 string 11 (pl=0): natural
 string 12 (pl=1): numeric_std
 string 13 (pl=0): standard
 string 14 (pl=2): std_logic_1164
 string 15 (pl=4): std_ulogic
 string 16 (pl=10): std_ulogic_vector
 string 17 (pl=0): t_slv_array
 string 18 (pl=1): test
 string 19 (pl=2): textio
 string 20 (pl=1): top
enum std_ulogic: 'U' 'X' '0' '1' 'Z' 'W' 'L' 'H' '-'
scalar: integer
subtype scalar: natural
array: std_ulogic_vector
array: t_slv_array
subtype array: <anon> (nbr_scalars=-3)
std_ulogic: wkt=3
0 scopes, 1 signals, 24 signal elements
ghw_read_base: error in section HIE
cannot read ghw file
```

**Expected behaviour**
I would expect the design to successfully simulate and be dumped out by ghwdump.
Note that the similar example provided in #538 pertaining to records does not fail anymore. 

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
library ieee;                                             
   use ieee.std_logic_1164.all;   
   use ieee.numeric_std.all;      

entity top is
end entity ; 

architecture arch_top of top is

    type t_slv_array is array(natural range <>) of std_logic_vector;

    signal test : t_slv_array(0 to 2)(7 downto 0);

begin

end architecture arch_top;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 top --wave=wave.ghw
ghwdump -vvv -t wave.ghw
```

**Context**
Please, provide the following information:

- OS: Ubuntu 18.04
- Origin:
  - [  ] Package manager: `version`
  - [  ] Released binaries: `tarball_url`
  - [x] Built from sources: `9b1191a6`

marph91, 2020-02-01:
I got the same issue and tried to check the code (`src/grt/ghwlib.c`) a bit. It seems the `nbr_scalars=-3` comes from `ghw_read_array_subtype()`. There the length of the outer dimension (= 3) seems to be correctly calculated. Further, `get_nbr_elements()` returns -1 for unbound types. The values get multiplicated to -3. I believe, `get_nbr_elements()` needs to be modified to fix this issue, but don't know how to proceed further.

It would be very helpful to be able to visualize unconstrained arrays.

nfrancque, 2020-02-01:
I went down a similar rabbit hole of messing with the code and saw the same
results.

Unfortunately not knowledgeable enough to make the right changes. For now I
am generating a vcd instead, in this case I can see all the other signals
at least.

timor, 2020-02-14:
We encountered the same problem.  Would the fix for this lie in the scope of the generator or the viewer?

nfrancque, 2020-02-14:
Reasonably confident it's the generator.  See @marph91's comment.  As I understand the viewer (primarily) using ghwdump.c.

tgingold, 2020-02-14:
Yes, at least the generator (ghdl) doesn't emit all the infos.

csantosb, 2020-03-01:
Same problem here. Using [this](https://gitlab.com/ip-vhdl/mux_fifo/-/tree/master/src) code, I manage to elaborate the `mux_fifo` unit (provided I pass the `mux_fifo_bus_width` generic, otherwise it crashes; just checked using [vivado19.2](https://gitlab.com/ip-vhdl/mux_fifo/-/tree/vivado19.2DoesIt/src), and there is no problem without it), but then I'm unable to open the resulting `wave.ghw` file using gtkwave 3.3.104-1, and produced with [this](https://gitlab.com/ip-vhdl/mux_fifo/-/blob/master/sim/runs/mux_fifo.ghdl/hdl/mux_fifo_tb.vhd) test bench.

Running latest (as for just now) `ghdl-llvm` under arch linux.

commit 36f25166 (2020-08-04): testsuite/gna: add a test for #1067

Close #1067
closed 2020-08-04 by tgingold