ghdl/ghdl #1069: synth: add support for inferring true dual-port memories

Xiretza, 2020-01-05:
**Is your feature request related to a problem? Please describe.**
Currently, synthesis seems to support memories with arbitrarily many differently-sized read and write ports, but they all have to be in the same clock domain. Most real-world FPGA implementations however support true dual port (TDP) memories with two completely separate read+write access ports, possibly with unrelated clocks.

In VHDL < 2008, this was rather easy to model, the array backing the memory just had to be declared a `shared variable`. Starting with VHDL 2008, shared variables have to be of a protected type, adding another layer of abstraction.

**Describe the solution you'd like**
Since ghdl doesn't allow mixing design sources of different VHDL versions, both styles of shared variable would have to be supported. If protected types pose too much of a problem, an alternative would be writing the memory in e.g. VHDL 93, and allowing this module to be imported into the surrounding VHDL 2008 design.

**Additional context**
Below is a slightly adapated design from Xilinx' [User Guide 901 "Vivado Synthesis"][ug901] with:

- Two asymmetric read+write ports with separate clocks
- Byte enables
- Output registers

```vhdl
library ieee;
use ieee.std_logic_1164.all,
    ieee.numeric_std.all;

entity tdp_ram is
    generic (
        ADDRWIDTH_A : positive := 12;
        WIDTH_A     : positive := 8;

        ADDRWIDTH_B : positive := 10;
        WIDTH_B     : positive := 32;

        COL_WIDTH : positive := 8
    );
    port (
        clk_a        : in  std_logic;
        read_a       : in  std_logic;
        write_a      : in  std_logic;
        byteen_a     : in  std_logic_vector(WIDTH_A/COL_WIDTH - 1 downto 0);
        addr_a       : in  std_logic_vector(ADDRWIDTH_A - 1 downto 0);
        data_read_a  : out std_logic_vector(WIDTH_A - 1 downto 0);
        data_write_a : in  std_logic_vector(WIDTH_A - 1 downto 0);

        clk_b        : in  std_logic;
        read_b       : in  std_logic;
        write_b      : in  std_logic;
        byteen_b     : in  std_logic_vector(WIDTH_B/COL_WIDTH - 1 downto 0);
        addr_b       : in  std_logic_vector(ADDRWIDTH_B - 1 downto 0);
        data_read_b  : out std_logic_vector(WIDTH_B - 1 downto 0);
        data_write_b : in  std_logic_vector(WIDTH_B - 1 downto 0)
    );
end tdp_ram;

architecture behavioral of tdp_ram is
    function log2(val : INTEGER) return natural is
        variable res : natural;
    begin
        for i in 0 to 31 loop
            if (val <= (2 ** i)) then
                res := i;
                exit;
            end if;
        end loop;

        return res;
    end function log2;

    function eq_assert(x : integer; y : integer) return integer is
    begin
        assert x = y;
        return x;
    end function eq_assert;
    
    constant COLS_A : positive := WIDTH_A / COL_WIDTH;
    constant COLS_B : positive := WIDTH_B / COL_WIDTH;

    constant TOTAL_COLS : positive := eq_assert(COLS_A * 2 ** ADDRWIDTH_A, COLS_B * 2 ** ADDRWIDTH_B);

    constant EXTRA_ADDR_BITS_A : positive := log2(COLS_A);
    constant EXTRA_ADDR_BITS_B : positive := log2(COLS_B);

    type ram_t is array(0 to TOTAL_COLS - 1) of std_logic_vector(COL_WIDTH - 1 downto 0);
    shared variable store : ram_t := (others => (others => '0'));

    signal reg_a : std_logic_vector(WIDTH_A - 1 downto 0);
    signal reg_b : std_logic_vector(WIDTH_B - 1 downto 0);
begin
    assert WIDTH_A mod COL_WIDTH = 0 and
           WIDTH_B mod COL_WIDTH = 0 and
           2 ** (ADDRWIDTH_A + EXTRA_ADDR_BITS_A) = TOTAL_COLS and
           2 ** (ADDRWIDTH_B + EXTRA_ADDR_BITS_B) = TOTAL_COLS
        report "Both WIDTH_A and WIDTH_B have to be a power-of-two multiple of COL_WIDTH"
        severity failure;

    process(clk_a)
    begin
        if rising_edge(clk_a) then
            for i in 0 to COLS_A - 1 loop
                if write_a = '1' and byteen_a(i) = '1' then
                    store(to_integer(unsigned(addr_a) & to_unsigned(i, EXTRA_ADDR_BITS_A))) :=
                        data_write_a((i+1) * COL_WIDTH - 1 downto i * COL_WIDTH);
                end if;

                if read_a = '1' then
                    reg_a((i+1) * COL_WIDTH - 1 downto i * COL_WIDTH) <=
                        store(to_integer(unsigned(addr_a) & to_unsigned(i, EXTRA_ADDR_BITS_A)));
                end if;
            end loop;

            data_read_a <= reg_a;
        end if;
    end process;

    process(clk_b)
    begin
        if rising_edge(clk_b) then
            for i in 0 to COLS_B - 1 loop
                if write_b = '1' and byteen_b(i) = '1' then
                    store(to_integer(unsigned(addr_b) & to_unsigned(i, EXTRA_ADDR_BITS_B))) :=
                        data_write_b((i+1) * COL_WIDTH - 1 downto i * COL_WIDTH);
                end if;

                if read_b = '1' then
                    reg_b((i+1) * COL_WIDTH - 1 downto i * COL_WIDTH) <=
                        store(to_integer(unsigned(addr_b) & to_unsigned(i, EXTRA_ADDR_BITS_B)));
                end if;
            end loop;

            data_read_b <= reg_b;
        end if;
    end process;
end behavioral;
```

```
$ ghdl --synth --std=93c tdp_ram.vhd -e tdp_ram
tdp_ram.vhd:63:24:error: multiple assignments for offsets 0:32767
```

[ug901]: https://www.xilinx.com/support/documentation/sw_manuals/xilinx2019_2/ug901-vivado-synthesis.pdf

Paebbels, 2020-01-05:
While implementing a TDP RAM with a shared variable is an accepted way in VHDL-93, it doesn't work in VHDL-87, and VHDL-2002 or newer.

The correct implementation of a true dual port RAM is by using one signal, one process, but with two clocks.

```vhdl
process(clock1, clock2)
begin
  if rising_edge(clock1) then
    ram(adr1) <= data1;
  end if;

  if rising_edge(clock2) then
    data2 <= ram(adr2);
  end if;
end process;
```

This form is also prooven to work with various synthesizers. See https://github.com/VLSI-EDA/PoC/blob/master/src/mem/ocram/ocram_tdp.vhdl?ts=2 for more details and a more complex implementation. In addition, the implementation in PoC fixes the incorrect simulation models of Xilinx and Altera/Intel for concurrent accesses in RAMs.

Xiretza, 2020-01-05:
> The correct implementation of a true dual port RAM is by using one signal, one process, but with two clocks.

Interesting, I had never seen that before, though VHDL inference models seem to be somewhat hard to come by in general. When the original example is modified as described (but changed to read-first for simplicity), ghdl crashes with an internal error:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1413-gdf5e3232) [Dunoon edition]
Compiled with GNAT Version: 9.2.0
Target: x86_64-pc-linux-gnu
/home/xiretza/temp/ghdl_synth_tests/
Command line:
ghdl --synth --std=93c tdp_ram.vhd -e tdp_ram
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-inference.adb:317
Call stack traceback locations:
0x48c5c2 0x48f6d8 0x5271ad 0x526e18 0x51fea4 0x51bf6d 0x558ff3 0x4dd732 0x406d11 0x7f876ec9d151 0x406d8c 0xfffffffffffffffe
******************************************************************
```

eine, 2020-01-05:
> If protected types pose too much of a problem, an alternative would be writing the memory in e.g. VHDL 93, and allowing this module to be imported into the surrounding VHDL 2008 design.

What about using a shared variable in VHDL 2008 with relaxed rules? It should be equivalent for this use case, since the only VHDL93 feature/requirement you want/have is precisely the shared variable.

hackfin, 2020-02-06:
I'd love to see that kind of support for TDP RAMs (which could be ROMs, too), with being able to pass the RAM data as a variable array instead of loading it from a file.
Throwing in a few models that have worked for me for a few synthesis tools/architectures. Kinda kept them simple and trying to do all guarding from a higher level (collisions..).
https://github.com/hackfin/MaSoCist/tree/opensource/hdl/ram
Also a crucial thing for me was/is the hint for the synthesizer to use either BlockRAM oder Distributed cells.

What ghdl-synth currently doesn't like though is this way of instancing it:
```
rom_l: DPRAM16_init_ce
 generic map (
  ADDR_W => 15 -2, -- 32 bit wide, LSB 0 aligned
  INIT_DATA => ram16_init_t(BOOTROM_PROG_INIT(0))
 )
...
```

with `../hdl/riscv/riscv32_soc.vhdl:260:28: unhandled type conversion (to array)`
Not sure if implementing this variant would open too many construction sites..

The reason for this 'fun' is that there could be RAM instances with a DATA_W(IDTH) of 13 bits but taking 16 bit initialization data (therefore DRAM16_init). Same thing with nbits > 16 and 32 bit initialization words.

hackfin, 2020-03-01:
I've started hacking a few test cases for different RAM types in MyHDL, reason for this is to spit out Verilog and VHDL code at the same and verify against working references by co-simulation.
Added it for the time being in the ghdlsynth-beta fork: https://github.com/hackfin/ghdlsynth-beta/tree/masocist/testsuite/myhdl/ram

What it does:
* When IMPLEMENTED is set to 1 in the *.py test script, it will synthesize the MyHDL->VHDL translation and write out a 'post map' Verilog file
* This file is compiled with iverilog and co-simulated with the test bench in the Python source
* If IMPLEMENTED == 0, the MyHDL simulator will be run only, when 2, the Verilog translation (pre-map) will be co-simulated

Quick starter to run it in a Docker container:
https://gist.github.com/hackfin/5bd9ce8f82d2ceacd24cf6889f1c2992

Currently, this is only run for the ECP5 architecture. There are already some discrepancies found between functionally identical Verilog and VHDL transfer, for example, write-through implementations of dual port rams are not identified correctly by yosys, or read-after-write may synthesize to different BRAM primitives (ECP5_DP16KD vs. TRELLIS_DPR16X4 distributed RAM).

tgingold, 2020-03-01:
I am late on this issue.  I still have to add support for dual-port memories with different clock.

hackfin, 2020-03-02:
There's another problem that there is no 'whitebox' simulation model available for the ECP5_DP16KD primitive, see also https://github.com/SymbiFlow/prjtrellis/issues/32. So proper post-map simulation requires the vendor models again (and they don't work out of the box due to some incomplete wiring from the yosys side, if I'm not mistaken)

tgingold, 2020-05-31:
I think this is now fixed.  Took a long time...

closed 2020-05-31 by tgingold