ghdl/ghdl #1073: Python libghdl documentation.

m-kru, 2020-01-08:
Is there any documentation for using libghdl as Python package? There is no documentation on [GHDL doc](https://ghdl.readthedocs.io/en/latest/). I know that it is used in the [ghdl-language-server](https://github.com/ghdl/ghdl-language-server), but it still requires analyzing the code of language server.

eine, 2020-01-08:
AFAIK, there is no documentation about libghdl-py (the term I use for https://github.com/ghdl/ghdl/tree/master/python/libghdl). However, @tingold recently pushed some info about the internals of GHDL (https://ghdl.readthedocs.io/en/latest/internals/Overview.html) and he is extending it. Please, let us know what are you trying to achieve, so we can better understand how you need to interact with the tool.

m-kru, 2020-01-08:
[vhdl-style-guide](https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/312). I have already evaluated the [pyVHDLParser](https://github.com/Paebbels/pyVHDLParser). Now I would like to do fiddle with libghl-py to compare the two.

Paebbels, 2020-01-08:
@m-kru the pyVHDLParser project got a lot of [new documentation](https://pyvhdlparser.readthedocs.io/en/latest/). At next it will get some bug fixes and improvements.

m-kru, 2020-01-08:
@Paebbels I really like the logic of pyVHDLParser, but I encounter errors on step 2 (Block Stream creation) for valid vhdl files. I guess they are valid, as `ghdl -s` does not return any errors.

Paebbels, 2020-01-08:
@m-kru maybe, we can continue discussions regarding pyVHDLParser and seen errors offline e.g. in [Gitter](https://gitter.im/Paebbels).

With respect to GHDL and Python, there was an offline discussion between me an @eine if both Python implementations could / should share the same data model / API.

tgingold, 2020-01-08:
No, there is no specific documentation.  It is a thin binding to the code.

There is also https://www.ohwr.org/project/vhdl-style/tree/master
but it uses a slightly older version of libghdl.

m-kru, 2020-01-08:
@Paebbels I have created issues in pyVHDLParsser repository. I am closing this issue as my question has been answered.

closed 2020-01-08 by m-kru