ghdl/ghdl #1076: synth: crash when else-generate block is used

Xiretza, 2020-01-10:
**Description**
When all conditions for an if-generate/elsif-generate compound fail, and an else-generate block is present, synthesis crashes.

**Expected behaviour**
The contents of the else-generate block are used if none of the previous if/elsif-generate statements have applied.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
	port (
		o : out std_logic
	);
end;

architecture a of ent is
begin
	gen: if false generate
		o <= '1';
	else generate
		o <= '0';
	end generate;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=08 ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 11a5be93

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1437-g11a5be93) [Dunoon edition]
Compiled with GNAT Version: 9.2.0
Target: x86_64-pc-linux-gnu
/home/xiretza/temp/ghdl_synth_tests/
Command line:
ghdl --synth --std=08 --workdir=ghdl_workdir else_generate.vhd -e ent
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:2816
Call stack traceback locations:
0x55a453fb0d42 0x55a453d622c6 0x55a453eb13a3 0x55a453e9a1a6 0x55a453e9a778 0x55a453e9a90b 0x55a453e8bfb0 0x55a453e8c0db 0x55a453e855eb 0x55a453ee3a3a 0x55a453e27aa4 0x55a453f96603 0x55a453be1d64 0x7f3ca9170151 0x55a453be058c 0xfffffffffffffffe
******************************************************************
```

closed 2020-01-11 by tgingold

hackfin, 2020-02-17:
There's another scenario where this pragma Assert occurs: When the condition uses a constant variable from a package.

Added patch for test case 1076:
[1076.patch.txt](https://github.com/ghdl/ghdl/files/4214159/1076.patch.txt)

tgingold, 2020-02-17:
Should be fixed.

commit 03d7679a (2020-01-11): synth: fix crash on else-generate.  Fix #1076