ghdl/ghdl #1077: synth: crash on shift/rotate operators

Xiretza, 2020-01-10:
**Description**
The shift and rotate operators defined for vector types (at least in VHDL 2008) can not be synthesized and cause a crash.

**Expected behaviour**
The operators are synthesized as per the standard (though notably [not the same as the related functions in `numeric_std`](http://jdebp.eu./FGA/bit-shifts-in-vhdl.html)).

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all,
    ieee.numeric_std.all;

entity ent is
	port (
		clk : in std_logic;
		i : in std_logic_vector(7 downto 0);

		amount : in integer range 0 to 7;

		const_half : out signed(7 downto 0);
		async_ror  : out std_logic_vector(7 downto 0);
		clocked_left : out unsigned(7 downto 0)
	);
end;

architecture a of ent is
	constant ONE : unsigned(3 downto 0) := x"4" srl 2;
begin
	const_half <= signed(i) sra to_integer(ONE);
	async_ror  <= i ror amount;

	process(clk)
	begin
		if rising_edge(clk) then
			clocked_left <= unsigned(i) sll amount;
		end if;
	end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=08 ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 11a5be93

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
shift_operations.vhd:19:52:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_NONE

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1437-g11a5be93) [Dunoon edition]
Compiled with GNAT Version: 9.2.0
Target: x86_64-pc-linux-gnu
/home/xiretza/temp/ghdl_synth_tests/
Command line:
ghdl --synth --std=08 --workdir=ghdl_workdir shift_operations.vhd -e ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-static_oper.adb:415
Call stack traceback locations:
0x5578e0932db0 0x5578e0928f19 0x5578e093d7a4 0x5578e0940cda 0x5578e094143f 0x5578e0941940 0x5578e0918f96 0x5578e09190db 0x5578e09125eb 0x5578e0970a3a 0x5578e08b4aa4 0x5578e0a23603 0x5578e066ed64 0x7f3426845151 0x5578e066d58c 0xfffffffffffffffe
******************************************************************
```

closed 2020-01-30 by tgingold

pedrorivera, 2021-11-08:
Hi. With much later GHDL (1.0.0.r856.gfbd853c8), I'm attempting to run synth of an entity making use or `ror` and I get:
```
MixG.vhd:65:47:error: unhandled predefined IEEE operator "ror"
            D <= (D_in xor (A_in + B_in + X)) ror 32;
                                              ^
../../src/ieee/v93/numeric_std.vhdl:663:12:error:  declared here
  function "ror" (ARG: UNSIGNED; COUNT: INTEGER) return UNSIGNED; --!V87
```
Is this there any relation between this issue and the error I'm getting?

umarcor, 2021-11-08:
@pedrorivera, please create a new issue and provide a reproducible example.

commit ce7b33d5 (2020-01-30): synth: handle some rotation and shifts.  Fix #1077