ghdl/ghdl #1087: synth: re-importing synthesized VHDL hits "synchronous code does not expect else part" error

antonblanchard, 2020-01-12:
Thanks for fixing the last set of bugs so quickly! I'm now able to synthesize the entire core of Microwatt. Next step is to reimport it into GHDL so we can simulate the synthesized VHDL.

I'm hitting a couple of issues in the icache at this stage:

```
git clone https://github.com/antonblanchard/microwatt
cd microwatt
make
ghdl  -a --std=08 icache.vhdl
ghdl  --synth --std=08 icache > icache2.vhdl
ghdl  -a --std=08 icache2.vhdl
ghdl  --synth --std=08 icache
```

Gives:

```
icache2.vhdl:3668:3:error: synchronous code does not expect else part
...
icache2.vhdl:121:48:error: synth_static_dyadic_predefined: unhandled IIR_PREDEFINED_IEEE_NUMERIC_STD_SUB_UNS_UNS
```

eine, 2020-01-12:
FTR, I'll label this kind of "bugs from feature looping" as `FeaReq: Synthesis`, even though the error is produced after executing synthesis.

tgingold, 2020-01-13:
I will fix the issues, but why do you want to *synthesize* the result.  In your initial message, you were speaking about simulating the output (which will be very slow, but I understand the purpose of it).

tgingold, 2020-01-13:
The first error is now almost a duplicate of #1069.

antonblanchard, 2020-01-13:
@tgingold good point, my focus indeed is to reimport and simulate. I thought it might be interesting to do another loop through ghdl and see if the output is stable, but I'll stick to simulation.

tgingold, 2020-01-13:
I am not sure you can have a stable output, as the names should be kept.

This was an interesting issue, but maybe more for me than for you.

antonblanchard, 2020-01-19:
Thanks @tgingold, I don't think there is anything left to do here so I'll close it.

closed 2020-01-19 by antonblanchard