ghdl/ghdl #1088: Machine epsilon for string to real conversion

eine, 2020-01-13:
According to https://en.wikipedia.org/wiki/Machine_epsilon#Values_for_standard_hardware_floating_point_arithmetics, the machine epsilon for 64 bit floats/doubles/reals is `2−52 ≈ 2.22e-16`. However, when `real'value` is used to convert a string to a real, the error seems to be twice as much.

```vhdl :file:tb.vhd
entity tb is
end;

architecture arch of tb is
begin
  process
    constant s : string := "3.33";
  begin
    report real'image(2.0**(-52));
    report real'image(3.33-real'value(s));
    wait;
  end process;
end;
```

```sh :image: ghdl/vunit:mcode ghdl/vunit:llvm
ghdl -a --std=08 tb.vhd
ghdl -e --std=08 tb
ghdl -r --std=08 tb
```

```
tb.vhd:9:5:@0ms:(report note): 2.220446049250313e-16
tb.vhd:10:5:@0ms:(report note): 4.440892098500626e-16
```

tgingold, 2020-01-13:
I think 'value is not fully precise.

eine, 2020-01-13:
Yes, and I'm ok with that, as it is expected. I'm trying to understand which is the theoretical maximum bound of the absolute error in the conversion. So far, I think it is `x ( 1 + e )`, where `e = 2.0**(-52)`. Therefore the bound is `~7.4e-16` and `4.441e-16` is ok.

However, I wonder whether a tighter theorecital bound exists, given that the representation is binary. E.g. `2**(round(log2(x))-1) * 2.0**(-52)` = `2**1 * 2.0**(-52)`. Note that this is *exactly* the error I get. Unfortunately, I don't know if this is true for any number, or it is just a coincidende with `3.33`.

tgingold, 2020-01-15:
Yes there is a theoretical bound, but the machine value should be the same for the same input.