ghdl/ghdl #1089: Interface to GHDL like verilator DPI/VPI possible?

BracketMaster, 2020-01-13:
I'm wondering if I can control simulation of the executable that GHDL generates in an external C program.

I know that you can start GHDL simulation by calling ghdl_main() as mentioned [here](https://ghdl.readthedocs.io/en/latest/using/Foreign.html#wrapping-and-starting-a-ghdl-simulation-from-a-foreign-program).

Verilator allows you to set input values and then call model.eval() in C++. Does GHDL offer something like that?

tgingold, 2020-01-13:
That's possible using VPI, but it's more tedious than the verilator interface.  You'd better to use a library like cocotb.

OTOH, I am planning to do a vhdl-ator: injecting the result of a synthesis to verilator (without generating intermediate verilog code).

eine, 2020-01-13:
Ref #1053

BracketMaster, 2020-01-13:
Cocotb actually handles this quite well!
Thanks!
Yehowshua

closed 2020-01-13 by tgingold