ghdl/ghdl #1091: synth: Another case of synchronous memory reads becoming asynchronous

antonblanchard, 2020-01-13:
I tried to run the Microwatt core through yosys, and I still see an issue with the icache not ending up in BRAM:

```
git clone https://github.com/antonblanchard/microwatt
cd microwatt
for i in decode_types common wishbone_types fetch1 fetch2 utils plru cache_ram icache; do ghdl -a --std=08 $i.vhdl; done
ghdl  --synth --std=08 icache
```

Which shows:

```
    n931_data <= n930_ports(to_integer (unsigned (n716_o)));
    n932_data <= n930_ports(to_integer (unsigned (n697_o)));
    n933_data <= n930_ports(to_integer (unsigned (n678_o)));
    n934_data <= n930_ports(to_integer (unsigned (n660_o)));
    n935_data <= n930_ports(to_integer (unsigned (n526_o)));
    n936_data <= n930_ports(to_integer (unsigned (n501_o)));
    n937_data <= n930_ports(to_integer (unsigned (n476_o)));
    n938_data <= n930_ports(to_integer (unsigned (n451_o)));
    if rising_edge(wrap_clk) and (n929_o = '1') then
      n930_ports (to_integer (unsigned (n669_o))) := n667_o;
    end if;
    if rising_edge(wrap_clk) and (n928_o = '1') then
      n930_ports (to_integer (unsigned (n688_o))) := n686_o;
    end if;
    if rising_edge(wrap_clk) and (n927_o = '1') then
      n930_ports (to_integer (unsigned (n707_o))) := n705_o;
    end if;
    if rising_edge(wrap_clk) and (n926_o = '1') then
      n930_ports (to_integer (unsigned (n725_o))) := n723_o;
    end if;
```

tgingold, 2020-01-13:
This is the memory for cache_valids.  I am not sure it could end up in BRAM.
(Ok, the traceability could be improved...)

I can investigate further, but this doesn't start well.

antonblanchard, 2020-01-13:
Sorry @tgingold the cache valids are indeed not meant to be in BRAM. Let me close this and reinvestigate why yosys is not synthesizing BRAM for the cache itself.

closed 2020-01-13 by antonblanchard