ghdl/ghdl #1092: bound check failure at ../../src/ieee/numeric_std-body.v93:2104

miguel9554, 2020-01-14:
**Description**
I'm getting the following error when running ghdl

`verification/tb_cordic/tb_cordic:error: bound check failure at ../../src/ieee/numeric_std-body.v93:2104
`

I'm running a testebench whose source files are located in `../../src`, I guess that's why that route is appearing. However, I have no `ieee` directory in `src`, I really can't understand from the error message where is the problem. Is there a way to get a more precise error message?

**Expected behaviour**
I would expect to get an error that points to the file and line causing the error. 

**How to reproduce?**
Since I don't know what's causing the problem, I can't provide a MWE. However, the error can be reproduced by cloning the [repo](https://github.com/miguel9554/sdigitales-tfinal), running the `generate_stimulus` script and then running `make tb_cordic`.

tgingold, 2020-01-14:
The file is for ieee.numeric_std

I will have a quick look.

tgingold, 2020-01-14:
I really need a reproducer because I cannot reproduce the bound check failure.

I had to remove the first line of stimulus.dat.  Is that expected ?
Then when I run tb_cordic, I get many assertion failures like:

verification/tb_cordic/tb_cordic.vhdl:97:13:@0ms:(report note): Reading line:  -231   386    52    -446.292000      56.469600    -446.389952      55.614845    0.022%    1.537%
verification/tb_cordic/tb_cordic.vhdl:160:13:@50ns:(assertion error): ERROR (X): expected -446, got -382
verification/tb_cordic/tb_cordic.vhdl:165:13:@50ns:(assertion error): ERROR (Y): expected 292000, got 637

and at the end:

verification/tb_cordic/tb_cordic.vhdl:186:9:@50050ns:(report note): Total lines processed: 1000
verification/tb_cordic/tb_cordic.vhdl:187:9:@50050ns:(report note): X errors: 998
verification/tb_cordic/tb_cordic.vhdl:188:9:@50050ns:(report note): Y errors: 1000
verification/tb_cordic/tb_cordic.vhdl:192:9:@50100ns:(assertion failure): Fin de la simulacion

miguel9554, 2020-01-14:
I'm really sorry, I had not pushed the broken version to the repo. The first line should not be giving problems if you are running `make tb_cordic`, and the assertion errors are the way of reporting errors in the testbench.

I've pushed the problematic version to the repo, If you clone/pull it should reproduce the error.

diogratia, 2020-01-14:
With a relatively recent version  of ghdl (Nov 29, 2019) your error does not occur:

```none
digoratia@Macbook: ls
/Users/digoratia/Desktop/lubuntu shared folder/sdigitales-tfinal-master/verification
tb_addsub/         tb_cordic/         tb_cordic_stage/   tb_signed_shifter/
digoratia@Macbook: cd ..
digoratia@Macbook: ls
/Users/digoratia/Desktop/lubuntu shared folder/sdigitales-tfinal-master
Makefile      README.md     src/          utils/        verification/
digoratia@Macbook: make tb_cordic
Generating WORK Library for verification/tb_cordic/work-obj93.cf
analyze tb_cordic.vhdl
analyze ../.././src/cordic.vhdl
analyze ../.././src/cordic_stage.vhdl
analyze ../.././src/signed_shifter.vhdl
analyze ../.././src/addsub.vhdl
elaborate tb_cordic
verification/tb_cordic/tb_cordic --fst=verification/tb_cordic/waves.fst
tb_cordic.vhdl:97:13:@0ms:(report note): Reading line: -231 386 393216 -444 595 -270.0825446513854 359.73937659732553
tb_cordic.vhdl:97:13:@50ns:(report note): Reading line: 477 486 1048576 532 987 324.5620740355165 598.6522029507315
tb_cordic.vhdl:97:13:@100ns:(report note): Reading line: -64 402 655360 -220 632 -132.8342636148873 384.7792333402241
tb_cordic.vhdl:97:13:@150ns:(report note): Reading line: 270 -85 -458752 428 -192 257.6285667537194 -117.27114560890217
tb_cordic.vhdl:97:13:@200ns:(report note): Reading line: -398 503 1638400 -944 471 -573.2874848961585 287.6707487066366
tb_cordic.vhdl:97:13:@250ns:(report note): Reading line: -326 184 327680 -564 257 -340.79612824347817 154.8870523131446
tb_cordic.vhdl:97:13:@300ns:(report note): Reading line: 337 47 -917504 558 -59 338.3599888481952 -35.9237796821162
tb_cordic.vhdl:97:13:@350ns:(report note): Reading line: -314 22 196608 -517 9 -314.72106495028095 5.536359504316263
tb_cordic.vhdl:97:13:@400ns:(report note): Reading line: 384 -437 1048576 804 -516 489.5780157323431 -314.2266164913177
tb_cordic.vhdl:97:13:@450ns:(report note): Reading line: -487 229 131072 -813 352 -494.6953175031723 211.86444449325495
 -494.6953175031723 211.86444449325495
################################
#                              #
#  ++====    ++  ++    ++=\\   #
#  ||        ||\\||    ||  \\  #
#  ++===     ++ \++    ++  ||  #
#  ||        ||  ||    ||  //  #
#  ++====    ++  ++    ++=//   #
#                              #
################################

tb_cordic.vhdl:186:9:@550ns:(report note): Total lines processed: 10
tb_cordic.vhdl:187:9:@550ns:(report note): X errors: 0
tb_cordic.vhdl:188:9:@550ns:(report note): Y errors: 0
tb_cordic.vhdl:192:9:@600ns:(assertion failure): Fin de la simulacion
verification/tb_cordic/tb_cordic:error: assertion failed
verification/tb_cordic/tb_cordic:error: simulation failed
make: *** [verification/tb_cordic/waves.fst] Error 1
digoratia@Macbook: ghdl --version
GHDL 0.37-dev (v0.36-892-ga462aa2) [Dunoon edition]
 Compiled with GNAT Version: GPL 2017 (20170515-63)
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
digoratia@Macbook: uname -a
Darwin Diogratia-MacBook.local 15.6.0 Darwin Kernel Version 15.6.0: Thu Jun 21 20:07:40 PDT 2018; root:xnu-3248.73.11~1/RELEASE_X86_64 x86_64
```

Perhaps you could provide information on the platform operating system and ghdl version?

(I also tried this on a pre ghdl-0.36 release  (v0.35-271-g5cd3de46) running on ubuntu 18.04 under VirtualBox with the same results.)

miguel9554, 2020-01-14:
Did you pull the latest commit from the repo? If so, please run again the script `generate_stimulus` (or just the script `utils\stimulus_generation\cordic.py`), as the data files are not included in the repo so they are not updated. The output I'm getting is the following

```
[miguel@localhost sdigitales-tfinal]$ make tb_cordic
Generating WORK Library for verification/tb_cordic/work-obj93.cf
analyze tb_cordic.vhdl
analyze ../.././src/cordic.vhdl
analyze ../.././src/cordic_stage.vhdl
analyze ../.././src/signed_shifter.vhdl
analyze ../.././src/addsub.vhdl
elaborate tb_cordic
verification/tb_cordic/tb_cordic --fst=verification/tb_cordic/waves.fst 
tb_cordic.vhdl:97:13:@0ms:(report note): Reading line: 590463589 2388303040 1900544 -1056347309 3911239318 -641441198.5679159 2375119333.7573075
verification/tb_cordic/tb_cordic:error: bound check failure at ../../src/ieee/numeric_std-body.v93:2104
verification/tb_cordic/tb_cordic:error: simulation failed
make: *** [Makefile:53: verification/tb_cordic/waves.fst] Error 1
```
I'm running Fedora 31 witht the following ghdl version

```
[miguel@localhost sdigitales-tfinal]$ ghdl --version
GHDL 0.37-dev (tarball) [Dunoon edition]
 Compiled with GNAT Version: 9.1.1 20190605 (Red Hat 9.1.1-2)
 GCC back-end code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2019 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

eine, 2020-01-15:
@miguel9554, do you know how was GHDL installed on that machine?

miguel9554, 2020-01-15:
@eine with DNF

eine, 2020-01-15:
Thanks. Then, according to https://src.fedoraproject.org/rpms/ghdl/commits/f31, it is likely to be https://src.fedoraproject.org/rpms/ghdl/blob/c83508e57c6c1785d6b70df181d09ce8c2e112e0/f/sources. It is unfortunate that `git` seems not to be available in Fedora's build procedure, so `(tarball)` is shown instead of a meaningful identifier. Ref #888.

eine, 2020-01-15:
I can reproduce the issue in a `fedora:31` docker container:

```sh
docker run --rm -it fedora:31 bash -c "
dnf install -y ghdl git python make
git clone https://github.com/miguel9554/sdigitales-tfinal
cd sdigitales-tfinal
cd utils
./generate_stimulus
cd ..
make tb_cordic
"
```

```
Generating WORK Library for verification/tb_cordic/work-obj93.cf
analyze tb_cordic.vhdl
analyze ../.././src/cordic.vhdl
analyze ../.././src/cordic_stage.vhdl
analyze ../.././src/signed_shifter.vhdl
analyze ../.././src/addsub.vhdl
elaborate tb_cordic
verification/tb_cordic/tb_cordic --fst=verification/tb_cordic/waves.fst
tb_cordic.vhdl:97:13:@0ms:(report note): Reading line: 590463589 2388303040 1900544 -1056347309 3911239318
verification/tb_cordic/tb_cordic:error: bound check failure at ../../src/ieee/numeric_std-body.v93:2104
verification/tb_cordic/tb_cordic:error: simulation failed
make: *** [Makefile:53: verification/tb_cordic/waves.fst] Error 1
```

I can reproduce it with `ghdl/ghdl:fedora30-gcc-8.2.0` too, which includes GHDL built from sources 5 days ago:

```sh
docker run --rm -it ghdl/ghdl:fedora30-gcc-8.2.0 bash -c "
dnf install -y git python3 make
ln -s /usr/bin/python3 /usr/bin/python
git clone https://github.com/miguel9554/sdigitales-tfinal
cd sdigitales-tfinal
cd utils
./generate_stimulus
cd ..
make tb_cordic
"
```

```
Generating WORK Library for verification/tb_cordic/work-obj93.cf
analyze tb_cordic.vhdl
analyze ../.././src/cordic.vhdl
analyze ../.././src/cordic_stage.vhdl
analyze ../.././src/signed_shifter.vhdl
analyze ../.././src/addsub.vhdl
elaborate tb_cordic
verification/tb_cordic/tb_cordic --fst=verification/tb_cordic/waves.fst
tb_cordic.vhdl:97:13:@0ms:(report note): Reading line: 590463589 2388303040 1900544 -1056347309 3911239318
verification/tb_cordic/tb_cordic:error: bound check failure at ../../src/ieee/v93/numeric_std-body.vhdl:2104
verification/tb_cordic/tb_cordic:error: simulation failed
make: *** [Makefile:53: verification/tb_cordic/waves.fst] Error 1
```

tgingold, 2020-01-15:
You are trying to convert one of this numbers (like the first one):
```
590463589 2388303040 1900544 -1056347309 3911239318
```
to a natural (using numeric_std.to_integer) and it fails because the number is larger than the highest natural value.

eine, 2020-01-15:
@tgingold, is it possible for GHDL to report where he is doing so, instead of producing the failure in the IEEE lib? Or is this something the user is expected to investigate (with backtrace)?

miguel9554, 2020-01-15:
I suspected that was the problem, but shouldn't the error refer to the line in which the number is trying to be converted? I'm getting this error in other modules and it's hard to find the problem when the error doesen't refer to a particular line (or even file)

tgingold, 2020-01-15:
You can get a better output if you enable the backtraces, or if you use gdb:
```
$ gdb ./verification/tb_cordic/tb_cordic 
(gdb) break __ghdl_fatal
(gdb) run
...
(gdb) bt
```
(Don't forget to build with -g)

closed 2020-01-20 by tgingold