ghdl/ghdl #1097: Overflow in osvvm.CoveragePkg

eschmidscs, 2020-01-15:
**Description**
When covering a large amount of items, an integer overflows in CoveragePkg. This does not happen with QuestaSim.
I've filed a bug report for osvvm first, see https://github.com/OSVVM/OSVVM/issues/43. I'm not sure which side should try to fix this - please check my latest comment of the mentioned issue.
If GHDL always produces an error if an integer overflows (as opposed to what a c program would do), a switch to change this to a silent wrap could solve to problem (then this would rather be a feature request).

**Expected behaviour**
The coverage should run much further. It will obviously have issues to cover more than 2**31 items, but currently it fails at 130606.

**How to reproduce?**
Run the testbench below. As it is given, it requires vunit.

```vhd :file: tb_osvvm.vhd
library ieee;
context ieee.ieee_std_context;

library vunit_lib;
context vunit_lib.vunit_context;

library osvvm;
use osvvm.RandomPkg.all;
use osvvm.CoveragePkg.all;

entity tb_osvvm is
  generic (
    runner_cfg    : string  := runner_cfg_default
    );
end entity tb_osvvm;

architecture tb of tb_osvvm is

  shared variable cov_fill   : CovPType;
begin
  ------------------------------------------------------------------------------------------------------------------------
  -- Test process
  ------------------------------------------------------------------------------------------------------------------------
  p_main : process
    variable rnd      : RandomPType;
  begin
    test_runner_setup(runner, runner_cfg);
    show_all(display_handler);

    while test_suite loop
      if run("random") then
        cov_fill.AddBins(GenBin(0, 3));
        while true loop
          cov_fill.ICover(rnd.RandInt(0, 3));
          cov_fill.WriteBin;
        end loop;
      end if;
    end loop;

    info(to_string(get_checker_stat));
    test_runner_cleanup(runner);
  end process p_main;

end architecture tb;
```

I get the following output (last part only):
```
...
%% WriteBin:
%% Bin:(0)   Count = 32326  AtLeast = 1
%% Bin:(1)   Count = 32691  AtLeast = 1
%% Bin:(2)   Count = 32532  AtLeast = 1
%% Bin:(3)   Count = 33056  AtLeast = 1

%% WriteBin:
%% Bin:(0)   Count = 32326  AtLeast = 1
%% Bin:(1)   Count = 32691  AtLeast = 1
%% Bin:(2)   Count = 32533  AtLeast = 1
%% Bin:(3)   Count = 33056  AtLeast = 1

/usr/local/bin/ghdl:error: overflow detected
  from: osvvm.coveragepkg.covptype.icoverindex at CoveragePkg.vhd:2072
/usr/local/bin/ghdl:error: simulation failed
fail (P=0 S=0 F=1 T=1) zf_radar.tb_osvvm.random (83.8 seconds)

==== Summary ====================================
fail zf_radar.tb_osvvm.random (83.8 seconds)
=================================================
pass 0 of 1
fail 1 of 1
=================================================
Total time was 83.8 seconds
Elapsed time was 83.8 seconds
=================================================
Some failed!
```

**Context**
Please, provide the following information:

- OS: Ubuntu 16.04 and 18.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `v0.36`

tgingold, 2020-01-15:
When count is 33056, you have for Index 4:
ordercount = 130607
CovBinPtr(Index).OrderCount = 2147414894

The sum of those results in an overflow.

tgingold, 2020-01-15:
(and no, the behaviour of an overflow in C for a signed integer is undefined behaviour)

eschmidscs, 2020-01-17:
Yes, I expected something similar to what you mentioned, but didn't have the time to follow that thought. Jim has made a change on his side by now.

And yes, it might be better I keep with HDL and don't do too much C and stuff... This has some pattern that I think something behaves a certain way in C and then it is actually undefined...

closed 2020-01-17 by eschmidscs