ghdl/ghdl #1108: 2019 release

tgingold, 2020-01-21:
It's yearly time for a new release.  In no specific order, a few points:

* For simulation, the uvvm regression should be fixed.

* For synthesis: this is the new major feature.

* Maybe it is time to change the major: switch to either 1.0 or (better) 2019.0

* Ideally, the release should happen before end of February.

eine, 2020-01-21:
First off, 🎉 

> * Maybe it is time to change the major: switch to either 1.0 or (better) 2019.0

IMHO, using `year.month` (shouldn't it be 2020.1 or 2020.2) is interesting for projects with a fixed schedule (every n months), such as Docker or VSCode. However, for GHDL, I think it might be misleading. Precisely, it would make it *difficult* to release two versions in the same month, should we need to fix any critical bug. In the end, the date is implicitly defined in the commit SHA that is tagged. Therefore, we might instead just change the version string to show the year/month along with the "semver" version.

CraigNoble, 2020-01-21:
For what it's worth I've always liked the kernel's method of versioning where odd numbers introduce new features (and are therefore inherently less stable) and even numbers contain bugfixes and minor improvements.

eine, 2020-01-21:
@CraigNoble, we had a similar discussion two years ago: #514. I think it would be desirable to continue this discussion there, or to open a new issue, so we keep this for the next release.

@tgingold, does switching to 1.0 imply any kind of commitment to keep the API stable? If so, I think we should try to do a cleanup before releasing. Basically, to check that the content of `ghdl --help` is up to date, and that all the commands/options are, at least, mentioned in the docs. This is also related to tgingold/ghdlsynth-beta#74, as it might be merged/moved.

tgingold, 2020-01-22:
By using `2019.n`, I didn't imply that `n` was the month; just the release number in the year. 

The command line API is very stable; the internal API is not.

arcturus140, 2020-01-22:
it feels like a big jump from version 0 to 2019. I agree with @eine this format is more common for fixed schedules. I'd be expecting roughly x versions per year.

arm-in, 2020-01-22:
Why not?

Xilinx Vivado has a Year.Number scheme.
https://www.xilinx.com/support/download.html

FlightGear also has it
https://www.flightgear.org/

A minor bug fix release gets an additional revision number, e.g.
Vivado Design Suite 2019.1.2 
FlightGear 2019.1.1
=> No problem with two releases within one month.

eine, 2020-01-22:
Yes, having a `year.number` instead of `year.month`, as I thought, would work. Still, I think it should be `2020.0`, instead of `2019.0`. Changing the version naming scheme to use `2019` can be misleading precisely when VHDL 2019 has been published, and considering that we are in 2020. It makes little sense for us, but unexperienced users might think that it has something to do with the version of the standard.

@arcturus140, GHDL's versioning scheme has never been very consistent. Most of the development was done between `0.31` and `0.37-dev` for no special reason. I think it is already big jump to `1.0` without having released `0.37` and moved to `1.0-dev`. For almost a year, users have seeing `0.37-dev`. Hence, as long as the new scheme is consistent, I think it is ok to do it now rather than in a few versions.

@arm-in, note that the format of Vivado is not exactly `year.number`, it's `year.version.hotfix`. And I am not sure if `version == quarter`. This is not to say that I am not ok with it, but that they use an additional sublevel compared to what Tristan commented.

Paebbels, 2020-01-23:
GHDL - as it does now - should adhere to [semantic versioning](https://semver.org). These rules are good rules and they exist for software since decades.

Marketing changes every few years. We had a lot of `95` programs followed by `98` programs, `ME` programs, `XP` programs and now we have a period were people use a year. Using YEAR.n shows the inflexibility of companies being able to release more then 2...4 times a year (Xilinx, Aldec, ...). Currently software (in average) in updated (released) between 1 once a day and once a month. There are kinds of software, that do 10...30 full releases (including new features) per day.

With GHDL, almost every push done by @tgingold can be considered a pre-release.

Using YEAR.n creates a lot of problem e.g. in automated version and compatibility checking. More over, GHDL has currently a high ranked certificate from Linux Foundation beeing on a good way for a Gold Level. This certificate requests open source projects to follow semantic versioning rules.

You'll also see that lots of projects have switched explicitly to semantic versioning in the last years.

tgingold, 2020-01-23:
Sorry, current year is indeed 2020...

We can still use year and semantic versioning, particularly if we release once a year.

CraigNoble, 2020-01-23:
It is worth noting that the Linux kernel itself deviates from true semantic versioning by incrementing it's version irrespective of what the release contains if the current version number gets "too high". I'm not sure if they've formally documented this or if it just happens whenever Linus feels like it.

https://lkml.org/lkml/2011/5/29/204

arm-in, 2020-01-23:
@eine, you are right. Vivado ought to have `version == quarter`. But they could not keep the pace. There is 2018.3 and 2019.2 at the end.

arcturus140, 2020-01-23:
> Vivado ought to have version == quarter. But they could not keep the pace. There is 2018.3 and 2019.2 at the end.

That's not to say they won't continue to have 4 releases a year. They just don't make it to 4 on every year. 

Xilinx seems to use x.0 releases to introduce new features, like a GUI redesign or API changes. In xilinx forums user also imply that x.0 releases have more bugs than others.  This could be subjective. 

is ghdl planning to release at least 1 version per year?

eine, 2020-01-23:
> GHDL - as it does now - should adhere to [semantic versioning](https://semver.org). These rules are good rules and they exist for software since decades.

Note that GHDL has never adhered to semantic versioning strictly. On the one hand, there are only two numbers in GHDL's versions. This will probably crash a bunch of semver libs. On the other hand, bumping versions has never been related to the inclusion of some specific feature. Otherwise, we might probable be at `0.1000`.

Anyway, I agree that adopting semantic versioning *properly* would be good. Well, using three numbers at least, even if the last one is always `0`.

> With GHDL, almost every push done by @tgingold can be considered a pre-release.

Although I initially suggested @CraigNoble to discuss it in #514, I feel that multiple users in this thread are thinking not only about the format, but also about the schedule. As @arcturus140 explicitly worded, I think that those of us who would like to have say 4 releases a year should raise our hands. 

> We can still use year and semantic versioning, particularly if we release once a year.

What about the following format? Let's suppose that we release 0.37 tomorrow and 1.0 next month:

`GHDL 0.37 (2020.1)`
`GHDL 1.0.0 (2020.2)`

As commented above, the meaning would be `semver (year.month)`, where `year.month` are automatically extracted from the date of the tag. This would not work if we release two versions the same month. However, on the one hand, we are discussing 1-4 releases a year. On the other hand, it is always possible to add the day.

bkzshabbaz, 2020-01-25:
> > Vivado ought to have version == quarter. But they could not keep the pace. There is 2018.3 and 2019.2 at the end.
> 
> That's not to say they won't continue to have 4 releases a year. They just don't make it to 4 on every year.
> 
> is ghdl planning to release at least 1 version per year?

At XDF 2018 they talked about moving away from four releases a year to two.  Mostly because the community thought the odd number versions contained new features and were buggy.

Paebbels, 2020-01-25:
@eine  
> > GHDL - as it does now - should adhere to semantic versioning. These rules are good rules and they exist for software since decades.

> Note that GHDL has never adhered to semantic versioning strictly. On the one hand, there are only two numbers in GHDL's versions. This will probably crash a bunch of semver libs. On the other hand, bumping versions has never been related to the inclusion of some specific feature. 

Yes, GHDL didn't use semantic version very strictly, but it didn't failed the rules. Versions in `0.minor.patchlavel` state can change APIs while incrementing the minor number. This will be more strict when reaching a major number >= 1.

According to the rules, a major.minor number is equal to major.minor.0, no need to print the patchlevel.

-------------
@bkzshabbaz 
> At XDF 2018 they talked about moving away from four releases a year to two. Mostly because the community thought the odd number versions contained new features and were buggy.

But exactly, that is the case with many Vivado releases, they introduced new features in *.1 and *.3 versions - and there where mostly not fully tested :).

arcturus140, 2020-01-25:
@Paebbels Xilinx changed their api from 2019.1 to 2019.2. I think these observations mentioned in the forum are merely subjective, there are no real patterns until a xilinx employee officially declares those as such.

arcturus140, 2020-01-25:
> According to the rules, a major.minor number is equal to major.minor.0, no need to print the patchlevel.

this is also what I head in mind, but after re reading the specification it sais: a normal version number MUST take the form x.y.z. I couldn't make out any reference for a being optional / omitted.

eine, 2020-01-26:
@Paebbels, @arcturus140, I think that you are describing the *de facto* behaviour of many libraries/tools, even if it is not strict. Anyway, since the least significant number is probably not going to be used in GHDL, even if added, I think it is pointless to discuss it.

I believe that the relevant discussion is whether we are going to adhere to semver in the sense that we will bump to `2.0` as soon as a breaking change is introduced to `1.*`. If we are going to do so, we should use proper semver syntax. Otherwise, if we want to just follow the syntax but not the semantics, the `year.number` format is equally valid.

tgingold, 2020-02-03:
My last idea: next version will be 20.1

Paebbels, 2020-02-03:
Can't we have an internal version number like `v1.x.y` and a - let's say - marketing number like `2020.1` ?

tgingold, 2020-02-04:
Two versions would be too confusing.

What is wrong with year as a version ?

* It follows semantic versioning

* It is obvious how old your program is

Xiretza, 2020-02-04:
>* It follows semantic versioning

Does it though? Year versions would mean you'd have to make exactly one incompatible API change every year, no more, no less.

eine, 2020-02-04:
Actually, it does not follow semantic versioning neither semantically, as @Xiretza points out, nor syntactically, as it is missing the third number.

As said, I think that the "marketing number" should not be set manually, but derived from the date of the tag. Nevertheless, I don't think it is necessary for the tag to follow semver syntax if it is not properly adopted.

arcturus140, 2020-02-04:
I agree that nothing feels more bleeding edge than GHDL 2020 on any GUI installer, white paper or article. But using the year as a version number doesn't really make much sense. A year is a year and not a version. As a user this tends to derive any semantics from that version, like _the 1st version of the year being less stable than others_, or _this program is almost 10 years old now, I should look out for an alternative_. This is merely subjective, of course, but the root cause is the message that we transmit.

Are we releasing a new version because of a change in development or are we releasing a new version because the year has changed?

tgingold, 2020-02-05:
For the version: we could use 20.1.0 to strictly follow semantic versioning if needed.

I am not sure we need to follow semver, as GHDL is not an API.

And yes, we do one version per year (currently).

tgingold, 2020-02-05:
Maybe I will postpone this decision for next year (and then use 1.x)

tgingold, 2020-02-29:
Now released.

closed 2020-02-29 by tgingold

eine, 2020-03-01:
Regarding previous discussions about semver and the git workflow, currently tag `v0.37` is not a parent of `master`. As a result "commits to master since..." is not shown in https://github.com/ghdl/ghdl/releases/tag/v0.37. Proper semver would imply having a `v0.37.0` tag pointing to https://github.com/ghdl/ghdl/commit/fb423bdd0aec26b3e41937b2f22a3409eda87d5a and `v0.37.1` pointing to the current `v0.37`. Tag `v0.37` would be removed and a branch would be created instead.

tgingold, 2020-03-01:
The first builds failed and I fixed that after increasing the release.
So we have two choices:

* Let the release as is.  For the semver, that's not a problem as it wasn't used before.  For the parent of master I don't think it could be fixed.

* Do a v0.37.1 and delete v0.37

I don't have a strong opinion.

eine, 2020-03-01:
IMHO, we should do an effort to use semver. I think we can create both `v0.37.0` and `v0.37.1` without breaking anything. For `v0.37.0`, the "parent of master" would be shown. Not for `v0.37.1`, but I believe that's ok.

I'm not sure about deleting `v0.37`, because it is already published. We can keep it, and next time (year) use a branch for `v1.0` instead of a tag.

tgingold, 2020-03-01:
I have created v0.37.0, I will remove the corresponding release (which won't build).

danilofuchs, 2020-03-08:
In https://github.com/ghdl/ghdl-language-server/issues/53, we are discussing the language server dependency on GHDL, and how to manage compatibility between them. So indeed, GHDL can be considered an API, and the language server a consumer of this API, and thus semantic versioning would be beneficial.

tgingold, 2020-03-08:
I am not sure how this will be that beneficial.

If we change the version only on a release, fixes in the master branch won't be usable.
If we change the version when an incompatible change is made (which might not be that easy to know), the version will be incremented very often.

The problem is very similar to versioning of compilers like gcc or llvm.

eine, 2020-03-08:
@tgingold, I think that the context won't change for users of master and/or users of tagged releases only. More frequent and semver compatible releases are useful for users that want to find which of the +1000 commits between v0.36 and v0.37 is compatible with one of the +80 commits in ghdl/ghdl-language-server. Note that tags can be released *post-mortem*. Hence, whenever a change is detected in the core which breaks the ls, it is possible to fix it and, then, tag and release a release candidate. It is not necessary to think about incompatible changes before.

arcturus140, 2020-03-08:
> If we change the version when an incompatible change is made [...] the version will be incremented very often.

This is not unexpected on a WIP. Development matures over time and API changes are less frequent. 

> (which might not be that easy to know)

> It is not necessary to think about incompatible changes before.

it is, as this reflects deprecated functionality. I don't see how this would be a concern, though, as this is an intended change and not something that has been changed by mistake.

eine, 2020-03-08:
> > (which might not be that easy to know)
> 
> > It is not necessary to think about incompatible changes before.
> 
> it is, as this reflects deprecated functionality. I don't see how this would be a concern, though, as this is an intended change and not something that has been changed by mistake.

I'm trying to propose a minor enhancement in the release flow, which does not alter how Tristan works on the code. Ideally, breaking changes would be planned, announced for a couple of released, and then removed. However, that's not how GHDL is actually developed. Frequently, things break by mistake (and are fixed ASAP). More frequently, experimental (not yet complete) features are pushed as users demand them; these are then modified in non-compatible ways and/or introduce regressions. As a result, I agree with Tristan in that it can be a hard task to decide if each commit deserves a version bump. That's why I suggest a less strict and *post-mortem* approach. Actually, it can be automated so that release candidates are pushed by a GitHub Action, only after a certain set of tests pass.

eine, 2020-03-08:
@arcturus140, R3 in the diagram below shows what I mean. The flow is read bottom to top. It shows how C2 and C3 can be pushed to master, which might include breaking changes. It is possible to create and release R3, even after those commits were pushed to master. Moreover, tag v0.3.0-rc.0 in this example would allow users to benefit from features added in C1; i.e. between v0.2 (v0.2.0) and v0.3 (v0.3.0). In the case of GHDL, C1 can be, in practice, +1000 commits.

@tgingold, note that the naming can be simplified for GHDL, because the last number in the semver can be used instead of the `-rc.0` suffix.

[image]