ghdl/ghdl #1110: synth: crash in output

Xiretza, 2020-01-21:
**Description**
When a signal uses the name of another signal as the initialization value, this value is not resolved to the constant initialization value of the named signal (or lack thereof), causing cascading failures in both netlist display and ghdlsynth-beta.

**Expected behaviour**
I'm actually not entirely sure how legal this is from an LRM perspective, but synthesis should definitely not crash either way.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
	port (
		clk : in std_logic;
		i : in std_logic_vector(7 downto 0);
		o : out std_logic_vector(7 downto 0)
	);
end;

architecture a of ent is
	signal a : std_logic_vector(7 downto 0) := x"42";
	signal b : std_logic_vector(7 downto 0) := a;
begin
	process(clk)
	begin
		if rising_edge(clk) then
			a <= i;
			b <= a;
		end if;
	end process;

	o <= b;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: caed711f

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
library ieee;
use ieee.std_logic_1164.all;
entity ent is
  port (
    clk: in std_logic;
    i: in std_logic_vector (7 downto 0);
    o: out std_logic_vector (7 downto 0)
  );
end;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of ent is
  signal wrap_clk: std_logic;
  signal wrap_i: std_logic_vector (7 downto 0);
  signal wrap_o: std_logic_vector (7 downto 0);
  signal a : std_logic_vector (7 downto 0);
  signal b : std_logic_vector (7 downto 0);
  signal n9_q : std_logic_vector (7 downto 0) := 
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1525-gcaed711f) [Dunoon edition]
Compiled with GNAT Version: 9.2.0
Target: x86_64-pc-linux-gnu
/home/xiretza/temp/ghdl_synth_tests/
Command line:
ghdl --synth ent.vhd -e ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-disp_vhdl.adb:361
Call stack traceback locations:
0x55d4511b788b 0x55d4511bce28 0x55d451340ed8 0x55d451369953 0x55d45136a2eb 0x55d4512aceda 0x55d45141ce59 0x55d451065d64 0x7f9eac201151 0x55d45106458c 0xfffffffffffffffe
******************************************************************
```

tgingold/ghdlsynth-beta@bfb4f21519374e8ffb88813722a2d26308ec685c crashes with the following message, probably caused by [ghdl.cc:695](https://github.com/tgingold/ghdlsynth-beta/blob/175123cda990ee2b5cfac461bd8ec44956da302a/src/ghdl.cc#L695):

```
1. Executing GHDL.
Importing module ent.
ERROR: Assert `is_fully_const() && GetSize(chunks_) <= 1' failed in kernel/rtlil.cc:3781.
```

tgingold, 2020-01-23:
Isn't it a real corner case ?

According to the LRM, you aren't suppose to use a signal in an expression for the initial value of a signal (see LRM08 14.4.1 note 1 - ok, not detected by ghdl).

Xiretza, 2020-01-27:
Reads like it to me too, in that case it should probably error during analysis.

dalex78, 2026-08-19:
# Issue #1110 — Crash: signal init value references another signal

Analysis by an AI agent. Recommend (optionally merging the test and) closing the ticket.

## High-level summary

`signal b : std_logic_vector(7 downto 0) := a;` (initializing one signal with another signal's name, `a` itself initialized to `x"42"`) crashed `ghdl --synth`'s netlist display with an internal `TYPES.INTERNAL_ERROR` at `netlists-disp_vhdl.adb:361` instead of resolving `b`'s initial value to `a`'s constant initial value (or lack thereof).

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`: `ghdl --synth ent.vhd -e ent` now produces a correct netlist where both `a`'s and `b`'s initial-value drivers are resolved to the constant `"01000010"` (`x"42"`):

```vhdl
signal n9 : std_logic_vector (7 downto 0) := "01000010";
signal n10 : std_logic_vector (7 downto 0) := "01000010";
```

instead of crashing. This must have been fixed as part of general improvements to init-value constant folding in `src/synth` since this was reported (originally against a 2019 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1110/` (the exact repro from the report) whose `testsuite.sh` synthesizes the design and asserts the constant `x"42"` init value appears resolved twice (once for `a`, once for `b`), with no `GHDL Bug occurred` crash. This locks in the current, correct behavior as a regression guard.

closed 2026-08-20 by tgingold