ghdl/ghdl #1113: synth: Using a signal as the index for an array of length 1 fails with an assertion error.

benreynwar, 2020-01-23:
**Description**
When accessing an element of an array of length one, using an address of length one as the index an assertion fails.  This assertion does not fail in other instances where the address signal is too large.  For example an array with 2 element can be accessed with an address of length two without problems.

**Expected behaviour**
I would expect this to synthesize without error.  Ideally the synthesized result would ignore the address and always output the first element of the array.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: memory_depth_one.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity memory_depth_one is
  port (
    address: in unsigned(0 downto 0);
    output: out std_logic
    );
end entity;

architecture arch of memory_depth_one is
  constant store: std_logic_vector(0 downto 0) := "0";
begin
  output <= store(to_integer(address));
end arch;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a memory_depth_one.vhd
yosys -m ghdl -p 'ghdl memory_depth_one'
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ x] Built from sources: `2fd7ad78`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1529-g2fd7ad78) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/home/ben/debug_ghdl_synth/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:1072
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tmeissner, 2020-01-23:
Am I wrong or is the `address` port too large, even in this example. The only valid value of address is `"0"`. If it's `"1"`, the array index is out of range.

benreynwar, 2020-01-23:
Yep.  I assume that the output when the address is 1 is undefined, so the synthesizer is free to do whatever it wants.  Just ignoring the address seems like the sensible solution.

I tested other designs where the address port was too large, e.g. an address of width 2 for a memory of depth 2, and synthesis was completed successfully.

The only advantage of this construction is for writing memories with a generic depth parameter without having to handle the special case of depth=1 separately.

commit c2168c67 (2020-01-25): synth: improve support of 0-width nets and gates.  Fix #1113
closed 2020-01-25 by tgingold