ghdl/ghdl #1115: macOS binary is unusable under the current version of the operating system

misterakko, 2020-01-25:
**Description**
The downloadable binary (pre-built release) at https://github.com/ghdl/ghdl/releases cannot be used on updated Mac computers.

**Expected behaviour**
./bin/ghdl should be an executable. But since it's not signed, the OS refuses to consider it so.
Issue can be easily fixed by distributing as a DMG file instead of a TGZ file, and digitally signing/notarizing  the file with the free tool inside the Xcode IDE. It's not necessary to register with Apple, nor pay any fee.

**How to reproduce?**
E.g. if you try
./ghdl  -a hello.vhdl
You will get a high level alert from the OS saying that the binary is unsigned and unnotarized, and could contain malware. The OS then offers to throw the GDHL executable in the trash.
Execution is stupid with the error message
./ghdl:error: exec error

**Context**
macOS 10.15 "Catalina"
See also https://stackoverflow.com/questions/39811791/mac-os-gatekeeper-blocking-signed-command-line-tool

tgingold, 2020-01-25:
Humm, is it just a matter of using hdiutil ?

eine, 2020-01-26:
FTR, currently, in the CI jobs macOS builds use the same script as GNU/Linux builds. There are some differences, but the prodecure is to configure (mcode), make, make install, generate a tarball and run the test suite. No specific macOS tool is used at all. Note that mcode backend is tested only.

Moreover, the macOS tarballs (mcode and llvm) available at [releases](https://github.com/ghdl/ghdl/releases) are contributed manually by @diogratia; unlike GNU/Linux and Windows tarballs, which are uploaded automatically from the CI jobs.

As a result, in the current status, we need @diogratia to voluntarily decide that he/she wants to investigate how to produce DMG files instead of tarballs.

Alternatively, the current CI jobs might be fixed and extended so that DMG files are generated and uploaded automatically, instead of requiring manual builds. See:

- #867: about enabling libghdl on macOS jobs. Users have reported that it works for them, so it must be some issue with the CI setup.
- https://github.com/eine/ghdl/commits/osx-hdiutil: an attempt to use hdiutil to extract the installer of GNAT, instead of using 7z installed through brew (which we are doing ATM).

Resources to reproduce/investigate:

- This is the CI job in GitHub Actions: https://github.com/ghdl/ghdl/blob/master/.github/workflows/push.yml#L29-L42.
  - As you see, GNAT is installed with a shell script: https://github.com/ghdl/ghdl/blob/master/dist/macosx/install-ada.sh. However, locally the DMG file can be installed manually.
- The relevant parts of the CI build script are:
  - Actually build GHDL: https://github.com/ghdl/ghdl/blob/master/dist/ci-run.sh#L402-L405 (note the CC=clang)
  - Then, run the test suite: https://github.com/ghdl/ghdl/blob/master/dist/ci-run.sh#L434-L436
- An example log of a job run: https://github.com/ghdl/ghdl/runs/390472921. As seen in https://github.com/ghdl/ghdl/runs/390472921#step:4:26 the command that is actually executed is `./dist/ci-run.sh -c -b mcode -p PACKAGE_NAME build`. This can be used to reproduce the build locally.
- Latest issues related to building GHDL on mac: #949, #959. Fortunatelly, @anquegi could have it done in the end.

/cc @BracketMaster @tmeissner

diogratia, 2020-01-28:
I reiterate that I have no capability of investigating Catalina problems other than anecdotally, not possessing a Mac capable of running MacOS 10.12 - 10.15. That extends to generating disk images or packages which include code signing and get notarized. My only experience with `codesign` has been to enable gdb to turn off memory page randomizing while debugging.

The required solution can apparently depend on the actual popup error message. There is [anecdotal evidence](https://apple.stackexchange.com/questions/367364/cant-execute-binary-on-macos-catalina) the August 2019 pre-release of Catalina could have the "can't be opened because its integrity cannot be verified" for a binary executable with the offer to move to the trash popup defeated by removing the com.apple.quarantine extended attribute.

This attribute is inherited, first upon download then through decorating the individual files by tools expanding archives.

The Catalina general release was October 7, 2018 there's no evidence the security features have been tightened. The underlying idea for Gatekeeper is to protect uninformed or inattentive users from executing  exploitative software (binaries, scripts, application bundles, etc.) individually downloaded, in disk images or packages. The idea doesn't (shouldn't) extend to scripts or executable binaries, application bundles, etc. produced by end users for their own use. It attacks the means of distribution and not the means of production. Gatekeeper isn't meant as a Soviet style 'everything that isn't permitted is prohibited' system.

Installer packages, disk images and other bundles  can still be opened and contents installed without notarization or code signing. Getting (a different) popup and opening Preferences => Privacy & Security, the general tab should allow the option to 'Open Anyway' resulting in another popup also available by right clicking 'open'. The entire mechanism for end consumer use is 'informed consent' as a product liability issue. The idea is how many times you are warned what you are doing is risky.

The extended attributes on individual executable files  can be deleted, e.g. `xattr -d com.apple.quarantine file_name`.  For coded signed apps this attribute is also used for inheritance for element files. You don't want to indiscriminately clear (`-c`) extended attributes, particularly recursively (`-r) across a file hierarchy containing bundles (applications, etc.).

[ghdl-0.36-macosx-llvm.tgz](https://github.com/ghdl/ghdl/releases) for example has no extended attributes (a 'resource fork' is contained in a separate file in the archive, they've been stripped out). The extracted archive contents can't be damaged by deleting or clearing extended attributes.

If removing operating system directed extended attributes doesn't work it'd be useful to show pictures of any popups, ls -l output, console output and extended attribute listings (`-l`) starting from opening the tgz archive.

BracketMaster, 2020-01-28:
Yes - I got your message - I'll take a look in March.
I'm a little busy with my Thesis and preparing for some conference workshops at the moments.

Best,
	Yehowshua

BracketMaster, 2020-01-28:
I have access to ample Macs in different configurations for testing out - so I'll take a look...

eine, 2020-01-28:
> Yes - I got your message - I'll take a look in March.
> I'm a little busy with my Thesis and preparing for some conference workshops at the moments.

It's ok. No need to rush at all. I just cced you so that you are informed as a mac user. Actually, this is the same content from gitter, with some more context.

BracketMaster, 2020-04-01:
Hello again, I meant to work on it last month, but have been doing school
stuff. Hopefully, my schedule clears up soon.

Yehowshua

eine, 2020-04-01:
No worries!

diogratia, 2021-02-03:
I can run existing binaries using Rosetta 2 on a Macbook Air M1 (MacOS 11.2 Big Sur).

System Preferences => Security & Privacy -> Privacy Button -> Developer tools 
```none
Allow the apps below to run software locally
that does not meet the system's security policy.
```
both Terminal (Apple's Terminal.app) and my preferred iTerm.app (iTerm 2) are checked. Big Sur will generally offer users with administrative rights pop-ups giving the option to go set permissions for various things or even installing Rosetta 2. You need at least the Xcode Command Line Tools to be installed.

eine, 2021-02-03:
@diogratia, that is probably because we are building the zipfiles (both mcode and LLVM) using `macos-latest`, which is macos-10.15 (https://github.com/actions/virtual-environments/#available-environments). I can add a couple of jobs for running the same scripts on `macos-11.0`. Could you please try them if I did so?

diogratia, 2021-02-03:
Sure, I'll give the MacOS 11 builds a try. For whatever reason I don't see notifications. I dragged and tagged a weblink to my Desktop.

eine, 2021-02-04:
Unfortunately, it seems that they disabled macos-11.0 temporaly. It works for orgs who had used it already, but new orgs/users are in a queue: https://github.com/actions/virtual-environments/issues/2486. Hence, tests fail: https://github.com/eine/ghdl/actions/runs/535662956. We'll need to wait until the feature is made public.

diogratia, 2021-02-04:
They may be getting ready to do a macos-11.2.

eine, 2021-02-04:
My understanding is that they have many more machines/licenses for running Linux/Windows than they have for macos. So, while it's cheap for them to offer free resources in those platforms, it is costly to do the same with macos. That's something that all CI services need to handle sooner or later, and that's why pricing is similar. I think that they didn't evaluate properly the demand that it would have (how many open source projects do want to test on macos).

diogratia, 2021-02-07:
A latest nightly binary for OSX ghdl-llvm won't run on my late 2008 aluminum Macbook (5, 1) running OSX 10.11.6:

```none
ghdl-1.0/ghdl-osx-llvm/bin/ghdl --version
GHDL 2.0.0-dev (1.0.0.r1.g40cf0da4) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm code generator
Written by Tristan Gingold.
...

ghdl -a level0.vhdl
dyld: lazy symbol binding failed: Symbol not found: __ZNSt3__118shared_timed_mutexC1Ev
  Referenced from: /usr/local/bin/ghdl1-llvm
  Expected in: /usr/lib/libc++.1.dylib

dyld: Symbol not found: __ZNSt3__118shared_timed_mutexC1Ev
  Referenced from: /usr/local/bin/ghdl1-llvm
  Expected in: /usr/lib/libc++.1.dylib

ghdl:error: exec error
```

What's happened here is ghdl1-llvm load command is not providing an rpath for resolution at run time and the name - /usr/local/opt/llvm/lib/libc++.1.dylib (presumably from the clang-llvm tool used in the build) doesn't exist. ld doesn't know where to find one and dithers around and finds one without the symbol.

I can build a ghdl-llvm locally and get a different name - @rpath/libc++.1.dylib,  which is going to provide a different search path rule (and will run on any compatible MacOS release - 10.11.5 through 11.2).

Attempting to change the  loader object name to match encounters a problem on MacOS 10.11.6:
```none
install_name_tool -change /usr/local/opt/llvm/lib/libc++.1.dylib @rpath/libc++.1.dylib ghdl1-llvm

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/u sr/bin/install_name_tool: object: ghdl1-llvm malformed object (unknown load command 10)
```

This error shows up with other tools, the command line tool `strings` for example. There's apparently a newer object format that the older system doesn't recognize:
```none
Load comnand 10
         cmd LC_BUILD_VERSION
     cmdsize 32
    platform 1
       minos 10.15
         sdk 10.15.2
      ntools 1
        tool 3
 version 609.8
```
while the matching object in the MacOS 10.11.6 ghdl-2.0dev build is
```none
Load command 9
      cmd LC_VERSION_MIN_MACOSX
  cmdsize 16
  version 10.11
      sdk 10.11
```
which does the same thing. We also see the actual issue appears to be what SDK the build was against - 10.15.2, limiting the target to maybe 15 percent of the Macs out there. Because the format is different an older platform doesn't inform the issue.

(These are extracted from otool -l bin/ghdl1-llvm for the respective builds.)

What we see is non-symmetrical forward and backward compatibility. This wants a build solution or multiple builds to not cut off the other 85 percent of eligible Macs. Patching loader objects is likely a bit too far out there, requiring an understanding of the ABI and really wanting it's own tool (if you can deal with the freed up space using the old object).

I'd be in favor of building on the [oldest MacOS practical](https://docs.travis-ci.com/user/reference/osx/#macos-version) to disenfranchise as few of potential Mac ghdl users as possible. In a year or so arm64 may make it big enough to add another platform.  Until then ghdl can't gain advantage of any MacOS newer features (other than Rosetta 2).

I'd also recommend potential ghdl users be informed of the limitations on the binary distribution.

tgingold, 2021-02-07:
We are currently building on github.
Yes, it would make sense to try to build on the oldest available macos version, but I fear this is already what we do.

The mcode variant should be easier to install.

eine, 2021-02-07:
> I'd be in favor of building on the [oldest MacOS practical](https://docs.travis-ci.com/user/reference/osx/#macos-version) to disenfranchise as few of potential Mac ghdl users as possible. In a year [...]

The main issue here is that Travis is no longer the company it used to be. They did an aggressive change in pricing a few months ago. Now, most open source projects run out of credits and maintainers are forced to writing e-mails for asking additional resources, each time. Therefore, although we could technically enable Travis again for macOS only, it's not going to be useful for keeping nightlies up to date. We would need to limit it to building tagged commits only.

Are you aware of any other CI service which provides free macOS machines for open source projects and which supports older versions?

As Tristan said, we are already using the oldest version available in GitHub Action. The only version available at the moment: 10.15.

> I'd also recommend potential ghdl users be informed of the limitations on the binary distribution.

Please, feel free to propose modifications to the docs. Forking and pushing to a feature branch will update `diogratia.github.io/ghdl` for you.

diogratia, 2021-02-08:
Putting information you know is subject to change into *the* documentation isn't ideal.

I was thinking more along the way of names, e.g.:
```none
ghdl-llvm-macos10.15-llvm-1.0.0.r2.g38025.tgz
```
Where you get the type of ghdl release, the minimum MacOS version required to to run it and where to find the source in the repository. Besides telling a potential consumer you must be at least this tall to ride they'd have enough information to organize installs/uninstalls of multiple ghdl versions or flavors.

eine, 2021-02-09:
Tarballs are updated automatically, and https://github.com/ghdl/setup-ghdl-ci uses fixed asset names for retrieving them and setting up CI. Therefore, having the specific version/sha in the tarball name is not an option. That info can be obtained through `ghdl version` or the recently added `ghdl version hash`. Nevertheless, I updated the workflow to include the specific macOS version. That's something we will need when more than one are tested.

With regard to documentation, I mean adding some note such as "pre-built tarballs will only work on the same version or later of macOS".

diogratia, 2021-06-24:
Recently the llvm version of GHDL for MacOS 10.15 builds switched from the static library integration to relying on a single dynamic library (libLLVM.dylib). The dynamic library is not API compatible across Hombrew llvm versions and discovering which  version is used requires reading through the Github Action build process and tracing the _current_ version used by [default](https://github.com/actions/virtual-environments/blob/main/images/macos/macos-10.15-Readme.md). Homebrew is the only available source for a binary distribution of this dynamic library without building llvm from scratch using nearly the same formula configuration and build process.

Using a nightly binary a ghdl end user must also be a developer with a willingness to use Homebrew which has various character flaws critical to end user adoption. It's .git overhead of around 600 MB and growing is installed on end user platforms replicating it's github repository. The x86_64 distribution assumes ownership control of /usr/local potentially displacing preexisting build environments with dependencies on required tools not found in /usr/bin (linked from either Xcode or Xcode Command Line Tools. The binary bottles for formulas can be large (the llvm bottle pours out 1.6 GB, compared to the 900 MB for a llvm.org binary distribution, here from having both a full set of static libraries as well as full set of dynamic libraries.

You could get the idea a Macbook Air using a limited local storage platform encouraging Cloud or external storage would 
be better off with either statically linked libLLVM libraries which requires ~26 MB versus the composite libLLVM.dylib at 88.6 MB were the llvm Homebrew binary distribution to be pared down and included  in a non-developer oriented ghdl binary distribution. It begs the question of whether ghdl caters only to tool developers or whether end user needs could be met at the same time.

Apple has come out with the M1 version of Big Sur MacOS 11, including the first Macbook Air M1. This requires another 'voyage of exploration' for the end user in discovering how to install x86_64 Homebrew and it's tool bottles from formulas. (The install script and brew are executed as an argument passed to Apple's arch command selecting target architecture for execution using Rosetta 2 with a -x86_64 command line argument or several variants, the purpose to manipulate a programs environment.)

Storage restrictions on small platforms can be overcome by external or Cloud storage but are stymied to some degree by libLLVM.dylib from Homebrew which references it's install path absolutely internally instead of using an rpath (it's likely possible to overcome this restriction by the use of libtool in lieu of building a binary distribution).  This again makes statically linked libLLVM libraries attractive or at least distributing the dynamic library. 

The underlying ideas presented here are that end user sophistication shouldn't need to be synonymous with tool developers and that Apple's Macbook Air M1 models will prove to be very popular in tertiary learning environments, potentially opening up more 'seats' for ghdl. Throw in a dash of platform storage limitations intended to encourage Cloud usage limiting the number of simultaneously supported development environments which ghdl does not inherently require other than by currently supported build process methodology currently aligned with Linux which has OS version to tool version tie-ins, previously avoided for MacOS by relying on statically linked libLLLVM libraries.

There's also the promise of increasing ghdl performance on M1 once supported by gcc/gant or a gnat-llvm effort proves sustainable. Both are being actively developed. You could note that current M1 platforms are available with optional storage sizes to 2 TB at $180 US per 250 MB, limited by discretionary spending, the Christmas spirit, or back to school budget.

ibkvictor, 2022-03-21:
`make install` error

Macbook air 2015
macOS 11.3

```bash
cstdio.o grt-cgnatrts.o grt-cvpi.o grt-cvhpi.o grt-cdynload.o fstapi.o lz4.o fastlz.o
ld: library not found for -lz
collect2: error: ld returned 1 exit status
gnatlink: error when calling /Users/user/opt/GNAT/2020/bin/gcc
gnatmake: *** link failed.
make: *** [ghdl_mcode] Error 4
```

please help

diogratia, 2022-03-22:
```none
gnatbind -aI../src -aI../src/vhdl -aI../src/synth -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/mcode -aI../src/synth -E -x ghdl_jit.ali
gnatlink ghdl_jit.ali -o ghdl_mcode -g memsegs_c.o chkstk.o jumps.o times.o grt-cstdio.o grt-cgnatrts.o grt-cvpi.o grt-cvhpi.o grt-cdynload.o fstapi.o lz4.o fastlz.o
ld: library not found for -lz
collect2: error: ld returned 1 exit status
gnatlink: error when calling /Users/user/opt/GNAT/2020/bin/gcc
gnatmake: *** link failed.
make: *** [ghdl_mcode] Error 4
```

This can also occur for later MacOS versions. (Here MacOS 12.3 with the correct Xcode Command Line Tools loaded.)

The proximate cause is Apple removing zlib.h from /usr/include in an earlier release and removing the compatibility mode from the Big Sur SDK.

The problem is illustrated in python-pillow[https://github.com/python-pillow/Pillow/issues/4816] where the zlib.h  isn't found in the /usr/bin/include search directories but is found in the /Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk/usr/include directory. 

The library libz is found relational from the include in the file system hierarchy:

```none
%: ls ../lib/libz*
/Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk/usr/lib
../lib/libz.1.1.3.tbd@  ../lib/libz.1.2.5.tbd@  ../lib/libz.1.tbd
../lib/libz.1.2.11.tbd@ ../lib/libz.1.2.8.tbd@  ../lib/libz.tbd@
%:
```
where the linker (/usr/bin/ld) knows how to interpret the suffix. 

There's a couple of ways to get the build to find zlib, and some depend on which SDKs you have present.

A quick fix is in a clean build directory, invoke configure with passed environment. e.g.:
```bash
%: ls
/Users/user/Desktop/ghdl/ghdl
COPYING.md      appveyor.yml    ghdl.gpr.in     pyGHDL/         src/
Makefile.in     build/          ghw/            pyproject.toml  testsuite/
NEWS.md         configure*      libraries/      scripts/
README.md       doc/            logo/           setup.py
%: 
```
```none
%: pwd
/Users/user/Desktop/ghdl/ghdl/build
%: ../configure LDFLAGS="-L/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib" CFLAGS="-I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include"
```
And this allows the build to complete (here with make, I'm not installing the mcode version in /usr/local):

```none
%: ./ghdl_mcode --version
GHDL 2.0.0-dev (1.0.0.r1015.g3a3e2f018) [Dunoon edition]
 Compiled with GNAT Version: Community 2020 (20200818-84)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%:
```
There may be easier fixes. I found and tested this one while writing this comment. For instance you can download older SDKs that may have a wider include search path MacOSX10.16 SDK for example when `export SYSTEM_VERSION_COMPAT=1` has been set prior to configuring in the environment. The earlier SDK isn't available on my MacOS 12.3 system without chasing down how to download and test it, and the method may have been extinguished.  Whoever gets the builds go using Xcode 12.3+ may have to research various methods to find the ghdl friendliest method.

I had actually installed zlib in /usr/local in 2013 for the same reason and it wasn't found which says /usr/local isn't in search path either. This would undoubtedly be a security feature.

diogratia, 2022-03-22:
That downloads the latest nightly "https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-macos-10.15-mcode.tgz" (which avoids the building under Big Sur or later issue).

From your first comment someone could be led to believe there was a specific reason to build ghdl from source instead of using a nightly build. The idea being nightly builds can change upto that often.

diogratia, 2022-07-07:
I successfully built an arm64 native version of ghdl-llvm on a Macbook Air M1 using Simon Wright's [GCC 12.1.0 (aarch64)](https://github.com/simonjwright/distributing-gcc/releases/tag/gcc-12.1.0-aarch64-1) package (download at bottom of the page installs in /opt) and the Homebrew arm64 llvm13 (installs in /opt/homebrew).

The configuration used:

%:  ../configure --prefix=/opt/ghdl  --with-llvm-config=/opt/homebrew/Cellar/llvm/13.0.1_1/bin/llvm-config CC=/usr/bin/gcc CXX=/usr/bin/c++  LDFLAGS="-L/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib" CFLAGS="-I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include"

where the LDFLAGS and CFLAGS enable the use of the SDK libz. These represent using the default SDK (with the Command Line Tools by default the last installed).

The resulting binary build had a couple of gotchas. It will fail linking ghdl-llvm which can be overcome by adding libgcc to the gnatlink command (here manually executed:

```none
gnatlink ortho_code_main.ali -o ghdl1-llvm -g llvm-cbindings.o grt-cstdio.o -lgcc --LINK=clang++ -L/opt/homebrew/Cellar/llvm/13.0.1_1/lib  -L/opt/gcc-12.1.0-aarch64/lib/gcc/aarch64-apple-darwin21/12.1.0 -Wl,-search_paths_first -Wl,-headerpad_max_install_names -lLLVM-13 -Wl,-no_compact_unwind
```

Then executing make, make install to complete the build.

Simulating runs into a problem with not passing `-Wl,-no_compact_unwind` to grt which can be overcome by appending it to lib/ghdl/grt.lst:

```none
@/libgrt.a
-L./
-lz
-Wl,-no_compact_unwind
```
Tristan tells me the Makefile could be modified to do the appending during build/install.

There's a run time requirement for libgcc:

```none
%: ldd *
ghdl:
	@rpath/libgcc_s.1.1.dylib (compatibility version 1.0.0, current version 1.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
ghdl1-llvm:
	/opt/homebrew/opt/llvm/lib/libLLVM.dylib (compatibility version 1.0.0, current version 13.0.1)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
	/opt/homebrew/opt/llvm/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1.0.0)
ghwdump:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.100.3)
%:
```
Here ldd is an alias to Otool -L which could be used to find how relocatable the location of the dylib actually is actually is.

Note the libc++ dynamic library was specified from llvm13 despite the setting for CXX.

Without regard to load library paths I've emptied the Homebrew install out preserving LibLLVM.dylib (90 MB) and libc++.dylib in situ (paths intact) and run the ghdl without the rest of the Homebrew/llvm13 1.6 GB overhead.

The libgcc_s.1.1.dylib provides math 'macros' specific to the ARM architecture too big to inline. libLLVM.dylib is the llvm compiler back end.

```none
%: ghdl --version
GHDL 3.0.0-dev (2.0.0.r444.ga33efb91d) [Dunoon edition]
 Compiled with GNAT Version: 12.1.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
```

I haven't tried anything worthy of benchmarking, it seems faster than the x86_64 ghdl-llvm nightly release using Rosetta 2. I'll also getting around to running the ghdl test suite.

tgingold, 2022-07-08:
Great!  Thank you for the whole work.

It would be nice if we could generate binaries like for the other platforms, but I haven't seen yet ARM macos platforms available for CI.

kms1212, 2024-09-20:
For the latest macOS environment, I succeeded in building GHDL using @diogratia 's method, but a little bit different. 

I downloaded and installed [Simon Wright's GCC 12.2.0](https://github.com/simonjwright/distributing-gcc/releases/tag/gcc-12.2.0-aarch64) instead of 12.1.0 since there's no package available for it and cannot build GCC 12.1.0 in latest macOS.

Then I downloaded GHDL 4.1.0 source and modified `ghdl-4.1.0/src/ortho/llvm6/Makefile` lines from 13 to 18.

**Before**
```makefile
$(ortho_exec): $(ortho_srcdir)/llvm6/ortho_llvm.ads force llvm-cbindings.o
	$(GNATMAKE) -o $@ -aI$(ortho_srcdir)/llvm6 -aI$(ortho_srcdir) \
	$(GNATFLAGS) ortho_code_main -bargs -E \
	-largs llvm-cbindings.o $(GNATLFLAGS) --LINK=$(CXX) \
	$(LLVM_LDFLAGS) $(LDFLAGS)
```

**After**
```makefile
$(ortho_exec): $(ortho_srcdir)/llvm6/ortho_llvm.ads force llvm-cbindings.o
	$(GNATMAKE) -o $@ -aI$(ortho_srcdir)/llvm6 -aI$(ortho_srcdir) \
	$(GNATFLAGS) ortho_code_main -bargs -E \
	-largs llvm-cbindings.o $(GNATLFLAGS) -lgcc \
	--LINK=$(CXX) -L/opt/gcc-12.2.0-aarch64/lib/gcc/aarch64-apple-darwin21/12.2.0 \
	`$(LLVM_LDFLAGS)` $(LDFLAGS) -v
```

Then I executed configure and make as follows:
```sh
$ ../configure --prefix=/opt/ghdl --with-llvm-config=/opt/homebrew/opt/llvm/bin/llvm-config CC=/usr/bin/gcc CXX=/usr/bin/c++ LDFLAGS="-L/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib" CFLAGS="-I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include"
$ make
```

I ran a simulation of my project with cocotb and it worked perfectly.

renaudguerin, 2026-09-13:
FYI because of this issue (it's [failing Gatekeeper checks](https://docs.brew.sh/Acceptable-Casks#platform-compatibility-and-macos-security-protections)), ghdl has been [disabled in homebrew](https://github.com/Homebrew/homebrew-cask/pull/252794
) since 1 September.

Paebbels, 2026-09-13:
We are not the maintainers of GHDL in Homebrew.

Please contact the appropriate maintainer.  
The maintainer is responsible for signing code build from sources.  
Moreover the recipe at Homebrew just downloads the binary we create for testing the macOS build scripts. The tarball can be used for a manual copy-deployment.

GHDL is successfully packaged for MSYS2, ArchLinux, Debian, Ubuntu, FreeBSD, ...

renaudguerin, 2026-09-13:
> We are not the maintainers of GHDL in Homebrew.
> Please contact the appropriate maintainer. The maintainer is responsible for signing code build from sources. 

That is correct for Homebrew _formulas_, but ghdl in Homebrew is currently a _Cask_, where binaries are downloaded directly from upstream. Whoever maintains this saw you had official macOS binary releases on GitHub and decided they could point users at them.
You're correct that it isn't the best practice for open source CLI tools in Homebrew (which they compile and sign themselves) and not technically your problem, but it is also not entirely unreasonable to expect upstream-provided binaries to be working out of the box.

> Moreover the recipe at Homebrew just downloads the binary we create for testing the macOS build scripts. The tarball can be used for a manual copy-deployment.

The formula uses the [macOS URL](https://github.com/Homebrew/homebrew-cask/blob/62cd09976aef81ef82adc928013c7a86d06df338/Casks/g/ghdl.rb#L53) from your [official GitHub releases page](https://github.com/ghdl/ghdl/releases#release-v6.0.0) : this is what Casks do, and I find it a bit curious to characterize this release as a random "testing" artifact.

Yes Homebrew should ideally build themselves, but again if you provide macOS binaries it is not entirely unreasonable to expect them to work.

> GHDL is successfully packaged for MSYS2, ArchLinux, Debian, Ubuntu, FreeBSD, ...

As you will be aware, macOS has more stringent signing requirements.

Paebbels, 2026-09-13:
According to AI search on Homebrew, GHDL is wrongly packaged as a Cask, which is a GUI application. GHDL is a TUI application and should be a Formulae (bottle?) - here: signing isn't required.