ghdl/ghdl #1116: synth: wrongly reported "direction mismatch in slice"

Xiretza, 2020-01-26:
**Description**
The example below exclusively uses `downto` SLVs, yet the slice on line 21 is reported as having a mismatched direction during synthesis. Analysis, elaboration and running are unaffected.

**Expected behaviour**
The example synthesises without error.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
	port (
		clk : in std_logic;
		i : in std_logic_vector(31 downto 0);
		o : out std_logic_vector(7 downto 0)
	);
end;

architecture a of ent is
	function switch_endianness(x : std_logic_vector(31 downto 0)) return std_logic_vector is
	begin
		return x(7 downto 0) & x(15 downto 8) & x(23 downto 16) & x(31 downto 24);
	end function;
begin
	process(clk)
	begin
		if rising_edge(clk) then
			o <= switch_endianness(i)(7 downto 0);
		end if;
	end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS: Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 2af18ead

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
ent.vhd:21:49:error: direction mismatch in slice

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-1535-g2af18ead) [Dunoon edition]
Compiled with GNAT Version: 9.2.0
Target: x86_64-pc-linux-gnu
/home/xiretza/temp/ghdl_synth_tests/
Command line:
ghdl --synth ent.vhd -e ent
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:1347
Call stack traceback locations:
0x5632eff3b122 0x5632efca6604 0x5632efe18284 0x5632efe3ae9a 0x5632efe1512e 0x5632efe185c8 0x5632efe20cfe 0x5632efe18b91 0x5632efe20cde 0x5632efe214f5 0x5632efe22f44 0x5632efe23101 0x5632efe14746 0x5632efe14871 0x5632efe0dbbb 0x5632efe6ddc6 0x5632efdaffca 0x5632eff209df 0x5632efb68d64 0x7f630e0eb151 0x5632efb6758c 0xfffffffffffffffe
******************************************************************
```

tgingold, 2020-01-26:
I think GHDL is correct.  The result of the concatenation has the 'to' direction.
(ok, it shouldn't crash).

Xiretza, 2020-01-26:
Interesting. Why is that, and how can it be changed to `downto`? If it really is an error, it should probably fail in simulation too.

Xiretza, 2020-01-26:
Of note is that the same thing happens even if a subtype with a range constraint is used:

```vhdl
subtype word is std_logic_vector(31 downto 0);

function switch_endianness(x : word) return word is
begin
	return x(7 downto 0) & x(15 downto 8) & x(23 downto 16) & x(31 downto 24);
end function;
```

I'm definitely no LRM expert, but this just seems wrong to me. Either way, the problem can be avoided by assigning to a variable, then returning that:

```vhdl
function switch_endianness(x : std_logic_vector(31 downto 0)) return std_logic_vector is
	variable result : std_logic_vector(31 downto 0);
begin
	result := x(7 downto 0) & x(15 downto 8) & x(23 downto 16) & x(31 downto 24);
	return result;
end function;
```
```

tgingold, 2020-01-26:
yes, when you specify the constraints, the result has those constraints.  I am fixing that.

commit f11a4cf6 (2020-01-26): synth: convert subtype on return.  Fix #1116
closed 2020-01-26 by tgingold