ghdl/ghdl #1117: Support command-line generic overrides for array types

Xiretza, 2020-01-27:
**Is your feature request related to a problem? Please describe.**
To implement [riscv-compliance](https://github.com/riscv/riscv-compliance) testing for my core, I need to create a test bench that is passed the memory location of the resulting test signature. At first, I tried to use a `natural` generic, but since the addresses are >`0x7FFF_FFFF`, they are out of range for this type.

Instead, an `unsigned` generic could be used, except that overriding these on the command line using `-g` is not implemented yet.

**Describe the solution you'd like**
Generics of any type, but especially array types, can be overridden using the `-g` command-line option. This would include binary and hex array literals, as well as integer literals for `numeric_std` types.

**Describe alternatives you've considered**
This can of course be worked around using several `natural` generics to add offsets, but it's not ideal.

**Additional context**

```vhd :file: ent.vhd
library ieee;
use ieee.numeric_std.all;

entity ent is
	generic (
		G : unsigned(31 downto 0)
	);
end;

architecture a of ent is
begin
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent '-gG=x"f000_0000"'
ghdl --elab-run ent '-gG=42'
ghdl --elab-run ent '-gG=16#f000_0000#'
```

eine, 2020-01-27:
Refs VUnit/vunit#588 VUnit/vunit#595

@Xiretza, [JSON-for-VHDL](https://github.com/Paebbels/JSON-for-VHDL) allows to pass complex generics as JSON objects. The generic itself can be a path to a JSON file, stringified JSON content or stringified and base16 encoded JSON content. Function `jsonLoad` supports all three cases. Hence, you can use JSON-for-VHDL to pass arrays, nested objects, etc. with simulators that support strings as generics. Moreover, it's synthesisable.

tgingold, 2020-01-27:
It should work with a string (like 11111111111111111111100000000000).  Ok, using a bit string is simpler in your case.

Xiretza, 2020-03-24:
Just stumbled across this again - seems like only the case of hex array literals for synthesis was fixed. All the testcases posted in the original bug report are still failing; integer literals aren't handled in synthesis either. Here's a more complete list of test cases:

```sh
ghdl -a ent.vhd
ghdl --elab-run ent '-gG=x"f000_0000"'
ghdl --elab-run ent '-gG=42'
ghdl --elab-run ent '-gG=16#f000_0000#'
ghdl --synth '-gG=x"f000_0000"' ent
ghdl --synth '-gG=42' ent
ghdl --synth '-gG=16#f000_0000#' ent
```

The `--elab-run`s fail with `ghdl:error: unhandled type for generic override of 'g'`, all the `--synth`s except for the first crash.

tgingold, 2020-03-25:
But the G generic type is unsigned.  So neither 42 nor 16#f000_0000# are correct literals for that type.

Xiretza, 2020-03-25:
Damn, I'm sorry, I must've confused assignment with comparison. In that case, only hex and binary array literals for simulation remain:

```
ghdl -a ent.vhd
ghdl --elab-run ent '-gG=x"f000_0000"'
ghdl --elab-run ent '-gG="11111111111111111111100000000000"'
```

Xiretza, 2020-06-18:
> only hex and binary array literals for simulation remain:
> 
> ```
> ghdl -a ent.vhd
> ghdl --elab-run ent '-gG=x"f000_0000"'
> ghdl --elab-run ent '-gG="11111111111111111111100000000000"'
> ```

```
./ent:error: unhandled type for generic override of 'g'
./ent:error: error during elaboration
./ent:error: unhandled type for generic override of 'g'
./ent:error: error during elaboration
$ ghdl --version
GHDL 1.0-dev (v0.37.0-760-g1666a4c7) [Dunoon edition]
 Compiled with GNAT Version: 10.1.0
 GCC back-end code generator
[...]
```

Could this be reopened?

eine, 2020-06-25:
@Xiretza, is this a regression? Does it work for synthesis (https://github.com/ghdl/ghdl/commit/55faf476ebc08240e9052d184221b57092e68db9) but not for simulation?

Xiretza, 2020-06-25:
It's not a regression, as far as I can tell it was only ever implemented for synth, not simulation.

commit 592e6bee (2020-02-02): configuration: allow hex-string for generic override.  Fix #1117