ghdl/ghdl #1118: GHDL on CentOS 8?

roeekalinsky, 2020-01-28:
I had been successfully using GHDL (built from source) on CentOS 7.  I'm now needing to migrate to CentOS 8 and am finding that GNAT, for reasons unknown to me, is not supported/available in CentOS 8 (why???).  Consequently, I can't even build GHDL from sources on CentOS 8.

Looking for help / advice on how to proceed in order to get GHDL on CentOS 8, as there doesn't appear to be a straight-forward path.  And I imagine that many others would find this useful, as RHEL / CentOS is a popular platform for FPGA development.

Needless to say, a GHDL package that could be installed in CentOS 8 through yum/dnf, if one were available, would be ideal (hint, hint!).

But short of that, step-by-step instructions on how to manually build/install GHDL on CentOS 8 would still be enormously helpful.

Thanks.

eine, 2020-01-28:
Hi @roeekalinsky, as suggested in the README and in the docs, if `gcc-gnat` is not available in your distribution, you should try with 'GNAT Community Edition' provided by AdaCore. I can confirm that GHDL can be successfully built with GNAT 2019 in a `centos:8` docker container, as I just tried it and the test suite passed. Note that you need, at least, `zlib-devel`, `glibc-devel` and `glibc-headers`. Moreover, the installation of GNAT is done through a GUI. It seems not possible to extract the content from the shell.

roeekalinsky, 2020-01-29:
Thanks.  Got it working.

Also worth noting that CentOS 8 comes with LLVM 8.0.1, which isn't officially supported by GHDL v0.36, but appears to work fine as far as I can tell.  The GHDL v0.36 install config script allows LLVM version up to 7.0 only, but modifying it to allow 8.0 let me get through the build/install, and all appears to be working well.

What LLVM version did you use in your CentOS 8 test suite?  To your knowledge, is there any known problem with using LLVM 8.0 with GHDL v0.36, or was it simply never tested before GHDL v0.36 was released?

Thanks.

eine, 2020-01-29:
Glad to hear that. I didn't try the LLVM backend, but mcode. Hence, I did not install/use llvm o clang.

You need not to worry. As you guessed, LLVM 8.0 was not tested when v0.36 was released. Actually, bumping support for it was a matter of modifying the configure script: https://github.com/ghdl/ghdl/commit/7328eae022b0b85c2071ea4d4aaad0c7a97ca436

Anyway, GHDL is a very rapidly rolling project. Hence, lots of new fixes and new features were added since v0.36. I strongly suggest to use a more recent SHA. Actually, you can use the Fedora/RHEL package as a reference: https://src.fedoraproject.org/rpms/ghdl/commits/master. @sharkcz has been taking care of bumping it lately (#888).

eine, 2020-01-29:
> Needless to say, a GHDL package that could be installed in CentOS 8 through yum/dnf, if one were available, would be ideal (hint, hint!).

Is this any different from https://src.fedoraproject.org/rpms/ghdl/commits/master. I know that RHEL, CentOS and Fedora are tightly related, but TBH I don't know any similar CI service for CentOS.

roeekalinsky, 2020-01-29:
Thanks, @eine.

I don't know what's involved in packaging for CentOS vs Fedora, not my area of expertise, but I imagine it is quite similar since CentOS 8 is a clone of RHEL 8 which is largely derived from Fedora 28.

About further improvements in GHDL since v0.36, I'm very happy to know that there's lots of development activity happening.  But for my current use of GHDL on a client project, I'm not in a position to use bleeding edge development snapshots.  I really need to be on a stable and thoroughly tested release of GHDL for my verification platform (same reason why I'm using CentOS and not Fedora -- stable and thoroughly tested).  So I'm looking forward to upgrading to the next GHDL release, whenever that might occur.

sharkcz, 2020-01-29:
https://fedoraproject.org/wiki/EPEL is the way to add packages from Fedora to RHEL/CentOS, but it would need adding the Ada front-end first, if I understand right. So doable, but I don't have the capacity.

eine, 2020-01-29:
> But for my current use of GHDL on a client project, I'm not in a position to use bleeding edge development snapshots.

@roeekalinsky, precisely, the point with GHDL is that every commit is more stable than the previous one. Tagged released are not guaranteed to be any more stable than any other prior or posterior commit. I understand that you don't want to expose yourself to transitive bugs that might be pushed to master. As long as v0.36 is feature-complete for your needs (say you use VHDL 1993 only for simulation only), no need to update. Furthermore, you are lucky now, since the next release is expected to be published in February. Nevertheless, should you need to update say next summer, because you found some issue with some VHDL 2008 features, please do consider using the SHA in the sources of the RPM packages as an "stable enough pre-release". That will likely be some weeks or months old, it will be proven (at least by all Fedora/RHEL users that update periodically), but not outdated.

Overall, I'm concerned with you getting the feeling that GHDL is less mature than it really is, just because of a yearly release schedule which is unrelated to stability.

roeekalinsky, 2020-02-07:
Hi @eine, I apologize for the delayed reply.  I don't think I was assuming ghdl is less mature than it is.  I've found it to be sufficiently mature and stable to use extensively in a professional setting.  I'm a fan.  (And you are correct, I'm not using VHDL 2008 features on my current project).

With respect to using a release of ghdl versus an arbitrary development snapshot, I was assuming, perhaps incorrectly as I understand your comments, that a release would be more than just a tag slapped on an arbitrarily chosen development snapshot.  In the environment in which I work, a release would generally be more thoroughly tested and more stable / less buggy than some arbitrary development snapshot.  Some time (typically a few weeks) before the target release date, you go into feature freeze (i.e. stop incorporating new features and potentially introducing new bugs) and focus on rigorous testing and fixing known bugs at the current feature set.  My assumption was that your release process would entail something similar.

eine, 2020-02-08:
No need to apologise at all ;)

> I was assuming, perhaps incorrectly as I understand your comments, that a release would be more than just a tag slapped on an arbitrarily chosen development snapshot

> My assumption was that your release process would entail something similar.

Your assumption was incorrect and you understood my comments. It is not exactly a tag slapped on an arbitrary commit because Tristan kind of internally handles his own schedule to prioritise features. That is, he likes to release a version around feb-march each year, so around november-december he starts suggesting the idea to other regular contributors. However, there is nothing like a freeze period, there are no feature or release branches at all (there are no hotfixes, although branches exist), no pre-releases are published, etc. GHDL is mostly a single-man linear history with no public work-in-progress.

I would like to ask you to please visit #1108 and, should you be kind to, expose your use case. As you will find, I'm an advocate of having a more clearly defined release scheme.

> stop incorporating new features and potentially introducing new bugs

I must say that, regardless of the not so orthodox release/versioning scheme, GHDL has had almost no regression bugs (I can only remember one in the last 3y, and I'm not even sure). That's why I started this conversation suggesting you to use pseudo-arbitrary commits (those chosen by Fedora package maintainers) as pre-releases.

Anyway, all this is just general knowledge so you can get familiar with the project. If you are using VHDL 1993 for simulation only, you should be good with v0.36 for now.

JimLewis, 2020-09-25:
It would be nice to have a build for CentOS.   Some EDA vendors only claim compatibility with RHEL, hence, CentOS is the choice for free distributions.

eine, 2020-09-25:
In practice, Fedora is compatible with RHEL too, most EDA vendors work ok on Debian/Ubuntu (some updated their constraints during these last years), and docker/podman images are available. Podman is installed by default in latest Fedora versions; I'm not sure about CentOS. Hence, I don't see the motivation for maintaining CentOS packages myself, as it would be additional work which I cannot handle. Nevertheless, I acknowledge there is a niche of CentOS users, specially in internal company servers that aim to be robust (minimize compatibility issues). Therefore, I'd be open to helping. As commented above, I did successfully build GHDL on CentOS, so it's known to work. A packager is needed only.

JimLewis, 2020-09-25:
When I had CentOS installed (I was too busy to do it myself when I having additional work done to the computer), I did not realize there was another variation of RHEL.   OTOH if CentOS's focus is stability - perhaps that would be better for me.   Does the Fedora build work on CentOS or would we need a separate build for CentOS?

JimLewis, 2020-09-25:
BTW, not it in for the fun of running linux and building my own software.   I have access to most EDA simulators at this point.   

Only doing this so I can test OSVVM with GHDL and submit issues against GHDL.   OSVVM seems to be on the bleeding edge of language implementation for some tools.   It would be nice to have a robust test environment for both windows and linux.  

Currently my windows build seems to be the least path of resistance.

eine, 2020-09-25:
There are several related but different questions here. Let's tackle them one by one:

---

> Does the Fedora build work on CentOS or would we need a separate build for CentOS?

Red Hat, CentOS, Fedora are all part of the same large family: https://upload.wikimedia.org/wikipedia/commons/1/1b/Linux_Distribution_Timeline.svg. Red Hat is the commercially supported product. CentOS is the community driven project that aims to be a replacement for RHEL. Fedora is the "playground" for both of them. That is, Fedora is typically on the bleeding edge of technology, implementing new standards, trying new tools, etc. When specific features/tools are very demanded and proven, those are added to RHEL/CentOS. Since RHEL/CentOS' release schedule is quite slow for nowadays standards, typically additional repos exist for providing new tools which are not in the "standard" repos yet. That's the context where GHDL packages for RHEL/CentOS might fit.

Althought not exactly the same, we might say that CentOS is Debian Stable and Fedora is Debian Sid (unstable). However, versions between CentOS and Fedora are typically further apart. Hence, Fedora packages are likely not to work on CentOS because of two reasons:

- The version of GCC/LLVM/Ada that was used for building the packages on Fedora will not match the ones available on CentOS. It might be possible to use a version built for an older version of Fedora. However, that's throwing a dice and crossing your fingers.
- As Dan commented at the top of this issue, the Ada frontend for GCC is currently missing in CentOS. That is, `gnat` or `libgnat` are not available. As a result, building on CentOS cannot be done as we do for any other "distro" (Debian, Fedora, Arch, termux, Windows -MSYS2-, etc.). Two possibilities exist:
  - Package the Ada frontend, as it is done in other distros.
  - Use AdaCore's Community package. This is what I reported above that I could succesfully use in a CentOS container. I can help anyone reproduce it, either locally or in CI. But I don't have bandwidth for maintaining it.

> OTOH if CentOS's focus is stability - perhaps that would be better for me.

Although several vendors claim to support RHEL only, I've been using Fedora and Debian for years without any issue I can remember. Some Mentor's tools required `redhat-lsb` and some specific version of freetype (see https://github.com/eine/hwd-ide/wiki/Continuous-Integration-(CI)#precision-inside-fedoralatest), but apart from that everything worked as expected (including flexlm licensing).

A very interesting case is Xilinx, who supported RHEL only some years ago. Then, they supported Ubuntu. Now, almost all their products are based on Ubuntu. I did not check what they claim lately, tho.

My point is that we are not in the early 2000s any more. Red Hat is not the only company offering commercial support for Linux. EDA vendors are entitled to fight against the reality, but we should not play that game.

> It would be nice to have a robust test environment for both windows and linux.

This is a completely different issue. You absolutely don't need to "natively" install the environments just for the purpose of testing OSVVM with GHDL (or any other vendor tool). You should install a single OS, the one you are more comfortable with. Then, use VMs/containers for testing any other platform.

If your main OS is a GNU/Linux distribution, you might want to have a separate Windows setup, because virtualizing Windows on Linux is tricky. However, if your main OS is Windows 10, just use either Docker or Windows Subsystem for Linux (WSL) in order to test the tools in multiple distributions.

Moreover, you definitely want to use CI. See [setup-ghdl-ci: Usage](https://github.com/ghdl/setup-ghdl-ci#usage). In ghdl-cosim you can see how it is used in practice for testing the project both on Linux (Ubuntu) and on Windows (MSYS2): https://github.com/ghdl/ghdl-cosim/blob/master/.github/workflows/test.yml#L49-L83. The same procedures are used in tdd-intro: https://github.com/VUnit/tdd-intro/blob/master/.github/workflows/tests.yml#L31-L67. In those examples, the "entrypoint" that triggers the tests is a pytest script. However, the same procedure can be used for executing shell scripts or any other tool. 

Using `setup-ghdl-ci` aims to reproduce the behaviour if you run the test locally on your Windows or Ubuntu workstation, assuming that you installed GHDL natively. Yet, in those same workflows, other jobs showcase how to use docker containers instead. See https://github.com/ghdl/ghdl-cosim/blob/master/.github/workflows/test.yml#L14-L28 and https://github.com/VUnit/tdd-intro/blob/master/.github/workflows/tests.yml#L21-L28.

Those docker commands might be executed on any platform where Docker/Podman are available (your Linux/Windows host), avoiding the native installation of GHDL. If you want to be on the bleeding edge, this is your workflow:

```sh
cd OSVVM_repo
docker pull ghdl/ghdl:buster-mcode
docker run --rm -v /$(pwd)://src -w //src ghdl/ghdl:buster-mcode YOUR_TEST_SCRIPT
```

NOTE: you might want to use `ghdl/ghdl:buster-llvm-7` (LLVM backend) or `ghdl/vunit:llvm` (includes Python), instead of `ghdl/ghdl:buster-mcode`. There are similar images for Ubuntu and Fedora: https://hub.docker.com/repository/docker/ghdl/ghdl/tags?page=1

Then, for installing GHDL natively on Windows (MSYS2):

```sh
wget https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-x86_64-ghdl-llvm-ci-1-any.pkg.tar.zst
pacman -U mingw-w64-x86_64-ghdl-llvm-ci-1-any.pkg.tar.zst

cd OSVVM_repo
YOUR_TEST_SCRIPT
```

> I will probably be installing linux on one of my other machines. Hence, maybe it will be a Fedora build.

That's my Linux distro of choice, so I'm obviously biased, but I think it is a good pick. Packaging of GHDL in Fedora seems to be more easy going for users than Debian/Ubuntu.

rodrigomelo9, 2021-02-08:
Hi. As I commented at #1484, I was compiling GHDL for Centos with GCC support. For Centos 8, I performed the flowing steps (with a free docker image of Centos 8):

* Go to https://www.adacore.com/community, click on `download`, select the Linux installer (bin).
* `chmod +x *.bin` and execute it to install.
* I created the following Dockerfile:
```
FROM centos:centos8

RUN yum update -y
RUN yum -y install gcc

RUN gcc --version

RUN yum -y install git curl bzip2 wget flex fontconfig libX11 zlib-devel make
```
* Run the Docker (having access to the installed GNAT) and execute
```
export PATH="<GNAT_INSTALL_PATH>/bin:$PATH"

mkdir ghdl && curl -L https://github.com/ghdl/ghdl/archive/master.tar.gz | tar -xz -C ghdl --strip-components=1
mkdir gcc && curl -L https://codeload.github.com/gcc-mirror/gcc/tar.gz/releases/gcc-8.3.0 | tar -xz -C gcc --strip-components=1
cd gcc && sed -i.bak s/ftp:/http:/g contrib/download_prerequisites && ./contrib/download_prerequisites && cd ..

mkdir build8 && cd build8 && ../ghdl/configure --with-gcc=../gcc --prefix=/usr/local && make copy-sources
mkdir gcc-objs && cd gcc-objs
../../gcc/configure --prefix=/usr/local --enable-languages=c,vhdl --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --disable-libquadmath
make -j2 && make install && cd .. && make ghdllib && make install
```
Could be useful to put somewhere? To solve this issue?

Regards,

eine, 2021-02-08:
Can you provide an script that downloads and installs AdaCore's GNAT automatically? If so, we can add CentOS containers to ghdl/docker.

rodrigomelo9, 2021-02-08:
The problem is, I don't know how to deal with the graphical installer of GNAT XD there is a script based installation at https://github.com/AdaCore/gnat_community_install_script, but it depends on **clicks**, so I understand that it needs a running X11. Moreover, I don't know what happens when a new GNAT version is launched (only the last seems available).

tgingold, 2021-02-08:
You can get older versions by going to the more page.

eine, 2021-02-08:
You can use https://github.com/mviereck/x11docker for starting a container with an X11 server, which can be xvfb in CI. The "clicks" script should work in there. See https://github.com/mviereck/x11docker/issues/201 for a related discussion.

Alternatively, as Tristan suggests, an older version of GNAT (maybe 2015 or 2017) should have a different installation binary.

robert-scheck, 2021-09-23:
Even it's a quite old topic, it might be worth to mention that the desired gcc-gnat RPM package for CentOS/RHEL/Rocky 8 is coming to EPEL 8 via [FEDORA-EPEL-2021-5848dcbea8](https://bodhi.fedoraproject.org/updates/FEDORA-EPEL-2021-5848dcbea8). Please note that `gnatgcc` rather `gcc` must be used, because the `gcc` in the distribution is indeed without any Ada/GNAT support further on.

umarcor, 2021-09-23:
FTR, we are currently building centos:7 and centos:8 images in ghdl/docker, using AdaCores community package:

- https://github.com/ghdl/docker/blob/master/build_centos.dockerfile
- https://github.com/ghdl/docker/runs/3627527570?check_suite_focus=true

I should be easy to adapt the dockerfile once the new gcc-gnat is available.

dlmiles, 2023-01-22:
On centos:stream8 I needed:

dnf install epel-release
dnf install gcc-gnat-8.5.0 make zlib-devel
## Do not use: dnf group install -y "Development Tools"
## first as it appears gcc-8.5.0-18 has an alias to gcc-gnat (FWIW gcc-gnat-8.5.0-17 is the latest at this time)

Trying to install package name as "gcc-gnat" either tries to install "gcc" only, or yields "gcc is already installed" message.

## For: make test
dnf install -y python3 python3-pytest python3-pip
pip3 install pyproject.toml
pip3 install -r pyGHDL/requirements.txt

robert-scheck, 2023-01-22:
@dlmiles, this is mainly because you are using CentOS Stream 8, which is the development branch of Red Hat Enterprise Linux 8 and its derivatives (such as Rocky Linux 8). However, EPEL 8 does not target CentOS Stream 8, but RHEL 8 (and its derivatives). Technically it will work as expected if the NEVRA ("version") of gcc-nat is at least equal to that one of gcc (which is not always the case if you use a development branch, as initially mentioned).